MarlinFirmware/Marlin #24547: When homing, X and Y just move 1mm and stop thinking they are at the endstop. Z home's normal.

NeutronKEKS, 2022-07-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I home the Printer, X and Y move's 1mm sops and thinks it's at X0 Y0 and continues tho home Z and crashes in the Frame, and Z homes completely normal. Also, all other movements function normally. I can move all axes correctly, heat the Nozzle/Bead, and can control all my fans.

I have a SKR 3 white EZ2226, standard microswitches as endstops, and 0.9° Steppers from E3D.

### Expected behavior

To home all axis. (X,Y,Z)

### Actual behavior

Axis X and Y just move 1mm, then stop and think the ar at X0 Y0.

### Version of Marlin Firmware

Marlin-2.1

### Printer model

Longer LK5 Pro

### Electronics

SKR 3, TFT35 V3, 0.9° Stepper E3D, BIQU H2,

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9184493/Configuration.zip)
[Log.txt](https://github.com/MarlinFirmware/Marlin/files/9184571/Log.txt)

EvilGremlin, 2022-07-25:
looks like user error, forgot to remove sensorless homing jumpers

NeutronKEKS, 2022-07-25:
No I have removed the jumpers. The issue was the same with sensorless homing.

Bob-the-Kuhn, 2022-07-25:
Try a bigger homing bump.  Maybe there's some mechanical hysteresis or electrical noise that makes the printer think the endstops are still active so you don't see the bump.

A video would be helpful.

NeutronKEKS, 2022-07-25:
https://user-images.githubusercontent.com/95710705/180874872-640d4e9f-b96a-45ce-a222-505fd6e6a1f7.mp4

NeutronKEKS, 2022-07-25:
I had to compress the video, so I didn't send it at first.

NeutronKEKS, 2022-07-25:
You can't see it very well, but right at the start, it moves a tiny bit, stops, and then moves to the "middle of the Bead" to home Z.

thisiskeithb, 2022-07-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-25 by thisiskeithb

Bob-the-Kuhn, 2022-07-26:
Is this correct?
- Does the X bump for homing
- Does the Y bump for homing
- Moves the head off the board
- Tries to home the Z axis

My guess is the X and Y values for DEFAULT_AXIS_STEPS_PER_UNIT are off by a factor of 2.  Did you recalibrate this after changing to the 0.9 degree steppers?

DEFAULT_AXIS_STEPS_PER_UNIT now:  { 161.97, 161.97, 2000, 932 }
DEFAULT_AXIS_STEPS_PER_UNIT should be ?  { 323.94, 323.94, 2000, 932 }

Bob-the-Kuhn, 2022-07-26:
I got that backwards.  It's already sending out too many steps.  Doubling the values would make it worse.

Best thing to do would be to command a 100mm move, measure the actual move and then adjust accordingly.

NeutronKEKS, 2022-07-26:
Yes, x moves like a bump in the wrong direction (to the right), and I think y also because -y is in the front of the printer and +y in the back.

The steps are correct. I can move the x and y normally. The 161,97 are the proper steps for 100mm travel. It only moves in the wrong direction when homing.

1. X moves right (wrong direction) 1mm and stops.
2. Y moves back 1mm and stops.
3. Bed move, and nozzle moves to the "middle" and crashes.
4. Z home in the wrong place but works correctly.

Bob-the-Kuhn, 2022-07-26:
It doesn't make sense to me, given the info I have so far.  Please assume something is wrong and try the following:
- Increase homing bump to 10 mm HOMING_BUMP_MM (10,10,10).  That'll make it very obvious what is happening.
- Command a 100mm move in each direction and measure the actual moves.
- Do a M502 followed by a M500 and repeat the 100mm move/measurements.
- Do a M122 to see if the TMC2209 comms are good.

I'm thinking the bump and the Z going off board are two seperate problems.

There is always mechanical hysteresis on the endstop switches.  Perhaps your's is just bigger than most.  10mm bump should be way more than needed.

The log agrees with your configuration files.  A move of the head that puts the Z probe in the middle of the bed was commanded.  The head moving further than expected can only mean that DEFAULT_AXIS_STEPS_PER_UNIT is wrong.  Either the calculation that generated those numbers is wrong or the TMC2209s are running 8 microsteps rather than the 16 shown in configuration_adv.h.  The wrong microstep settings could be because the EEPROM didn't get updated since the last change (M502 then M500).  Maybe the TMC2209s are running in standalone mode.

NeutronKEKS, 2022-07-26:
I have done as you sad. Because howe I measure I could only measure 10mm. At first, I had X=10.12mm and Y=10mm. After M502 and M500, X=9.99mm and Y=10mm. I think that the Z going off the bed is not a problem on its one. I believe ist the aftermath of the not correct homing X and Y axes. It's not that the head moves too far. It just starts, not from the endstopp, because after it moves for 1mm, it thinks it has homed and then travels from there to the "middle."

And I have as stepper drivers, the new EZ2226 from BTT should be the same as the TMC2226, only a new form factor.

The driver should work in URAT mode. I have configured them right. Or not?
I also have made a better video of the problem.

https://user-images.githubusercontent.com/95710705/181096100-3c0d6c7c-1829-4b34-9dde-e6875b39c5ca.mp4

thisiskeithb, 2022-07-26:
This isn't a Marlin bug. Please use the resources [linked in my comment above](https://github.com/MarlinFirmware/Marlin/issues/24547#issuecomment-1194644641) for configuration / troubleshooting.

NeutronKEKS, 2022-07-26:
Sorry, I am waiting to be accepted into the Facebook Group and the Marlin forum. And Bob-the-Kuhn responded after you closed the thread, so I also responded. 😅

Bob-the-Kuhn, 2022-07-26:
Sorry - didn't realize it was still visible to the group.