MarlinFirmware/Marlin #24548: [BUG] (bug summary)S_CURVE_ACCELERATION compilation error for STM32F070x: selected processor does not support `umull r2,r3,r3,r4' in Thumb mode

DD-Y41613, 2022-07-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

by enabling "S-Curve Acceleration" ("#define S_CURVE_ACCELERATION" in "Configuration.h"), I run into the error (see below) during compilation.
It is basically the assembler routine in "stepper.cpp", which uses the "UMULL" instruction - the STM32F070-based controller does not support in "Thumb Mode". And the compiler run with the parameters ... '-mcpu=cortex-m0' '-mthumb' ...

Work-Around for me is ... simply not enable the "S-Curve Acceleration".
Hopefully the developer of "stepper.cpp" could have a look and decides, how to deal with this topic.

I did run the test with 2.1 release version of Marlin as well as with the hotfix of the 22th July and bugfix of today.
Originally I do have testing with my configuration for the Malyan Mini Delta.
For this bug report, I took the sample configuration for Malyan M200 (made a minor change regarding the fans) and the changes required to reproduce the issue
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/9187261/Marlin_config.zip)

Configuration.h
```
:
#define MOTHERBOARD BOARD_MALYAN_M300
:
#define S_CURVE_ACCELERATION
:
```

For any questions, don't hessitate to contact me, or reply on my post.

Kind regards,
DD-Y 41613

=== VSCode "Auto Build Marlin" page ===
Marlin Firmware Auto Build
Firmware 2.1
:
Board: MALYAN M300
STM32F070-based delta
:

### Bug Timeline

new, at least for me ... haven't tried to compile Marlin for the M300 board

### Expected behavior

If the S-Curve Acceleration is not supported for the Malyan M300 board (aka MPMD), it's fine with me ... if during the compilation a meaningful message comes up.

I do have no idea, how much effort it will be to allow the S-Curve Acceleration on that Malyan M300 board. And I do have also no idea what other boards will be effected.
But it would be nice to have this feature.  :-)

### Actual behavior

Compilation fails

Output of compilation output with "-v"
```
:
COLLECT_GCC_OPTIONS='-o' '.pio\build\malyan_M300\src\src\module\stepper\indirection.cpp.o' '-c' '-Wno-register' '-std=gnu++14' '-std=gnu++14' '-fno-threadsafe-statics' '-fno-rtti' '-fno-exceptions' '-fno-use-cxa-atexit' '-g3' '-v' '-fmax-errors=5' '-Os' '-mcpu=cortex-m0' '-mthumb' '-ffunction-sections' '-fdata-sections' '-Wall' '-nostdlib' '--param' 'max-inline-insns-single=500' '-D' 'PLATFORMIO=60103' '-D' 'STM32F070xB' '-D' '__MARLIN_FIRMWARE__' '-D' 'NDEBUG' '-D' 'HAL_STM32' '-D' 'USBCON' '-D' 'USBD_USE_CDC' '-D' 'TIM_IRQ_PRIO=13' '-D' 'ADC_RESOLUTION=12' '-D' 'HAL_PCD_MODULE_ENABLED' '-D' 'DISABLE_GENERIC_SERIALUSB' '-D' 'HAL_UART_MODULE_ENABLED' '-D' 'BOARD_F_CPU=48000000L' '-D' 'SERIAL_RX_BUFFER_SIZE=128' '-D' 'SERIAL_TX_BUFFER_SIZE=64' '-D' 'USART_RX_BUF_SIZE=128' '-D' 'USART_TX_BUF_SIZE=64' '-D' 'STM32F0xx' '-D' 'ARDUINO=10808' '-D' 'ARDUINO_ARCH_STM32' '-D' 'ARDUINO_MALYANM300_F070CB' '-D' 'BOARD_NAME="MALYANM300_F070CB"' '-D' 'HAL_UART_MODULE_ENABLED' '-D' 'VECT_TAB_OFFSET=0x2000' '-D' 'CUSTOM_STARTUP_FILE' '-D' 'HAL_PCD_MODULE_ENABLED' '-I' 'Marlin' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\Wire\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\Servo\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\SoftwareSerial\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\EEPROM\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\SPI\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\IWatchdog\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\avr' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\OpenAMP' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F0xx_HAL_Driver\Inc' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F0xx_HAL_Driver\Src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\STM32F0xx' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\open-amp\lib\include' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\libmetal\lib\include' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\virtual_driver' '-I' '%USERPROFILE%\.platformio\packages\framework-cmsis\CMSIS\Core\Include' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F0xx\Include' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F0xx\Source\Templates\gcc' '-I' '%USERPROFILE%\.platformio\packages\framework-cmsis\CMSIS\DSP\Include' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\variants\MALYANMx00_F070CB' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\variants\MALYANMx00_F070CB' '-mfloat-abi=soft' '-march=armv6s-m'
Compiling .pio\build\malyan_M300\libd15\SPI\utility\spi_com.c.o
 c:/users/dietricl/.platformio/packages/toolchain-gCOLLECT_GCC_OPTIONS='-o' '.pio\build\malyan_M300\src\src\module\temperature.cpp.o' '-c' '-Wno-register' '-std=gnu++14' '-std=gnu++14' '-fno-threadsafe-statics' '-fno-rtti' '-fno-exceptions' '-fno-use-cxa-atexit' '-g3' '-v' '-fmax-errors=5' '-Os' '-mcpu=cortex-m0' '-mthumb' '-ffunction-sections' '-fdata-sections' '-Wall' '-nostdlib' '--param' 'max-inline-insns-single=500' '-D' 'PLATFORMIO=60103' '-D' 'STM32F070xB' '-D' '__MARLIN_FIRMWARE__' '-D' 'NDEBUG' '-D' 'HAL_STM32' '-D' 'USBCON' '-D' 'USBD_USE_CDC' '-D' 'TIM_IRQ_PRIO=13' '-D' 'ADC_RESOLUTION=12' '-D' 'HAL_PCD_MODULE_ENABLED' '-D' 'DISABLE_GENERIC_SERIALUSB' '-D' 
'HAL_UART_MODULE_ENABLED' '-D' 'BOARD_F_CPU=48000000L' '-D' 'SERIAL_RX_BUFFER_SIZE=128' '-D' 'SERIAL_TX_BUFFER_SIZE=64' '-D' 'USART_RX_BUF_SIZE=128' '-D' 'USART_TX_BUF_SIZE=64' '-D' 'STM32F0xx' '-D' 'ARDUINO=10808' '-D' 'ARDUINO_ARCH_STM32' '-D' 'ARDUINO_MALYANM300_F070CB' '-D' 'BOARD_NAME="MALYANM300_F070CB"' '-D' 'HAL_UART_MODULE_ENABLED' '-D' 'VECT_TAB_OFFSET=0x2000' '-D' 'CUSTOM_STARTUP_FILE' '-D' 'HAL_PCD_MODULE_ENABLED' '-I' 'Marlin' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\Wire\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\Servo\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\SoftwareSerial\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\EEPROM\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\SPI\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\libraries\IWatchdog\src' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\avr' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\OpenAMP' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F0xx_HAL_Driver\Inc' '-I' '%USERPROFILE%\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32 [... cut ...]
Visual Studio Code - Version 1.69.2
PlatformIO IDE - Version v2.5.0
Auto Build Marlin - Version v2.1.41
2. extract Marlin-bugfix-2.1.x.zip
3. replace the files Configuration.h and Configuration_adv.h by those from the attached zip archive
4. run "build"

### Version of Marlin Firmware

2.1

### Printer model

Monoprice Mini Delta (aka MPMD)

### Electronics

stock electronics with STM32F070CB76 controller

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-07-26:
firstly add these to your config to stop all the endstop warnings
 
```
#define X_MIN_PIN -1
#define Y_MIN_PIN -1

```

And try the following (uses non assembly optimized version of code instead) 

```DIFF
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 593c8f7c6f..fb5f04049f 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -1353,7 +1353,7 @@ void Stepper::set_directions() {
     }
 
     FORCE_INLINE int32_t Stepper::_eval_bezier_curve(const uint32_t curr_step) {
-      #if (defined(__arm__) || defined(__thumb__)) && !defined(STM32G0B1xx) // TODO: Test define STM32G0xx versus STM32G0B1xx
+      #if (defined(__arm__) || defined(__thumb__)) && !(defined(STM32G0B1xx) || defined(STM32F070xB)) // TODO: Test define STM32G0xx versus STM32G0B1xx
 
         // For ARM Cortex M3/M4 CPUs, we have the optimized assembler version, that takes 43 cycles to execute
         uint32_t flo = 0;

```

sjasonsmith, 2022-07-26:
@ellensp I think you might be able to make this more generic be testing for `__thumb2__` rather than `__thumb__`. You’d want to verify it still uses the optimized section for Cortex M3 targets.

if that doesn’t work then I think you could check for `__ARM_ARCH >= 7`.

One of those would let you remove the checks for specific ST product families.

zvonka, 2022-07-30:
I was under the impression that s_curve_acceleration did not work on deltas? I might be confusing it with another setting however that does not work on delta & I might have the names switched in my head.

thinkyhead, 2022-08-04:
According to [this page](https://developer.arm.com/documentation/dui0473/m/arm-and-thumb-instructions/umull) "This 32-bit Thumb instruction is available in ARMv6T2 and above," so we just need to add a test for "ARMv6T2 and above" before accepting `__thumb__` for this block of code. I'm looking into how to detect that ARM version now….

ellensp, 2022-08-04:
user never got back to us that C version works... MPU is very slow 48 MHz,  evnl slower than stm32f103 and a slower architecture...  

Until we know the C version works, its pointless changing code to use C version.

thinkyhead, 2022-08-04:
What is the recommended assembly code substitution for MULL in this situation?

thinkyhead, 2022-08-04:
For now, the additional condition `&& __ARM_ARCH >= 6` (if that pertains to "ARMv6T2 and above") may be added to prevent the incompatible assembly being used until there is a rewrite for STM32F07 in thumb mode.

commit 1c4fc460 (2022-08-04): 🩹 Fix Malyan M300 with S-Curve compile

Fixes #24548
closed 2022-08-04 by thinkyhead