MarlinFirmware/Marlin #24551: [FR] Specify the ABL measure points

fsadak, 2022-07-26:
### Describe the feature you want

I use Prusa MK42 table and Pinda sensor on my 3D printer.  My control board is SKR 1.4 Turbo.  I installed Marlin, everything works fine, but I can't make the bed level.  The reason is that pinda does not sense the places other than the measuring points on the MK42.  I am looking for a method to individually identify these 9 points to measure when using a bilinear abl.  Is it possible to add a setting to specify points to measure in Marlin?  For example;  Is it possible to make an adjustment by giving the starting point, the bed x and y dimensions, the total measuring point and the distances between the x/y measuring points?

EvilGremlin, 2022-07-26:
This existed, like, always - set PROBING_MARGIN_ [LRFB] to finely position probe over points.

fsadak, 2022-07-26:
Cevabınız için teşekkür ederim. Arada kalan diğer 5 nokta için bir ayar gerekmez mi? Marlin noktalar arası mesafeyi nereden bilecek?

fsadak, 2022-07-26:
Thank you for your answer.  Don't you need a setting for the other 5 points in between?  How will Marlin know the distance between points?

EvilGremlin, 2022-07-26:
by simple division, distances are always equal

closed 2022-07-26 by thisiskeithb