MarlinFirmware/Marlin #24552: [FR] bump platform-ststm32 version

dominikberse, 2022-07-26:
### Is your feature request related to a problem? Please describe.

I experienced problems while working with SoftwareSerial due to the fact that the platformio platform-ststm32 version is outdated. The `common_stm32` target declared in `stm32-common.ini` relies on version `12.1` from March 21 which in turn uses Arduino Core version `v1.9.0` from May 2020.

### Describe the feature you want

Update to a newer version (current version is `15.4` with underlying Arduino Core version `v2.2.0`).

I did a quick test build for Ender-5 Pro (STM32F103RE_creality) with 15.4. There seems to be a change in the way board variants are organized, so I had to rename `variant.h` and `variant.cpp` to `variant_generic.h` as well as `variant_generic.cpp` (in `buildroot/share/PlatformIO/variants/MARLIN_CREALITY_STM32F103Rx/`). I assume that this needs to be done for all STM32 variants.

Other than that, everything worked as expected and I had the SoftwareSerial fix I needed.

thisiskeithb, 2022-07-26:
> There seems to be a change in the way board variants are organized
> I assume that this needs to be done for all STM32 variants.

This is why we haven't bumped the version since it'll require quite a bit of work & testing.

dominikberse, 2022-07-27:
I see, thanks for the info. Maybe I can support you to bring this forward? I would probably bump the version number locally and see if I can make the changes required for all tests to pass. For manual testing however the only option I have is the STM32F103 on the Creality and BigTreeTech boards.

zvonka, 2022-07-30:
I have a similar board with similar cpu STM STF446ZET6 iirc, along those lines, it's a BTT octopus & I'm looking to do the same thing you did. Would you be able to share the changes you made specifically so that I may adapt my firmware to take advantage of the fix as well please?

dominikberse, 2022-08-01:
@zvonka For me it was simply the changes I stated above:
 - Change the platform version number in `ini/stm32-common.ini` under the `[common_stm32]` section as required (i.e. `platform = ststm32@~15.4`)
 - Rename the `variant.*` files under `buildroot/share/PlatformIO/variants/<platform>/` to `variant_generic.*` accordingly