MarlinFirmware/Marlin #24558: [BUG] 2.0.9.5 Compilation Failure on BIQU B1 SKR 1.4 appears related to stepper/planner header changes

tssva, 2022-07-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to compile 2.0.9.5 for my Biqu B1 SKR 1.4 with a BLTOUCH. I am using the 2.0.9.5 Biqu B1 SKR 1.4 example configuration file modified to enable the bltouch, USE_PROBE_FOR_Z_HOMING and Z_SAFE_HOMING. Compilation fails with errors which seem like they may be related to the stepper/planner changes. 2.0.9.4 and prior compiles properly.

```
Marlin\src\inc\Warnings.cpp:573:6: warning: #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is enabled. (See https://bit.ly/2ZPRlt0) (Define DIAG_PINS_REMOVED to suppress this warning.)" [-Wcpp]
  573 |     #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is enabled. (See https://bit.ly/2ZPRlt0) (Define DIAG_PINS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
Marlin\src\module\planner.cpp: In static member function 'static void Planner::buffer_sync_block(BlockFlagBit)':
Marlin\src\module\planner.cpp:2864:35: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'block_t' {aka 'struct block_t'} with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
 2864 |   memset(block, 0, sizeof(block_t));
      |                                   ^
In file included from Marlin\src\module\planner.cpp:65:
Marlin\src\module\planner.h:190:16: note: 'block_t' {aka 'struct block_t'} declared here
  190 | typedef struct block_t {
      |                ^~~~~~~
In file included from c:\users\tim\marlin\marlin\src\inc\MarlinConfigPre.h:37,
                 from c:\users\tim\marlin\marlin\src\inc\marlinconfig.h:28,
                 from Marlin\src\module\stepper\trinamic.cpp:28:
c:\users\tim\marlin\marlin\src\core\macros.h:716:7: error: expected unqualified-id before ',' token
  716 |     ( , DEFER2(__MAPLIST)()(_MAP_OP,V) ) \
      |       ^
c:\users\tim\marlin\marlin\src\core\macros.h:630:24: note: in definition of macro 'EVAL1'
  630 | #define EVAL1(V...)    V
      |                        ^
c:\users\tim\marlin\marlin\src\core\macros.h:629:30: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:628:24: note: in expansion of macro 'EVAL2'
  628 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:24: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:30: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:628:30: note: in expansion of macro 'EVAL2'
  628 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:627:24: note: in expansion of macro 'EVAL4'
  627 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:24: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:30: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:628:24: note: in expansion of macro 'EVAL2'
  628 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:24: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:629:30: note: in expansion of macro 'EVAL1'
  629 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:628:30: note: in expansion of macro 'EVAL2'
  628 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:627:30: note: in expansion of macro 'EVAL4'
  627 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                              ^~~~~
c:\users\tim\marlin\marlin\src\core\macros.h:626:24: note: in expansion of macro 'EVAL8'
  626 | #define EVAL16(V...)   EVAL8(EVAL8(V))
[... 558 lines cut ...]
Started with 2.0.9.5

### Version of Marlin Firmware

2.0.9.5

### Printer model

BIQU B1

### Electronics

SKR V1.4

### Add-ons

BLTOUCH

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9212547/configuration.zip)

NiklasOlofsson, 2022-07-28:
Same here. MKS Gen L v1. TMC2208.
2.0.9.5 no go
2.0.9.4 ok

ellensp, 2022-07-29:
line 987 Marlin/src/module/stepper/trinamic.cpp

`  MAPLIST(SA_NO_TMC_SW_C, ALL_AXIS_NAMES)`

should be

`  MAP(SA_NO_TMC_SW_C, ALL_AXIS_NAMES)`

ellensp, 2022-07-29:
fixed in https://github.com/MarlinFirmware/Marlin/commit/761c76a196f9ac890378e29b87b59509ef44524c

closed 2022-07-29 by ellensp