MarlinFirmware/Marlin #24573: [BUG] Marlin 2.1 MKS Tinybee Part Cooling Fan not working

3DSmitty, 2022-07-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Unable to turn on the Part Cooling Fan with M106 on the MKS Tinybee in Marlin 2.1. This works fine with MKS source Marlin 2.0. It does not turn on via command line or LCD. Command Line responds with "ok" and it shows fan is on. LCD shows fan is on. Connected fan directly to 12v and it works fine so no hardware issue. Tried swapping pin numbers between FAN_PIN and FAN1_PIN with same results so it is not a mosfet issue. LED on mosfet circuit does not light up.

### Bug Timeline

new bug started with 2.1

### Expected behavior

Part Cooling fan should turn on using M106.

### Actual behavior

Fan does not turn on.

### Version of Marlin Firmware

2.1

### Printer model

Monoprice MP10

### Electronics

MKS Tinybee

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_2.1_Bug.zip](https://github.com/MarlinFirmware/Marlin/files/9229072/Marlin_2.1_Bug.zip)

descipher, 2022-07-31:
There are a number of changes in the MKS TinyBee source that have not made into the Marlin upstream. Some are specific to FAN_PIN1 etc.  ping @solawc

thisiskeithb, 2022-07-31:
> There are a number of changes in the MKS TinyBee source that have not made into the Marlin upstream. Some are specific to FAN_PIN1 etc. ping @solawc

There's some other weird stuff in MKS' Tinybee firmware like including a [`.pio` folder with modified libraries](https://github.com/makerbase-mks/MKS-TinyBee/tree/main/firmware/mks%20tinybee%20marlin/.pio), so moment you hit clean in VSCode, all of their custom changes are dumped/reset.

descipher, 2022-07-31:
The config upload will no work ...
```
#define FAN_PIN                              147
//#define FAN1_PIN                           148
#define HEATER_BED_PIN                       144
```
No FAN1_PIN

closed 2022-07-31 by thisiskeithb

3DSmitty, 2022-07-31:
@descipher @thisiskeithb Why is this closed? I have pin 148 defined in Configuration_adv.h:
`#define E0_AUTO_FAN_PIN 148`
That is the extruder heatsink fan not the Part Cooling Fan which is:
`#define FAN_PIN                              147`
The extruder heatsink fan works just fine as soon as the hotend reaches 50 it kicks on. But that is not the issue. The issue is with FAN_PIN which is the Part Cooling Fan.

descipher, 2022-08-01:
Your config is not correct.
Return the pins file back to stock config and set the Extruder auto fans correctly.
You cannot use FAN_PIN as E0_AUTO_FAN_PIN 
e.g.
#define E0_AUTO_FAN_PIN FAN1_PIN

3DSmitty, 2022-08-01:
@descipher ok thats fine but that is still not going to solve the problem. 

> You cannot use FAN_PIN as E0_AUTO_FAN_PIN

There must be some confusion I never said I wanted to use FAN_PIN as E0_AUTO_FAN_PIN. I want to control FAN_PIN ( Part Cooling Fan) with M106 as it should be.

descipher, 2022-08-01:
If you do not have FAN1_PIN defined the fan count will be wrong.

3DSmitty, 2022-08-01:
@descipher ok I just changed back and re-flashed. FAN_PIN 147 is still not controllable via M106 (does not turn on) so the issue still exists. 
`#define FAN1_PIN                             148`
`#define E0_AUTO_FAN_PIN FAN1_PIN`

descipher, 2022-08-01:
Does M106 S100 P1 work on the heatsink fan?

3DSmitty, 2022-08-01:
@descipher 

`M106 S100 P1
Ok
ok`

Is the response from the command line but the fan does NOT turn on. If I heat the nozzle up above 50C it kicks on and then back off below 50C.

descipher, 2022-08-01:
That is the correct behaviour for an auto fan, there was some work done on the expander and fan timing control recently. Will look at that to see if it’s causing the M106 FAN0 control problem. Just not likely if the auto fan control works.

3DSmitty, 2022-08-01:
@descipher ok thank you.

3DSmitty, 2022-08-01:
@descipher @thisiskeithb I got it. I had to enable this for it to work:

`#define FAN_SOFT_PWM`

descipher, 2022-08-01:
Should work with or without FAN_SOFT_PWM the code, Looked at the HAL and without FAN_SOFT_PWM it would need FAST_PWM_FAN. The set_pwm_duty call emulates PWM with the HC595 expander.
```
  #if ENABLED(I2S_STEPPER_STREAM)
    if (pin > 127) {
      const uint8_t pinlo = pin & 0x7F;
      pwm_pin_t &pindata = pwm_pin_data[pinlo];
      const uint32_t duty = map(invert ? v_size - v : v, 0, v_size, 0, pindata.pwm_cycle_ticks);
      if (duty == 0 || duty == pindata.pwm_cycle_ticks) { // max or min (i.e., on/off)
        pindata.pwm_duty_ticks = 0;  // turn off PWM for this pin
        duty ? SBI32(i2s_port_data, pinlo) : CBI32(i2s_port_data, pinlo); // set pin level
      }
      else
        pindata.pwm_duty_ticks = duty; // PWM duty count = # of 4µs ticks per full PWM cycle
    }
    else
  #endif
```

3DSmitty, 2022-08-02:
Yes it should but it doesn't. The original Marlin on MKS github for Tinybee does not have FAN_SOFT_PWM enabled and it works so something must have changed.