MarlinFirmware/Marlin #24576: TMC2208 Drivers go nuts during printing

mawais15, 2022-07-31:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I am using Marlin 2.0 with TMC 2208 in UART mode with SKR V1.3 and BTT MOT expansion board. I am using tripple Z auto align feature. Problem i am facing is during printing, my Z2 and Z3 motors start spinning at double the speed that of Z1 (which is true speed) and print gets distorted. I have enabled debugging in the firmware using M122 and i found that when this happens, Z2 and Z3 drivers are drawing more current and their actual steps/mm are halved. Here is debug report when this happens.
SENDING:M122
		X	Y	Z	Z2	Z3	E
Enabled		false	false	false	false	false	false
Set current	700	700	600	600	600	700
RMS current	673	673	581	1049	1049	673
MAX current	949	949	819	1479	1479	949
Run current	21/31	21/31	18/31	18/31	18/31	21/31
Hold current	10/31	10/31	9/31	9/31	9/31	10/31
CS actual	10/31	10/31	9/31	31/31	31/31	10/31
PWM scale
vsense		1=.18	1=.18	1=.18	0=.325	0=.325	1=.18
stealthChop	false	false	true	true	true	true
msteps		16	16	16	8	8	16
interp		true	true	true	true	true	true
tstep		max	max	max	max	max	max
PWM thresh.
[mm/s]
OT prewarn	false	false	false	false	false	false
triggered
 OTP		false	false	false	false	false	false
pwm scale sum	12	12	32	104	84	41
pwm scale auto	0	0	3	0	0	3
pwm offset auto	36	36	93	104	84	111
pwm grad auto	14	14	24	28	30	18
off time	3	3	3	3	3	3
blank time	24	24	24	36	36	24
hysteresis
 -end		-1	-1	-1	-3	-3	-1
 -start		1	1	1	6	6	1
Stallguard thrs
uStep count	376	312	536	784	272	648
DRVSTATUS	X	Y	Z	Z2	Z3	E
sg_result
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x80:0A:00:00
		Y	0x80:0A:00:00
		Z	0xC0:09:00:00
		Z2	0xC0:1F:00:00
		Z3	0xC0:1F:00:00
		E	0xC0:0A:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing Z3 connection... OK
Testing E connection... OK

The drivers are actively cooled and there is no sign of heating however the motors Z2 and Z3 were warm to touch which is understandable because drivers is sending more current than needed.

### Version of Marlin Firmware

2.0.x

### Bed Leveling

ABL 3-point

### Your Slicer

Simplify3D

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-31:
> Did you test the latest bugfix-2.1.x code?
> No, but I will test it now!

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

> Don't forget to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You also forgot to do this.

ellensp, 2022-07-31:
```
              X  Y  Z  Z2  Z3 E
msteps 16 16 16 8    8   16

```
All Z's micro stepping and steps/mm must be the same.

This shows that Z2 and Z3 have 1/8 micro stepping while Z has 1/16th..    Z2 and Z3 will move twice the distance of Z

mawais15, 2022-08-01:
> ```
>        X  Y  Z  Z2  Z3 E
> msteps 16 16 16 8   8  16
> ```
> 
> All Z's micro stepping and steps/mm must be the same.
> 
> This shows that Z2 and Z3 have 1/8 micro stepping while Z has 1/16th.. Z2 and Z3 will move twice the distance of Z

Sorry I forgot to mention that it happens in middle of prints randomly. Steps/mm are correct for all Z steppers. Sometimes in middle of print this issue occurs and if I reset the printer or re-initialize the Tmc2208 through command line, it gets fixed.

mawais15, 2022-08-01:
[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9231091/configurations.zip)
Here is the Configuration_adv.h and Configuration.h files attached.

EvilGremlin, 2022-08-01:
CONFIGURATION_H_VERSION 02000902
this is old, use bugfix-2.1 for starts

closed 2022-10-12 by github-actions[bot]