MarlinFirmware/Marlin #24581: [BUG] MKS H43 costumize failed

FreeZuu69, 2022-08-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Although I did all the costum steps as mentioned on github, there was no change in the theme or the initial logo. I have attached the relevant files and images. Those who have done it before or have knowledge about dgus, please I am waiting for your help.

### Bug Timeline

dwinset dosyalarını attıktan sonra.

### Expected behavior

The dwinset file had to be discarded with a card formatted as fat32 and 4096 and the theme had to be changed. Even though I did dgusta generate, there was no change in the theme.

### Actual behavior

nothing changed and when I logged into the print files, the backspace above stopped working.

### Steps to Reproduce

The first step is to replace the jpeg files.
The second step is to open and generate the files by dgus.
The third step is to start the printer by inserting the file named dwinset in the socket connected to the screen with the micro sd card formatted with fat32 4096.

### Version of Marlin Firmware

Marlin-2.0.9.3

### Printer model

Corexy

### Electronics

Mks S gen L 2.0 , Mks Pwc, Mks H43 LCD, bltouch

### Add-ons

none

### Bed Leveling

ABL 3-point

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[https://www.hizliresim.com/3omt4oq](url)
[https://www.hizliresim.com/rore38t](url)
[https://www.hizliresim.com/heay9q6](url)

ellensp, 2022-08-03:
Not Marlin firmware. 
https://github.com/makerbase-mks/MKS-H43-firmware/tree/main/H43_Firmware_1.31_for_Marlin

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-08-03 by ellensp

FreeZuu69, 2022-08-03:
I am very sorry. I tried to contact all channels of the seller, I did not get any response. I could not get a response from all these channels on Discord, facebook, twitter, aliexpress. I finally decided to post here. I'm so sorry if it's inappropriate here.

EvilGremlin, 2022-08-03:
This is a problem with DGUS software/hardware itself, not Marlin nor MKS.
Most likely your new assets takes too much space, make them same size as original. Read DGUS manuals for requirements.

FreeZuu69, 2022-08-03:
Thanks a lot for your comment. Please excuse me for taking your time.