MarlinFirmware/Marlin #24583: [BUG] reprap rotary encoder not working

gyorgylevente, 2022-08-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The rotary encoder of my reprap discount smart controller with ramps 1.4+mega2560 doesnt work. (btw Marlin is configured to drive a K40 Laser)

### Bug Timeline

Came with fresh install

### Expected behavior

I expect to be able to navigate in the menu with the rotary encoder.

### Actual behavior

I'm unable to navigate in the menu. Click works.

### Steps to Reproduce

1. Configure latest release of marlin 2.1 with REPRAP_DISCOUNT_SMART_CONTROLLER on ramps 1.4+arduino mega 2560
2. Enter the menu with a click
3. turn rotary encoder
 it does not change the selection

### Version of Marlin Firmware

Marlin 2.1

### Electronics

Ramps 1.4 + Mega 2560

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/9255081/conf.zip)

ellensp, 2022-08-04:
Provided configuration doesn't have a LCD enabled.

ellensp, 2022-08-04:
Tested on marlin bugfix 2.1.x and Marlin bugfix 2.0.x 
Works as expected

closed 2022-08-04 by ellensp