MarlinFirmware/Marlin #24592: [BUG] UBL not honoring mesh height changes

Bob-the-Kuhn, 2022-08-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Changing UBL mesh height (via G29 P6 cxxx) isn't affecting my actual Z height.

### Bug Timeline

I noticed this with bugfix_2.1.x from 3 Aug and verified it's still present in the 5 AUG version

### Expected behavior

When I change the UBL mesh height I expect the actual height of the nozzle above the board to change for a given G0 command.

### Actual behavior

It's acting as if bed leveling is off.  Changing the mesh level via G29 P6 Cxxx doesn't affect how far off the bed is when I command G0 Z7.8.  The nozzle is always just touching the bed.

### Steps to Reproduce

1. Flash the firmware
2. G28
3. G29 P1
4. G29 P3 several times to fill out the mesh
5. G29 S1
6. G29 A
7. M500
8. M420 V3 - shows reasonable mesh, active, slot 1
9. power cycle
10. G28
11. G0 Z15
12. Use LCD to lower head until paper drags on nozzle
13. LCD reads 7.8
14. G29 P6 C-7.8
15. G29 S1
16. M500 - says mesh saved in slot 1
17. M420 V3 - shows reasonable mesh, lower by 7.8, active, slot 1
18. power cycle
19. G28
20. G0 Z15
21. Use LCD to lower head until paper drags on nozzle
22. LCD reads 7.8

Doesn't matter what constant I use with G29 P6 Cxxx -  LCD always reads 7.8 when the nozzle is just touching the plate.  I've tried positive numbers and negative numbers.  M420 V3 shows the P6 command is being implemented.

### Version of Marlin Firmware

Send: M115 Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Aug  5 2022 21:30:18) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:TEVO Little Monster EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff Recv: Cap:SERIAL_XON_XOFF:0 Recv: Cap:BINARY_FILE_TRANSFER:0 Recv: Cap:EEPROM:1 Recv: Cap:VOLUMETRIC:1 Recv: Cap:AUTOREPORT_POS:0 Recv: Cap:AUTOREPORT_TEMP:1 Recv: Cap:PROGRESS:0 Recv: Cap:PRINT_JOB:1 Recv: Cap:AUTOLEVEL:1 Recv: Cap:RUNOUT:1 Recv: Cap:Z_PROBE:1 Recv: Cap:LEVELING_DATA:1 Recv: Cap:BUILD_PERCENT:0 Recv: Cap:SOFTWARE_POWER:0 Recv: Cap:TOGGLE_LIGHTS:0 Recv: Cap:CASE_LIGHT_BRIGHTNESS:0 Recv: Cap:EMERGENCY_PARSER:0 Recv: Cap:HOST_ACTION_COMMANDS:0 Recv: Cap:PROMPT_SUPPORT:0 Recv: Cap:SDCARD:1 Recv: Cap:MULTI_VOLUME:0 Recv: Cap:REPEAT:0 Recv: Cap:SD_WRITE:1 Recv: Cap:AUTOREPORT_SD_STATUS:0 Recv: Cap:LONG_FILENAME:0 Recv: Cap:LFN_WRITE:0 Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0 Recv: Cap:EXTENDED_M20:0 Recv: Cap:THERMAL_PROTECTION:0 Recv: Cap:MOTION_MODES:0 Recv: Cap:ARCS:1 Recv: Cap:BABYSTEPPING:1 Recv: Cap:CHAMBER_TEMPERATURE:1 Recv: Cap:COOLER_TEMPERATURE:0 Recv: Cap:MEATPACK:0 Recv: Cap:CONFIG_EXPORT:0

### Printer model

Teva Little Monster (delta printer)

### Electronics

stock MKS SBASE (LP1768)

### Add-ons

Duet Smart Effector

### Bed Leveling

UBL Bilinear mesh

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9274310/Marlin.configuration.zip)

Bob-the-Kuhn, 2022-08-08:
2.0.9.3 works
2.0.9.4 fails

Bob-the-Kuhn, 2022-08-09:
Per git bisect,  the offending commit is:
e0deb75764938995e32bc589d646ec397c4a3019
Date:   Thu May 19 22:15:15 2022 +0100
    🐛 Fix Leveling apply/unapply (#24188)

This is part of PR: Leveling fixes #24188

Bob-the-Kuhn, 2022-08-09:
Just in case it's of any use ... here's a log of the problem with debug leveling turned on.  All it shows is that leveling is on.  
[log2.txt](https://github.com/MarlinFirmware/Marlin/files/9286628/log2.txt)

Basically the head always goes to the same spot whether leveling is on or leveling is off or if the matrix height is changed.

Bob-the-Kuhn, 2022-08-11:
Still scratching my head.  Suggestions appreciated.

I've sprinkled some debug prints around and see that the correct (leveled) destination Z is being passed to **planner.buffer_line**.   In the log below, raw.z is the 10 from "G0 Z10",  "z_cxcy:7.94" is the computed leveling offset and "new raw.z:17.94" is what is being passed to the **planner.buffer_line**.   As best I can tell this is all correct.

M114 D from a passing build is almost identical to the one in the log.  The only differences can be explained by the probing undertainties during G29 P1.  

```
Send: G28
...
Send: G0 Z10
Recv: current_position.z:493.52
Recv: first raw.z:493.52
Recv: leveling is active
Recv: raw.z:10.00
Recv: new raw.z:17.94   z_cxcy:7.94
Recv:   buffer_segment FR:25.00 A:382.16 (461699->203820) B:382.16 (461699->203820) C:382.16 (461699->203820) E:0.00 (0->0)
Recv: ok
[...]
Send: M114 D
Recv: 
Recv: Logical: X: 0.000 Y: 0.000 Z: 10.000
Recv: Raw:     X: 0.000 Y: 0.000 Z: 10.000
Recv: Leveled: X: 0.000 Y: 0.000 Z: 17.940
Recv: UnLevel: X: 0.000 Y: 0.000 Z: 10.000
Recv: DeltaK:  X: 382.162 Y: 382.162 Z: 382.162
Recv: Stepper: X:203820 Y:203820 Z:203820 E:0
Recv: FromStp: X: 0.000 Y:-0.000 Z: 17.940 E: 0.000
Recv: Diff:    X: 0.000 Y:-0.000 Z: 0.000 E: 0.000
Recv: ok
```

Bob-the-Kuhn, 2022-08-11:
I think I've found the smoking gun.  Basically, Bugfix-2.1.x is leveling but the Z home position is wrong so the nozzle is actually gliding along a plane that is incorrect.  The incorrect plane is the Z offset in the mesh for the Z homing position.

I can see why this went un-noticed for so long.  My offsets are huge (about 8mm) compared to most printers so it's much more apparent on my printer.

Attached are two logs.  One is from 2.0.9.3 (shows correct behavior).  The other is from bugfix-2.1.1 (shows incorrect behavior). 
 Both have leveling enabled, then debug_leveling enabled, then G28, then M114 D then G0 Z10 and finally M114 D

In the below screen shot the two are compared side by side.  2.0.9.3 is on the left.  Notice that after G28, the sync_plan_position  and leveling active comments are swapped which results in the final positions being different.  The 2.0.9.3 number is lower by the amount of leveling compensation for the homing Z location.  

[image]
[LOG_bugfix-2.1.x (fails).txt](https://github.com/MarlinFirmware/Marlin/files/9307118/LOG_bugfix-2.1.x.fails.txt)
[LOG_2.0.9.3 (works as expected).txt](https://github.com/MarlinFirmware/Marlin/files/9307119/LOG_2.0.9.3.works.as.expected.txt)

thisiskeithb, 2022-08-12:
Closing since you've opened a PR to fix this:

- https://github.com/MarlinFirmware/Marlin/pull/24622

closed 2022-08-12 by thisiskeithb