MarlinFirmware/Marlin #24594: [MKS Base 1.6 support] (Setting up new printer)

electrovo, 2022-08-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I set up a new printer with MKS base v1.6 board with TFT35 touch screen. I used Visual Basic/PlatformIO to upload the Marlin 2.1. Copied the example file of Configuration.h and Configuration_adv.h to Marlin folder. Change the board name to “#define MOTHERBOARD BOARD_MKS_BASE_16.
However, the test fails and nothing working. Z, Y and X axis aren't moving. Can someone help me with this one please? 

### Bug Timeline

[MKS Base 1.6 support] (Setting up new printer)

### Version of Marlin Firmware

2.1

### Printer model

CTC bizer dual 

### Your Slicer

Simplify3D

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-08-07:
Two things I noticed in your configuration.h file:
- Driver type should be A4988, not DRV8825.  Probably doesn't make a difference.
- I didn't see a LCD defined.  There are YouTube videos on how to setup your MKS TFT35 V1.0 in Marlin.

thisiskeithb, 2022-08-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-08-07 by thisiskeithb

ellensp, 2022-08-07:
seems to be a MKS TFT35 V1.0, it has no lcd emulation mode, so no need for a lcd to be defined.
All it needs is a serial port and a baudrate that matches what the lcd is set to.
exp-1 is serial 0 on the mks base 1.6