MarlinFirmware/Marlin #24603: [BUG] (bug summary)

dwstanley1, 2022-08-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I hope I'm posting this is the right area as I usually find someone has already posted what I'm looking for.  This time I haven't found the my issue.

I am trying to use the T-D500 thermistor and cannot seem to get it to work.  I am using firmware version 2.1.x and the thermistor that came with the spider hotend kit works fine; however, when swapping to the T-D500 with flashing the firmware with the suggested updates (detailed below), the thermistor does not register, it reads 0 (zero).  I noticed the system was already calling for heat but the nozzle was not heating.  I went ahead and tried to increase the temp thinking maybe it would start to read but the system immediately errored/halted.  
config.h:
TEMP_SENSOR_0 66
HEATER_0_MINTEMP 21
HEATER_0_MAXTEMP 500 (I don't need it this hot but I was following instructions)
config.adv.h:
MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 5
MILLISECONDS_PREHEAT_TIME 30000

It was suggested to me by the seller that I also modify the following but it did not change anything.
config.h:
DEFAULT_Kp  14.00
DEFAULT_Ki 0.50
DEFAULT_Kd 125.00
After that didn't work I also updated the following that were referenced.
DEFAULT_Kp_LIST {  14.00,  14.00 }
DEFAULT_Ki_LIST {   0.50,   0.50 }
DEFAULT_Kd_LIST {  125.00,  125.00 }

Any help you can give me would be great!

David
____________________
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/9277778/Configuration.files.zip)

### Bug Timeline

never worked

### Expected behavior

I expected to see a reading registered above 0 (zero) upon starting.

### Actual behavior

nothing

### Version of Marlin Firmware

2.1.x

### Printer model

ender 3 pro

### Electronics

board 4.2.2

### Add-ons

BLTouch, spider hotend nozzle 

### Your Slicer

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-08-08:
Does the indicated temperature change if you remove the thermistor from the controller?  I'm wondering if it is open or there is a bad connection.

dwstanley1, 2022-08-08:
Thanks for your reply,

I went back and forth on the connection, even used the backup thermistor thinking maybe the first one was faulty. I even wondered if they were both faulty so I checked the resistance and got a reading of about 4.5MΩ which I was told was the reading I would see at room temperature. 

David

closed 2022-10-17 by github-actions[bot]