MarlinFirmware/Marlin #24611: [BUG] Cr touch not compensating on skr mini v3

Smithy0822, 2022-08-09:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Hey, so I'm trying to get my cr touch to work with my ender 3 with a skr mini v3 board. The config that they supplied for auto leveling didn't work, so I compiled my own using teaching techs tutorial. It probes fine but when I try to print a larger object is doesn't compensate enough, so usually the left side is too high, mid is perfect, right is too low. I had the same problem with my 4.2.2 board. My start gcode:  ; Ender 3 Custom Start G-code

M190 S{material_bed_temperature} ;Heat bed to Cura and wait

M104 S150 ;Heat hotend, no wait

G92 E0 ; Reset Extruder

G28 ; Home all axes

G29 ; BLtouch bed level

G1 Z12 F3000 ;Set nozzle for heatup

G1 X0 Y0 ;Set nozzle to wipe

M109 S{material_print_temperature} ;Final hotend heat

G92 X-5.0 Y-10.0 ; Home offset for MicroSwiss

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position

G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line

G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little

G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

G92 E0 ; Reset Extruder

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
; End Of Custom Start G-code

### Bug Timeline

When I started using CR touch a few months ago

### Expected behavior

I expected it to fully compensate for the bed level

### Actual behavior

It didn't compensate enough

### Steps to Reproduce

1. Install Cr touch on an ender 3 pro with skr mini e3 v3
2.  Compile firmware for auto leveling
3. Print large object

### Version of Marlin Firmware

2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

Skr Mini E3 V3

### Add-ons

Creality glass bed

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/9294259/Configs.zip)
I tried testing marlin 2.1 but I vs code said the provided configs were outdated

Bob-the-Kuhn, 2022-08-10:
Auto leveling isn't perfect. 

Sometimes increasing the grid size (number of sample points) is all that is needed.

I like to use the G26 mesh validation pattern to see where the problems are.  I'm using the UBL leveling system because it has the G29 P2 phase where I can adjust individual mesh points as needed based on the G26 pattern.

thisiskeithb, 2022-08-10:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Smithy0822, 2022-08-10:
I’ll do the test tomorrow, but the the thing is I can’t use the 2.1 firmware because the provided config for 2.1 firmware is “for an old version of marlin” according to vscode

Bob-the-Kuhn, 2022-08-10:
I just downloaded the bugfix-2.1.x examples from [here.](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%203.0) It compiled with no issues.

thisiskeithb, 2022-08-10:
> I’ll do the test tomorrow, but the the thing is I can’t use the 2.1 firmware because the provided config for 2.1 firmware is “for an old version of marlin” according to vscode

That's because you're not using the [2.1.1 release configs](https://github.com/MarlinFirmware/Configurations/archive/refs/tags/2.1.1.zip) with the [2.1.1 release of Marlin](https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.1.1.zip).

To test, you'll need to download [bugfix-2.1.x configs](https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip) with the [Marlin bugfix-2.1.x](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip).

Smithy0822, 2022-08-11:
Pain… I tried using Google gcode sender, which worked but didn’t save the whole log. Tried cura, but it doesn’t even have an output log. Tried pronterface but upon enter M111 S247 if just echos m105 constantly.

Smithy0822, 2022-08-11:
Yeah ok this “might” be my problem

CRCinAU, 2022-08-12:
It's strange, but I've noticed this too - however I've been trying to rule out a hardware problem first... My bed probes as:

```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.120 -0.061 +0.009 +0.051 +0.109
Recv:  1 -0.136 -0.077 -0.005 +0.053 +0.074
Recv:  2 -0.110 -0.065 +0.001 +0.054 +0.091
Recv:  3 -0.159 -0.087 -0.005 +0.062 +0.114
Recv:  4 -0.100 -0.062 +0.009 +0.064 +0.111
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
```

Levelling should be able to handle those levels of difference.

CRCinAU, 2022-08-12:
So after a re-level, checking the X gantry, resetting the Z offset for the probe etc. This is the first part of a print.

[image]

Note that the outer circle is 1 line width - and is much more compressed on the left side of the print.

The inner circles are similar - much multiple line widths.

The grid used on this print:
```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.101 -0.041 +0.006 +0.062 +0.101
Recv:  1 -0.106 -0.079 -0.001 +0.048 +0.077
Recv:  2 -0.106 -0.056 +0.006 +0.066 +0.080
Recv:  3 -0.125 -0.086 -0.010 +0.061 +0.111
Recv:  4 -0.104 -0.059 +0.015 +0.071 +0.101
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok P15 B3
```

One interesting point to note - these are done with ARC commands. the g-code for the inner arcs (which is a brim) is:
```
;TYPE:Skirt/Brim
;WIDTH:0.42
G1 F1200
G3 X85.249 Y77.443 I33.749 J34.508 E0.14903
G1 X88.621 Y74.987 E.12424
G1 X91.526 Y73.203 E.10156
G3 X98.092 Y70.106 I24.183 J42.762 E0.21641
G1 X102.064 Y68.776 E.12478
G3 X107.204 Y67.595 I13.650 J47.633 E0.15714
G1 X110.976 Y67.093 E.11335
G1 X114.373 Y66.895 E.10136
G3 X121.622 Y67.269 I1.078 J49.538 E0.21640
G1 X125.64 Y67.951 E.12137
G3 X130.491 Y69.253 I-10.164 J47.555 E0.14966
G1 X134.4 Y70.706 E.12423
G3 X139.092 Y72.989 I-20.379 J47.846 E0.15547
G1 X142.662 Y75.18 E.12478
G3 X150.563 Y81.822 I-27.104 J40.262 E0.30799
G1 X153.333 Y84.965 E.12478
G3 X156.420 Y89.245 I-39.714 J31.898 E0.15727
G1 X158.498 Y92.814 E.12298
G3 X160.613 Y97.373 I-42.608 J22.536 E0.14977
G1 X161.965 Y101.194 E.12071
G3 X163.578 Y108.271 I-46.780 J14.384 E0.21640
G1 X163.967 Y111.67 E.10189
G1 X164.114 Y115.825 E.12385
G3 X162.949 Y126.094 I-48.654 J-0.319 E0.30839
G1 X161.861 Y130.14 E.12478
G3 X160.014 Y135.052 I-48.320 J-15.366 E0.15638
G1 X158.181 Y138.776 E.12363
G1 X156.623 Y141.436 E.09183
G3 X152.302 Y147.269 I-41.353 J-26.117 E0.21641
G1 X149.503 Y150.25 E.12179
G3 X145.751 Y153.557 I-33.749 J-34.508 E0.14902
G1 X142.379 Y156.012 E.12425
G1 X139.474 Y157.797 E.10156
G3 X132.908 Y160.894 I-24.183 J-42.762 E0.21640
G1 X128.936 Y162.224 E.12478
G3 X123.796 Y163.405 I-13.650 J-47.633 E0.15714
G1 X120.024 Y163.907 E.11334
G1 X116.627 Y164.105 E.10137
G3 X109.378 Y163.731 I-1.078 J-49.538 E0.21640
G1 X105.36 Y163.049 E.12137
G3 X100.509 Y161.747 I10.164 J-47.555 E0.14966
G1 X96.6 Y160.294 E.12423
G3 X91.908 Y158.011 I20.379 J-47.846 E0.15547
G1 X88.338 Y155.82 E.12478
G3 X80.422 Y149.162 I27.080 J-40.232 E0.30865
G1 X77.667 Y146.035 E.12412
G3 X74.579 Y141.755 I40.015 J-32.125 E0.15727
G1 X72.502 Y138.186 E.12298
G3 X70.387 Y133.627 I42.608 J-22.536 E0.14977
G1 X69.035 Y129.806 E.12071
G3 X67.422 Y122.729 I46.780 J-14.384 E0.21640
G1 X67.033 Y119.33 E.10189
G1 X66.886 Y115.175 E.12385
G3 X68.051 Y104.906 I48.622 J0.315 E0.30839
G1 X69.139 Y100.86 E.12478
G3 X70.986 Y95.948 I48.320 J15.366 E0.15637
G1 X72.819 Y92.224 E.12364
G3 X78.698 Y83.731 I42.618 J23.219 E0.30822
G1 X81.456 Y80.794 E.12
G1 X84.067 Y83.362 F9000
G1 F1200
G3 X82.835 Y84.615 I31.433 J32.138 E8.36051
G1 X84.025 Y83.405 E.05057
G1 X84.294 Y83.669 F9000
G1 F1200
G3 X80.783 Y87.538 I31.199 J31.839 E8.18656
G3 X84.252 Y83.712 I33.964 J27.309 E0.15390
G1 X84.521 Y83.976 F9000
G1 F1200
G3 X80.242 Y88.846 I30.974 J31.530 E8.07848
G3 X84.478 Y84.019 I34.733 J26.208 E0.19147
G1 X84.747 Y84.283 F9000
G1 F1200
G3 X116.273 Y71.683 I30.753 J31.207 E1.03841
M73 P0 R680
G3 X84.705 Y84.326 I-0.768 J43.816 E7.16101
G1 X84.974 Y84.59 F9000
G1 F1200
G3 X78.130 Y93.345 I30.525 J30.914 E7.79871
G3 X84.932 Y84.633 I36.998 J21.875 E0.33014
G1 X85.201 Y84.897 F9000
G1 F1200
G3 X78.453 Y93.539 I30.295 J30.611 E7.73258
G3 X85.158 Y84.940 I36.704 J21.706 E0.32569
G1 X85.427 Y85.205 F9000
G1 F1200
G3 X77.097 Y96.849 I30.062 J30.308 E7.56091
G3 X85.385 Y85.247 I38.279 J18.583 E0.42670
G1 X85.654 Y85.512 F9000
G1 F1200
G3 X134.863 Y153.125 I29.848 J29.990 E4.34731
M73 P0 R679
G3 X85.612 Y85.554 I-19.361 J-37.622 E3.56985
G1 X85.88 Y85.819 F9000
G1 F1200
G3 X77.012 Y98.849 I29.617 J29.690 E7.37601
G3 X85.838 Y85.862 I38.240 J16.496 E0.47053
G1 X86.107 Y86.126 F9000
G1 F1200
G3 X78.121 Y97.331 I29.381 J29.388 E7.36596
G3 X86.065 Y86.169 I37.190 J18.060 E0.40995
G1 X86.334 Y86.433 F9000
G1 F1200
G3 X77.706 Y99.146 I29.163 J29.077 E7.24675
G3 X86.292 Y86.476 I37.502 J16.170 E0.45857
G1 X86.56 Y86.74 F9000
G1 F1200
G3 X78.052 Y99.294 I28.936 J28.770 E7.18224
G3 X86.518 Y86.783 I37.199 J16.052 E0.45260
G1 X86.787 Y87.047 F9000
G1 F1200
G3 X120.676 Y75.403 I28.711 J28.424 E1.10584
M73 P0 R678
G3 X86.745 Y87.090 I-5.168 J40.096 E6.45845
G1 E-.35 F4800
;WIPE_START
G1 F7200
G1 X86.903 Y86.93 E-.1425
;WIPE_END
G1 E-.0075 F4800
G1 Z.5 F9000
G1 X76.43 Y115.138
G1 Z.2
G1 E.5 F2400
```

That makes me wonder - is it possible that the mesh isn't applied correctly to ARC print commands?

closed 2022-10-23 by github-actions[bot]