MarlinFirmware/Marlin #24612: BUG BTT SKR2 revB   reset

qulet, 2022-08-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin po ok 5 -15 min samoczynnie następuje  RESET

### Bug Timeline

Testowałem wcześniejsze wersje . Ten błąd dla mojej płyty się powtarza

### Expected behavior

 Praca stabilna berz resetu.

### Version of Marlin Firmware

Marlin 2.09.5 BIQU B!

### Printer model

Core XY DIY

### Electronics

BTT SKR 2 revB

### Add-ons

Karta SD

### Bed Leveling

ABL Linear grid

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9294902/Marlin.zip)

thisiskeithb, 2022-08-10:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23618

closed 2022-08-10 by thisiskeithb

qulet, 2022-08-11:
Dlaczego  temat został zamknięty ?
Jaki konflikt softwerowy powoduje ten RESET ?  
post #23618  dotyczył Marlina v 2.09.02. i nie został rozwiązany. Wersja 2.09.05   ma ten sam problem.