MarlinFirmware/Marlin #24613: [BUG] (Z-home does not work on the Makertech 3D Profoge 2s Dual)

Fhalo48, 2022-08-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When Z-home is active the printer move 10mm or so up, then stop. It will not move down toward the bed until the inductive sensor is activated.

But, Y and X homing are working fine.

The printer also works fine with the factory installed Marlin V1.1.x 

>  https://github.com/Makertech3D/Proforge-2S/tree/master/Printer%20Firmware

Attached are the factory configuration file, V2.1.x configuration file. Settings from working printer and settings from failed printer.
[workingDefaultConfig.zip](https://github.com/MarlinFirmware/Marlin/files/9298750/workingDefaultConfig.zip)

[FailedBugFix2xConfig.zip](https://github.com/MarlinFirmw
[WorkingProforge.zip
[PrinterSettingsV2-1-1.zip](https://github.com/MarlinFirmware/Marlin/files/9298770/PrinterSettingsV2-1-1.zip)
](https://github.com/MarlinFirmware/Marlin/files/9298768/WorkingProforge.zip)
are/Marlin/files/9298763/FailedBugFix2xConfig.zip)

### Bug Timeline

Started when upgrading the firmware from factory version to V2.1.x

### Expected behavior

The Z home should work regardless of the printer head position.

### Actual behavior

When I start the Z home. It move the print head up few mm then pause, it move the print head up few mm then stop,

### Steps to Reproduce

1. Power off the printer
2. Move the bed some distance from the probe/nozzle, maybe about 30mm
3. Turn on the printer, then select Auto Home

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Makertech 3D, Proforge 2s Dual

### Electronics

MKS_GEN_L

### Add-ons

No changes to stock hardware

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The Problem is repeated if I use the Pronterface, or the touch screen display on the printer.

ellensp, 2022-08-10:
This is symptomatic of the Z endstop already being triggered...

What does M119 say?

Fhalo48, 2022-08-10:
Give me few min. I get the info. Thanks for quick reply

Fhalo48, 2022-08-10:
I've re-installed the V2.1.1 release version. Ran M119 and it return the following:-
>>> M119
SENDING:M119

> Reporting endstop status
> x_max: open
> y_max: open
> z_min: TRIGGERED

ellensp, 2022-08-10:
and there is the issue...   z_min: TRIGGERED

So hold something under the inductive sensor...  and send a new M119.
does z_min change to open?
In which case change #define Z_MIN_ENDSTOP_INVERTING

If its not changing, it's something else

Fhalo48, 2022-08-10:
The LED on the inductive sensor turn red. But, M119 still shows trigger. Will try changing #define Z_MIN_ENDSTOP_INVERTING

ellensp, 2022-08-10:
wont work if it doesn't change...

Is the probe really plugged into to Z-Min endstop? 
Or is it plugged in somewhere else and they changed the pins file

Fhalo48, 2022-08-10:
Is the probe really plugged into to Z-Min endstop?
Yes, as it worked with the factory firmware.

I  could compare the pin file between the factory firmware & the V2.1.1

Changed #define Z_MIN_ENDSTOP_INVERTING to true. It failed to compiled:-

Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
=============================================================================== [FAILED] Took 16.02 seconds =============================================================================== 

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:16.020
========================================================================== 1 failed, 0 succeeded in 00:00:16.020 ========================================================================== 
PS C:\DATA\MerlinRelease> echo "done" >"C:\Users\ds\AppData\Local\Temp\ipc"
PS C:\DATA\MerlinRelease> platformio run --target upload -e mega2560
Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr @ 5.1.0
 - tool-avrdude @ 1.60300.200527 (6.3.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1805:8: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
        ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1

Fhalo48, 2022-08-10:
Attached pins used in the factory firmware:-
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Arduino Mega with RAMPS v1.4 (or v1.3) pin assignments
 *
 * Applies to the following boards:
 *
 *  RAMPS_14_EFB (Hotend, Fan, Bed)
 *  RAMPS_14_EEB (Hotend0, Hotend1, Bed)
 *  RAMPS_14_EFF (Hotend, Fan0, Fan1)
 *  RAMPS_14_EEF (Hotend0, Hotend1, Fan)
 *  RAMPS_14_SF  (Spindle, Controller Fan)
 *
 *  RAMPS_13_EFB (Hotend, Fan, Bed)
 *  RAMPS_13_EEB (Hotend0, Hotend1, Bed)
 *  RAMPS_13_EFF (Hotend, Fan0, Fan1)
 *  RAMPS_13_EEF (Hotend0, Hotend1, Fan)
 *  RAMPS_13_SF  (Spindle, Controller Fan)
 *
 *  Other pins_MYBOARD.h files may override these defaults
 *
 *  Differences between
 *  RAMPS_13 | RAMPS_14
 *         7 | 11
 */

#if !defined(__AVR_ATmega1280__) && !defined(__AVR_ATmega2560__)
  #error "Oops!  Make sure you have 'Arduino Mega' selected from the 'Tools -> Boards' menu."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME "RAMPS 1.4"
#endif

//
// Servos
//
#ifdef IS_RAMPS_13
  #define SERVO0_PIN        7   // RAMPS_13 // Will conflict with BTN_EN2 on LCD_I2C_VIKI
#else
  #define SERVO0_PIN       11
#endif
#define SERVO1_PIN          6
#define SERVO2_PIN          5
#ifndef SERVO3_PIN
  #define SERVO3_PIN        4
#endif

//
[... 479 lines cut ...]
        #define SHIFT_CLK       44
        #define SHIFT_LD        42
        #define BTN_EN1         64
        #define BTN_EN2         59
        #define BTN_ENC         63
      #elif ENABLED(PANEL_ONE)
        #define BTN_EN1         59   // AUX2 PIN 3
        #define BTN_EN2         63   // AUX2 PIN 4
        #define BTN_ENC         49   // AUX3 PIN 7
      #else
        #define BTN_EN1         37
        #define BTN_EN2         35
        #define BTN_ENC         31
      #endif

      #if ENABLED(G3D_PANEL)
        #define SD_DETECT_PIN   49
        #define KILL_PIN        41
      #endif

    #endif
  #endif // NEWPANEL

#endif // ULTRA_LCD
`

ellensp, 2022-08-10:
#define Z_MIN_PIN 63   <-----   this is not stock z_min pin

 stock for mks_gen_l is  #define Z_MIN_PIN  18

In your Configuration.h add the line
#define Z_MIN_PIN 63 
after   #define MOTHERBOARD BOARD_MKS_GEN_L

To make this change in the config

Fhalo48, 2022-08-10:
Thanks I'll do that.

ellensp, 2022-08-10:
Not a Marlin bug.

Fhalo48, 2022-08-10:
You are correct. Without your help I would not be able to resolve this. Thanks again. Proforge 2s Dual is now 2.1.x. I will need to run more tests.

Final output when Z is home:-
SENDING:M119
Reporting endstop status
x_max: open
y_max: open
z_min: open

Thanks again.

closed 2022-08-10 by Fhalo48

ellensp, 2022-08-10:
You should upload the configs to https://github.com/MarlinFirmware/Configurations when your sure its all working.
To help out the next person

Fhalo48, 2022-08-10:
I'll be happy to do that. I'll also give you credit foe helping me out.

Fhalo48, 2022-08-10:
Hi

I'm getting remote: Permission to MarlinFirmware/Configurations.git  denied. I do have github account. When I try to push the configuration file

ellensp, 2022-08-10:
you can only create PR's against the IMPORT branch of Configurations

ellensp, 2022-08-10:
If your really struggling, just create a new job in Configurations, with new configs to add, and someone will turn it into a pr

Fhalo48, 2022-08-10:
Thanks again. Files updaloded :)