MarlinFirmware/Marlin #24617: [System Variables] (Variables for parsing information within macros, used for probing and custom marco calls)

MasterofNull, 2022-08-11:
### Is your feature request related to a problem? Please describe.

Probing and custom macro calls need to parse/pass information to variables within macros. 

### Describe the feature you want

System variables.
Example:
#100
#101
#102
#200
#201
#202

Bob-the-Kuhn, 2022-08-12:
How extensive do you envision this support?  I'm concerned that you're looking for a macro language that has conditional branches.  They exist but at a higher level than Marlin.  I'm wondering if you're hoping for a simple python like language that can parse Marlin's responses and make decisions based on the responses.

MasterofNull, 2022-08-12:
I think just simple variable storage and let the slicer and G/M code handle
the conditional branching.

I am worried that the temp reading M code might read a temp close to the
actual desired temp. That would lead to compounding changes in temp if
called often.

"I'm wondering if you're hoping for a simple python like language that can
parse Marlin's responses and make decisions based on the responses."
I think that is what I am looking for.

[FR] Custom temperature macro #24614
https://github.com/MarlinFirmware/Marlin/issues/24614

This is the current application to this feature I am trying to implement.
Trying to sus out the best approach for this kind of feature.

Bob-the-Kuhn, 2022-08-13:
You're not the first to want this capability.  So far no traction.

What do you see as the value add for the average user?

Personally, I don't see this happening.  This would require a major shift in Marlin's philosophy plus supporting changes by the organizations that supply the slicers and host interfaces.  Non-trivial.

From a practical standpoint, I expect that the macro would be part of the gcode stream produced by the slicer which implies the entire gcode needs to be stored within the controller running Marlin which implies a huge increase in the available memory which means more cost.

I could see the new feature running as an application on a PC or a coprocessor.  There is one klipper oriented 3D controller that has a PC on a chip as a co-processor.

MasterofNull, 2022-08-13:
"You're not the first to want this capability. So far no traction."
You mean I'm not special?/s

"What do you see as the value add for the average user?"
Did you see the referenced temperature macro? Simplifies temp change over
and nozzle leakage during wait time(immediate). Allows for future logic
branching and decision making, off loaded to other devices.

"Personally, I don't see this happening."
I can understand that this flirts with the code logic that is not supported
by Marlin. But I really think that this simple function can unlock some
very power capabilities. Custom code strings and logic in slicers for
start, end, before z change, and after z change. They can be implemented
without the major immediate changes for slicer outputs.

"From a practical standpoint"
You lost me here

"I could see the new feature running as an application on a PC"
This is the more advanced capability that I see possible. Would need to be
coupled with some kind of print or output statement. Would love to see some
branch capability off loaded to USB, SD, PC, and co-proccessors.

MasterofNull, 2022-08-13:
For the referenced custom temperature macro. Replace [NEW_VAR1] = #101
[NEW_VAR2] = #102, ...ect.

MasterofNull, 2022-08-19:
Need code review for variable call function! Would anyone be interested in looking over my code to see if I am getting close to a working variable call? I am able to get it to compile without error. Just need someone to verify that it is doing what it should.
[Marlin-2.1.1(variables).zip](https://github.com/MarlinFirmware/Marlin/files/9378365/Marlin-2.1.1.variables.zip)

MasterofNull, 2022-08-19:
Here is a link to my Marlin 2.1.1 with Variable_Support added branch. https://github.com/MasterofNull/Marlin

thinkyhead, 2022-08-19:
> Here is a link to my Marlin 2.1.1 with Variable_Support added fork.

NOTE: You'll want to do your work in a branch not named `bugfix-2.1.x` so you have a pristine branch and can submit a (draft) PR with it.

MasterofNull, 2022-08-19:
Thanks. How about this? https://github.com/MasterofNull/Marlin/tree/variable-2.1.x Still trying to get GitHub figured out. Not super intuitive for me.

MasterofNull, 2022-08-19:
Submitted PR with Variable_Support functionality.

closed 2022-08-19 by MasterofNull