MarlinFirmware/Marlin #24621: [BUG] Build fails when SWITCHING_TOOLHEAD enabled but not TOOL_SENSOR

aamott, 2022-08-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Building Marlin fails when SWITCHING_TOOLHEAD is enabled but not TOOL_SENSOR. It compiles fine when SWITCHING_TOOLHEAD is commented. 

### Expected behavior

Firmware compiles when SWITCHING_TOOLHEAD is properly configured.

### Actual behavior

This error is thrown (clipping off the end of the build log):

```
...
Marlin\src\module\tool_change.cpp: In function 'void switching_toolhead_tool_change(uint8_t, bool)':
Marlin\src\module\tool_change.cpp:540:5: error: 'switching_toolhead_lock' was not declared in this scope
     switching_toolhead_lock(false);
     ^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\tool_change.cpp:540:5: note: suggested alternative: 'switching_toolhead_tool_change'
     switching_toolhead_lock(false);
     ^~~~~~~~~~~~~~~~~~~~~~~
     switching_toolhead_tool_change
Compiling .pio\build\at90usb1286_dfu\lib58a\SPI\SPI.cpp.o
*** [.pio\build\at90usb1286_dfu\src\src\module\tool_change.cpp.o] Error 1
```

### Steps to Reproduce

1. Configure 2 extruders, thermistors, heaters, etc. 
2. Enable `SWITCHING_TOOLHEAD` but *not* `TOOL_SENSOR`
3. Attempt to compile

### Version of Marlin Firmware

2.1.1

### Printer model

Custom

### Electronics

Printrboard Rev E

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configurations: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9313003/Configuration.zip)

I was able to patch it with 2 small changes, but it may affect the ability to use a tool sensor. 
My first change was in `tool_change.cpp`, moving this code from the `#if ENABLED(TOOL_SENSOR)` section to the `#if ENABLED(SWITCHING_TOOLHEAD)` section:

```cpp
  inline void switching_toolhead_lock(const bool locked)
  {
    #ifdef SWITCHING_TOOLHEAD_SERVO_ANGLES
        const uint16_t swt_angles[2] = SWITCHING_TOOLHEAD_SERVO_ANGLES;
        servo[SWITCHING_TOOLHEAD_SERVO_NR].move(swt_angles[locked ? 0 : 1]);
    #elif PIN_EXISTS(SWT_SOLENOID)
        OUT_WRITE(SWT_SOLENOID_PIN, locked);
        gcode.dwell(10);
    #else
    #error "No toolhead locking mechanism configured."
    #endif
  }
```

When I did, a new error was thrown: 

```
...
.pio\build\at90usb1286_dfu\src\src\MarlinCore.cpp.o: In function `setup':
MarlinCore.cpp:(.text.setup+0x1f4): undefined reference to `swt_init()'
collect2.exe: error: ld returned 1 exit status
```

`swt_init()` is defined in `tool_change.cpp` under the `#if ENABLED(TOOL_SENSOR)` section too. I fixed it by moving the definition of `swt_init` from the `#if ENABLED(TOOL_SENSOR)` section to the `#if ENABLED(SWITCHING_TOOLHEAD)` section.   
After that It compiled and is running on my printer now. Still, I can't confirm how this will affect systems using TOOL_SENSOR, particularly `MAGNETIC_PARKING_EXTRUDER` and 'ELECTROMAGNETIC_SWITCHING_TOOLHEAD', as they do not have these definitions.

DerAndere1, 2022-08-23:
I have the same patch applied. and saw the same solution in yet another branch. So even without testing, the proposal seems to be a reasonable fix.

thisiskeithb, 2022-09-29:
Closing since you’ve opened a PR.

- https://github.com/MarlinFirmware/Marlin/pull/24814

closed 2022-09-29 by thisiskeithb