MarlinFirmware/Marlin #24625: [BUG] Some arguments in G-code are parsed as Hexadecimal values

mriscoc, 2022-08-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sending the G-code `G0Y0X10' results as Y target position parsed as 16 and X as 10.
G0Y0X20 is parsed as Y=32 and X=20.

### Expected behavior

If a hexadecimal value is intended to be parsed, then the X axis should be kept unchanged, if the hexadecimal value is not to be parsed, then Y and X should keep only the decimal values.

### Actual behavior

Hexadecimal value is parsed if a figure 0Xnn is found in the argument.

### Steps to Reproduce

Send to printer the G-code `G0Y0X10' the printer moves to Y=16 X = 10

### Version of Marlin Firmware

Latest bugfix 2.1.x

### Bed Leveling

No Bed Leveling

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mriscoc, 2022-08-14:
For G0Y0X20:
https://github.com/MarlinFirmware/Marlin/blob/4bd4c1f3bc00056da4fe008de9aeda8424422d3f/Marlin/src/gcode/parser.h#L257-L275
**value_ptr** contains 0X20 when Y value is parsed and 20 for X.

mriscoc, 2022-08-14:
If hexadecimal values are forbidding, this could be fixed changing:
```cpp
        if (c == 'E' || c == 'e') {
```
to
```cpp
        if (c == 'E' || c == 'e' || c == 'X' || c == 'x') {
```

Bob-the-Kuhn, 2022-08-14:
Confirmed.  

G0Y0X10 and G0 Y0X10 both result in X=10, Y=16
G0 Y0 X10 behaves as expected.

thinkyhead, 2022-08-14:
That's a funny one. See the way we deal with "e" within a number for the solution. I'll post the patch soon.

mriscoc, 2022-08-14:
Is there some way to disable the hexadecimal parsing? maybe that could save some program memory bytes.

ellensp, 2022-08-15:
no PR or anything, how is this completed?

mriscoc, 2022-08-15:
> no PR or anything, how is this completed?

I can submit a PR, with [this](https://github.com/MarlinFirmware/Marlin/issues/24625#issuecomment-1214274003) solution, but it is more as a patch/workaround, I think that could be better remove the support for hexadecimal in the parser with the hope of gain some bytes in program memory.

thinkyhead, 2022-08-16:
> Is there some way to disable the hexadecimal parsing? maybe that could save some program memory bytes.

In my first cursory pass at the parser, all the string-to-integer functions are specifying base-10. But maybe the string-to-float functions are falling back to integer functions without the base-10 specifier. So, I need to run some tests in the debugger and see which functions are acting up. If it's the float-fallback thing then we'll need to use the same hack as for E-exponent.

descipher, 2022-08-16:
If MARLIN_DEV_MODE is defined then yes it will parse as HEX after x.
undef MARLIN_DEV_MODE and it will not parse any HEX.
But I would ask, what use case requires HEX support?

mriscoc, 2022-08-16:
MARLIN_DEV_MODE was disabled in the tests.

descipher, 2022-08-16:
Confirmed with SIM. 
Only when we place Y first in the gcode string.
G0X0Y10 is parsed as X=0, Y=10
G0Y0X10 is parsed as Y=10, X=16

descipher, 2022-08-16:
Maybe we should adjust the bug description to reflect the order issue.

descipher, 2022-08-16:
https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374

Based the RS-274 and subsequent standards, HEX numbers are not permitted.
I vote to remove the possibility of them and only conform to the correct supported industry standard gcode formats.

https://fabricesalvaire.github.io/pythonic-gcode-machine/gcode-reference/rs-274/index.html

```
7.1.4.3. Number
The following rules are used for (explicit) numbers. In these rules a digit is a single character between 0 and 9.

A number consists of (1) an optional plus or minus sign, followed by (2) zero to many digits, followed, possibly, by (3) one decimal point, followed by (4) zero to many digits — provided that there is at least one digit somewhere in the number.
There are two kinds of numbers: integers and decimals. An integer does not have a decimal point in it; a decimal does.
Numbers may have any number of digits, subject to the limitation on line length.
A non-zero number with no sign as the first character is assumed to be positive. Notice that initial (before the decimal point and the first non-zero digit) and trailing (after the decimal point and the last non-zero digit) zeros are allowed but not required. A number written with initial or trailing zeros will have the same value when it is read as if the extra zeros were not there.
```
There is no constraint to order in any of the observed standards docs.

descipher, 2022-08-16:
> If hexadecimal values are forbidding, this could be fixed changing line 264 from:
> 
> ```c++
>         if (c == 'E' || c == 'e') {
> ```
> 
> to
> 
> ```c++
>         if (c == 'E' || c == 'e' || c == 'X' || c == 'x') {
> ```

+1 to this solution.

ellensp, 2022-08-26:
fixed with https://github.com/MarlinFirmware/Marlin/commit/af1c7e1a81f969a60c24be71b6a35b2d010166ad

closed 2022-08-26 by ellensp