MarlinFirmware/Marlin #24632: Build issue with TFT35 LGVL_UI with DELTA AUTO CALIBRATION

Orafilynie, 2022-08-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Error Details :

*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1455:6: error: #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
 1455 |     #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
      |      ^~~~~
 1766 |   #error "MESH_EDIT_GFX_OVERLAY requires AUTO_BED_LEVELING_UBL and a Graphical LCD."
      |    ^~~~~
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32\MarlinSerial.cpp.o] Error &

Seems to be LGVL not compatible with delta calibration and i dont know why and also GFX_OVERLAY error

Also i dont know the type of levelling of my printer as the FLSUN Q5 use a probe

### Bug Timeline

Surely old but didnt helped me

### Expected behavior

Build successful

### Actual behavior

Nope.
Marlin hates me and wont build.

See error in desc.

### Steps to Reproduce

SET to LGVL_UI and use FLSUN Q5 config files

### Version of Marlin Firmware

X

### Printer model

FLSUN Q5

### Electronics

X

### Add-ons

X

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://www.mediafire.com/file/lpmqtgg2i5bu7o2/Marlin.rar/file

ellensp, 2022-08-15:
Provide configuration_adv.h is version 02000902 while Configuration.h is version 02000905

This will never work.

Orafilynie, 2022-08-15:
ok let me retry for the 17th time

Orafilynie, 2022-08-15:
Now the autobuild even tell me its not a valid workspace ._.

ellensp, 2022-08-15:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-08-15 by ellensp

Orafilynie, 2022-08-15:
was very helpful i reccomand -_-

ellensp, 2022-08-15:
DELTA_CALIBRATION_MENU does not work on your User interface TFT_LVGL_UI
MESH_EDIT_GFX_OVERLAY does not work on your User interface TFT_LVGL_UI

TFT_CLASSIC_UI is the only UI that supports DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY on your LCD display

Orafilynie, 2022-08-15:
Ah ok, so is the tft24 works on classic ?