MarlinFirmware/Marlin #24633: [FR] Make TFT_LVGL_UI compatible with DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY if technically possible :)

Orafilynie, 2022-08-15:
### Is your feature request related to a problem? Please describe.

Related to my last issue, only the CLASSIC_UI is compatible with DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY and that's a shame that i can't use my TFT color screen to its full potential.

### Describe the feature you want

Make TFT_LVGL_UI compatible with DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY if possible so i can use the full potential of my TFT Screen

jmz52, 2022-09-22:
> only the CLASSIC_UI is compatible with DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY

MESH_EDIT_GFX_OVERLAY is indeed to feature limited to 12864 UI, however the UBL functionality can be used without it.
COLOR_UI with automatic delta calibration and UBL works just fine on my FLSUN Q5

[image]

Orafilynie, 2022-09-22:
Yeah i saw that it could buy on color so i will reactivate ubl and ill test when i receive my tmc and flsun q5 printer tommorow

Orafilynie, 2022-09-22:
Thanks for letting me know

Orafilynie, 2022-09-22:
Solved

closed 2022-09-22 by Orafilynie

Orafilynie, 2022-09-24:
> > only the CLASSIC_UI is compatible with DELTA_CALIBRATION_MENU and MESH_EDIT_GFX_OVERLAY
> 
> MESH_EDIT_GFX_OVERLAY is indeed to feature limited to 12864 UI, however the UBL functionality can be used without it. COLOR_UI with automatic delta calibration and UBL works just fine on my FLSUN Q5
> 
> [image]

also how did you flash your flsun q5 as mine stuck at updating 100%