MarlinFirmware/Marlin #24638: [BUG] Bed and extruder temperature reports the same temperature

kelroy1990, 2022-08-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, we are working with the BOARD_BRICOLEMON_LITE_V1_0, with have the SAMD51 IC from microchip.

On the last versions (>2.1) of marlin, everything was working well. Right now i am testing the release and the bugfix of Marlin 2.1.X, but i see that we have a problem with the temperature reports, the extruder and the bed are reporting the same temperature, and the printer stops to work properly (wich its totally normal as the halted prevention flag start because the temperature of the bed doesnt change).

Both temperatures report the Extruder temperatur, so if you modify the extruder temperature, both start to change.

Test steps:
- Tested with Marlin GCODE M105
--> Tested with and without temperature thermistor (to see the variations)

Regards, Carlos.

### Expected behavior

Extruder and Bed temperature should be independent one from another.  

### Actual behavior

Both temperatures report the extruder temperature, working on Marlin < 2.1.X

### Steps to Reproduce

1.- Donwload Marlin 2.1 bugfix from github.
2.- Configure a SAMD51 board, on this case the BOARD_BRICOLEMON_LITE_V1_0
3.- Configure the bed temperature from 0 to 1.
4.- Configure the first USB Serial port to -1 instead of 0.
5.- Uploads the firmware to the board.
6.- Connect pronterface to the board with the USB cable.
7.- Start to work with the GCODE M105
8.- Try to heat the Extruder/Bed independent from another
---> See that both temperatures report extruder temperature.

### Version of Marlin Firmware

Marlin Bugfix 2.1 at 17/08/2022 with the CR10stock display

### Printer model

Ender 3

### Electronics

Bricolemon Lite (SAMD21 with TMC2209 steppers)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/9358902/Marlin_config.zip)
[image]

Bob-the-Kuhn, 2022-08-17:
Long shot - check the file **pins_BRICOLEMON_LITE_V1_0.h** and verify that the temperature inputs are assigned differently.

KelPolux, 2022-10-21:
I am checking the issue, and the behaivour is still be there. The temperature report its the same for the bed and the extruder.

kelroy1990, 2022-10-21:
I am checking the issue, and the behaivour is still be there. The temperature report its the same for the bed and the extruder.

closed 2023-01-01 by github-actions[bot]

thinkyhead, 2026-11-15:
PR #28857 is targeted to fix this issue. Let's hope that's the solution!