MarlinFirmware/Marlin #24642: [BUG] Servo timing with ACTUONIX PQ12R (works with any servo but not with this)

studiodyne, 2022-08-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

2 x PQ12R actuators ,  with RUMBA32 stm32f4 board
The actuator works at startup , but when I ‘home ‘ the machine , never works again
But classic servos , micro servos and any servos works fine
A timing problem , I have set  full priority to timer but always the same problem (may be i believe i have set the priorities, not really sure )
These actuators need precise timing,  and it works on Lulzbot TAZ PRO 'not the same cpu'

Please HELP

### Bug Timeline

old

### Expected behavior

simply works , lowering and raising servos

### Actual behavior

Works before homing and motor enabled

### Steps to Reproduce

T0 / T1 for toolchange

### Version of Marlin Firmware

2.1

### Printer model

CUSTOM

### Electronics

RUMBA32

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

IdeaMaker

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Nothin
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9375056/Marlin.zip)
g

Bob-the-Kuhn, 2022-08-18:
Have you tried using M280 to find the sweet spots for your servos on your board?

I don't see where you're specifying the servo angles.

studiodyne, 2022-08-18:
SWITCHING_NOZZLE toolhead

studiodyne, 2022-08-19:
@Bob-the-Kuhn , I know who you are , and I know you can help me
I repeat , all servos works ' analog , mini , micros and more ' , but the digital one ' Actuonix PQ12r' , works only after boot , and when I make the homing , never move , but If i plug any servos , the pwm works , the servos works too , but not the actuonix because need a precise pulse OR NOTHING

On the TAZ PRO ' lulzbot' , based on ARCHIM2 'atsam cpu ' , it works fine , and 'deactive after move ' is not defined in config.h , if not defined it's because of perfect pulse ANYTIME
I'm really tired of STM32 ,  I want to use NEOPIXEL , but not working cause ' not precise timing ' , I use digital servos , I have the same issue
Do I have to buy an ARCHIM board ' arduino due ' to make it functionnal ? Really tired of timming issue !

studiodyne, 2022-08-19:
May be , you know how to freeze all computation when servos moving , to have absolute timing ?

studiodyne, 2022-08-21:
SOLVED !
0.1µf capacitor between signal line and ground
After seeing signal with scope ,  the switching power supply make noise when motors on load
+ 1 meter cables ...

All is fine
Thks for help

closed 2022-08-21 by studiodyne