MarlinFirmware/Marlin #24645: [BUG] Marlin crashes/CPU reset if G34 is executed

Hiabuto-de, 2022-08-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin crashes if G34 is executed with 3-Z-Axis and auto-alignment. G28 and G29 work fine. 
Used Tag 2.1.1 from branch 2.1.x and also tried bugfix-2.1.x
Board: Rumba (AtMega2560)

### Bug Timeline

new Issue

### Expected behavior

BedLeveling with 3 axis should work or at least some Error-Message if anything is wrong

### Actual behavior

Marlin/CPU crashes and resets itself.

### Version of Marlin Firmware

2.1.1

### Printer model

HevORT

### Electronics

Rumba-oard

### Add-ons

BLTouch 3.0, TMC1560-Driver (SPI), GLCD-Display (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_G34_Issue.zip](https://github.com/MarlinFirmware/Marlin/files/9376481/Marlin_G34_Issue.zip)

ellensp, 2022-08-19:
You have a lot of stuff enabled for a a mega2560 based controller.

try disabling #define USE_WATCHDOG, see if it stops rebooting.

Hiabuto-de, 2022-08-19:
No, same Result - it instantly resets if I send G34, even with WDT disabled.

Hiabuto-de, 2022-08-26:
I stripped down my Config to a minimum as attached and try to run G34 again. This happens: First I do a G28 which works fine and Z moves 10mm down after Probing as it should. Then I start G34. Extruder moves to front left (near x=0, y=0), deployes BLTouch and probes (Z goes Up to Z=0). Then, Z stays at this Level and  Extruder moves to the Center and tries to deploy BLTouch, which fails, because the Z axis is still at Z=0. - used bugfix-2.1.x, comit 766ee15a38626807c73bbc9960eab3115d3c7c3
[Config_Stripped.zip](https://github.com/MarlinFirmware/Marlin/files/9431323/Config_Stripped.zip)

Hiabuto-de, 2022-08-26:
with Release 2.0.9.5 and all enabled Features as in my first Comment Leveling works fine!

sjasonsmith, 2022-09-29:
I tried to reproduce this in the simulator, but it seemed ok. I tried to get as close as I could to your config, but can't currently simulate the BLTouch, an AVR processor, or actual discrepencies in probe heights.

I used the latest state of bugfix-2.1.x, at commit a5c5bb96601c263087fd1223df5a7607e117aa78, then re-tested with the commit you mentioned with the same results. Could you verify that you still have problems with the latest state of bugfix-2.1.x?

If it still fails with the latest state of `bugfix-2.1.x`, can you enable bed level debug output and post the log here? You already have the feature enabled, but you need to enable the prints before running your G34.

``` C++
/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
#define DEBUG_LEVELING_FEATURE
```

I have somewhat inconvenient access to a AVR-based 3-Z printer, which I can use to debug if needed. It has a SOLENOID_PROBE, which may be an issue if the problem is BLTOUCH-specific.

closed 2022-12-09 by github-actions[bot]