MarlinFirmware/Marlin #24646: [BUG] FYSETC_CHEETAH_V20 compilation failed "#error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN.""

J0hnMatrix, 2022-08-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Starting from Marlin-2.1.1, compilation failed if fans pins are declared in the configuration_adv.h

`Compiling .pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_flash.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
Compiling .pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_if_iic.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\HAL.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_if_iic.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2199:6: error: #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 2199 |     #error "You cannot set E1_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_flash.cpp.o] Error 1
*** [.pio\build\FYSETC_CHEETAH_V20\src\src\HAL\STM32\eeprom_if_iic.cpp.o] Error 1`

[image]
[image]

I see that some pins are declared differently in the 'pins_FYSETC_CHEETAH_V20.h' file and I don't know if I can safely use the file from 2.1 (that compile successfully) into 2.1.1

Please find attached 'configuration.h' and 'configuration_adv.h' from Marlin 2.1 and 2.1.1
[Marlin-2.1.1_cheeta
[Marlin-2.1_cheetah_v2.zip](https://github.com/MarlinFirmware/Marlin/files/9379922/Marlin-2.1_cheetah_v2.zip)
h_v2.zip](https://github.com/MarlinFirmware/Marlin/files/9379919/Marlin-2.1.1_cheetah_v2.zip)

### Bug Timeline

New bug

### Expected behavior

Compile success

### Actual behavior

Compile failed

### Version of Marlin Firmware

2.1.1

### Printer model

Creality Ender 3

### Electronics

FYSETC Cheetah 2.0 board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-08-19:
This was patched in bugfix 13 days ago..  https://github.com/MarlinFirmware/Marlin/pull/24593

ellensp, 2022-08-19:
nope its not it...

same sort of issue though...

ellensp, 2022-08-19:
The above PR fixes it 
https://github.com/MarlinFirmware/Marlin/pull/24648/files

closed 2022-08-19 by ellensp