MarlinFirmware/Marlin #24653: [BUG] Compiled firmware results in blank screen when compiling on Linux (Ubuntu 22.04)

brian200508, 2022-08-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Prerequisites:

- OS: Linux Ubuntu 22.04
- Python 3.10.4
- pip 22.0.2
- PlatformIO Core, version 6.1.4 - installed via 'pip install -U platformio' (and 99 udev rules too)
- VS Code 1.70.2
- PlatformIO IDE v2.5.2
- git 2.34.1

Reproduce the issue:

- cd ~/Develop
- git clone https://github.com/MarlinFirmware/Marlin.git
- cd Marlin
- git status -> On branch bugfix-2.1.x, Your branch is up to date with 'origin/bugfix-2.1.x'.
- modified Configuration.h(as attached): custom steps, RAMPS 1.6, RepRapDiscount FULL GRAPHIC Smart Controller, SD card support
- pio run --target clean (did same on VS Code too)
- pio run (press Build in VS Code)
- upload ~/Develop/Marlin/.pio/build/mega2560/firmware.hex -> OK, but blank display
- power off/on printer -> still the same blank display

If I do the same steps on Windows 11 everything is fine. I can see the Marlin welcome screen and seconds later the main info screen.

What's wrong? Please help.
Thanks in advance...
Brian
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9382597/Configuration.zip)

### Bug Timeline

dectected since autumn 2021 - but did not compile on linux before

### Expected behavior

I can see the Marlin welcome screen and seconds later the main info screen.

### Actual behavior

I see a blank screen and nothing happens - sometimes it beeps.

### Steps to Reproduce

1. cd ~/Develop
2. git clone https://github.com/MarlinFirmware/Marlin.git
3. cd Marlin
4. git status -> On branch bugfix-2.1.x, Your branch is up to date with 'origin/bugfix-2.1.x'.
5. modified Configuration.h(as attached): custom steps, RAMPS 1.6, RepRapDiscount FULL GRAPHIC Smart Controller, SD card support
6. pio run --target clean (did same on VS Code too)
7. pio run (press Build in VS Code)
8. upload ~/Develop/Marlin/.pio/build/mega2560/firmware.hex -> OK, but blank display
9. power off/on printer -> still the same blank display

### Version of Marlin Firmware

branch bugfix-2.1.x checked out today (2022, August 19th)

### Printer model

self made

### Electronics

Arduino mega2560 with RAMPS 1.6 and RepRapDiscount FULL GRAPHIC Smart Controller

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

brian200508, 2022-08-19:
Linux is INTEL x64 based architecture.

EvilGremlin, 2022-08-19:
On fresh OS install? Did you try to completely purge and reinstall platformio at least?

brian200508, 2022-08-19:
yes, I have installed PlatformIO for first time on this PC (and Linux installation) today. All the other attempts in there past were on ubuntu 20.04.

EvilGremlin, 2022-08-19:
No issues at in Mint for me, except pio suicide update sometimes. You're probably the first one.

ellensp, 2022-08-19:
No issues on  Ubuntu 22.04.1 LTS for me
Used command line like  you instructed. not vscode...
uploaded with pio run -t upload

processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 96
model name	: AMD Ryzen 7 4800H with Radeon Graphics

ellensp, 2022-08-19:
Note. I do most of my dev work on a ramps + REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER on Ubuntu.
Was Ubuntu 20.04.4 LTS till a week go, but now Ubuntu 22.04 LTS. 
I have not seen this issue.

brian200508, 2022-08-19:
Is there anything I have to keep in mind regarding the installation? Or do I have to edit the platformio.ini file in some waxy?

brian200508, 2022-08-19:
[platformio.ini.txt](https://github.com/MarlinFirmware/Marlin/files/9382981/platformio.ini.txt)

brian200508, 2022-08-19:
just by the way: i have the same effect when compiling 1.1.x with Arduino IDE 1.8.x:

- firmware compiled on Windows: works
- firmware compiled on Linux: blank screen

Is there some difference concerning the pin out?

EvilGremlin, 2022-08-19:
Nope, there's no differnce between OS at all.

brian200508, 2022-08-19:
Is there a recommended way to setup a Marlin build environment on Linux? E.g. installing PlatformIO via pip or better using the script or just via VS Code. Install as root or as user, etc.

EvilGremlin, 2022-08-19:
However you want - it compiles exactly same.

ellensp, 2022-08-19:
a Test

Edit Marlin/src/gcode/host/M115.cpp
find `    "FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION " (" __DATE__ " " __TIME__ ") "`
replace with
`    "FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION`
 
now the code will not change between builds

Build on both machines and you should get identical firmware.hex files.

brian200508, 2022-08-19:
Thanks, I will try it...

brian200508, 2022-08-19:
In my case it produces different output; see attached files
[build_MarlinFirmware.zip](https://github.com/MarlinFirmware/Marlin/files/9384453/build_MarlinFirmware.zip)
.
The first indicator is the commandline output:
108604 bytes vs. 108600 bytes 

Linux:
Linking .pio/build/mega2560/firmware.elf
Checking size .pio/build/mega2560/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  53.9% (used 4417 bytes from 8192 bytes)
Flash: [====      ]  42.8% (used 108604 bytes from 253952 bytes)
Building .pio/build/mega2560/firmware.hex
========================= [SUCCESS] Took 68.37 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:01:08.368
========================= 1 succeeded in 00:01:08.368 =========================

Windows:
Linking .pio\build\mega2560\firmware.elf
Checking size .pio\build\mega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  53.9% (used 4417 bytes from 8192 bytes)
Flash: [====      ]  42.8% (used 108600 bytes from 253952 bytes)
Building .pio\build\mega2560\firmware.hex
============================================= [SUCCESS] Took 64.84 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:01:04.835
============================================= 1 succeeded in 00:01:04.835 =============================================
[build_MarlinFirmware.zip](https://github.com/MarlinFirmware/Marlin/files/9384488/build_MarlinFirmware.zip)

ellensp, 2022-08-20:
There are differences in .hex generated from windows and linux... very odd.  But also a platformio issue not marlin.

Since you provided a firmware_linux.hex I flashed this to my controller.

It works as expected.

closed 2022-08-20 by thisiskeithb

brian200508, 2022-08-20:
Thanks for Your investigations. It's essential for me to know, that the compiled binary itself works. I will check the USB driver and the Board. 

Best regards 
Brian