MarlinFirmware/Marlin #24655: [BUG] PID tune for extruder heater works perfectly but fails when setting temp manually

ansonl, 2022-08-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I upgraded my Ultimaker Original/2+ to have two extruder with the [community DXU modification](https://community.ultimaker.com/topic/24553-dxu-efficient-dual-extrusion-upgrade-for-um2/) and heating the second extruder heater does not work correctly. Temp sensor is PT100.

When I set a temperature `M104 T1 S210` the second heater heats up to about 20 degrees under the target temperature before stopping heating. I have measured the power when the extruder temperature starts falling and it is no longer using any energy and is basically cooling normally. 

When I autotune temperature `M303 E1 C8 S210` the second heater works perfectly. It is able to maintain the temperature well. See below. Each time I set the temperature manually, it stops heating 20 degrees under the target temp and eventually halts with a heater failed error. I have set the PID values to what is found by the PID autotune.

[image]

I have swapped my two extruder heater wires and this issue only occurs when using the second heater and thermometer ports on the Ultimaker Controller v2.1.4. The extruder heater wiring works fine with another heater port and the second heater port only works with PID autotune.  

I have not been able to figure out why the PID autotune works and setting a heating temperature as normal fails. Is there some difference in the heating routine between the **M303** PID autotune and **M104** that might be causing this? Thanks. 

I have attached my configuration and schematics for the ultimaker controller board v2.1.4. End of the terminal log with extra logging available also attached. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9385475/Configuration.zip)
[ultimainboard-v2-1-4.zip](https://github.com/MarlinFirmware/Marlin/files/9385472/ultimainboard-v2-1-4.zip)
[gcode.log](https://github.com/MarlinFirmware/Marlin/files/9385532/gcode.log)

### Bug Timeline

New bug, unknown when it started

### Expected behavior

I expect the heater to succeed in reaching the target temperature in both PID autotune and manual heating. 

### Actual behavior

PID autotune works perfect, manual heating always fails

### Steps to Reproduce

Attach heater and PT100 thermometer in heating block to Ultimaker board v2.1.4 ports: **Heater2** and **Temp2**

Run PID autotune to temperature such as 210. `M303 E1 C8 S210`
Set extruder temp to 210. `M104 T1 S210` Temperature will drop after reaching within 20 degrees of target temp. Heating failed error. 

### Version of Marlin Firmware

2.1.1 bugfix

### Printer model

Ultimaker 2+

### Electronics

Stock ultimainboard v2.1.4

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-08-19:
Did you try enabling **PID_PARAMS_PER_HOTEND**?

ansonl, 2022-08-19:
@Bob-the-Kuhn Yes, I just tried it out and the same problem occurs.

ansonl, 2022-08-19:
@Bob-the-Kuhn After a few tries, heating with the second heater port works! Thanks for the tip.

I just duplicated the first PID values in the Configuration to the second PID values. Shouldn't the printer just use the same PID values for both heaters when using a single PID in the Configuration or am I missing where multiple heaters explicitly need their own PID value in the config?

Bob-the-Kuhn, 2022-08-19:
You may also want to increase **WATCH_TEMP_PERIOD** if you get any further heating failed messages.

closed 2022-08-20 by ansonl