MarlinFirmware/Marlin #24657: [BUG] Moving Z axis moves extruder

JuliusSenkus, 2022-08-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While moving Z axis extruder moves together and while moving extruder no stepper moves.

Just a guess but since instead of E0 an E1 (in LCD) is shown (in LCD) and movement of driver 2 and 3 seems to be in sync, it seems that firmware thinks that extruder is another Z motor.

Tried on master, bug fix and just 2.1 branches with `mks_monster8_usb_flash_drive` and `mks_monster8` environments ([full config](https://drive.google.com/file/d/1hGirHUtGG9zkeZIjN3YAsm7qzAzBjwUT/view?usp=sharing)).

<details>
  <summary>Configuration diff</summary>
  <p>
  
  
```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index b30ba9b8d3..2592987a2f 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -87,7 +87,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_MKS_MONSTER8_V1
 #endif
 
 /**
@@ -98,7 +98,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
@@ -135,7 +135,7 @@
 //#define BLUETOOTH
 
 // Name displayed in the LCD "Ready" message and Info menu
-//#define CUSTOM_MACHINE_NAME "3D Printer"
+#define CUSTOM_MACHINE_NAME "Cetus printer"
 
 // Printer's unique ID, used by some programs to differentiate between machines.
 // Choose your own or use a service like https://www.uuidgenerator.net/version4
@@ -158,9 +158,9 @@
  *          TMC5130, TMC5130_STANDALONE, TMC5160, TMC5160_STANDALONE
  * :['A4988', 'A5984', 'DRV8825', 'LV8729', 'TB6560', 'TB6600', 'TMC2100', 'TMC2130', 'TMC2130_STANDALONE', 'TMC2160', 'TMC2160_STANDALONE', 'TMC2208', 'TMC2208_STANDALONE', 'TMC2209', 'TMC2209_STANDALONE', 'TMC26X', 'TMC26X_STANDALONE', 'TMC2660', 'TMC2660_STANDALONE', 'TMC5130', 'TMC5130_STANDALONE', 'TMC5160', 'TMC5160_STANDALONE']
  */
-#define X_DRIVER_TYPE  A4988
-#define Y_DRIVER_TYPE  A4988
-#define Z_DRIVER_TYPE  A4988
+#define X_DRIVER_TYPE  TMC2130
+#define Y_DRIVER_TYPE  TMC2130
+#define Z_DRIVER_TYPE  TMC2208
 //#define X2_DRIVER_TYPE A4988
 //#define Y2_DRIVER_TYPE A4988
 //#define Z2_DRIVER_TYPE A4988
@@ -172,7 +172,7 @@
 //#define U_DRIVER_TYPE  A4988
 //#define V_DRIVER_TYPE  A4988
 //#define W_DRIVER_TYPE  A4988
-#define E0_DRIVER_TYPE A4988
+#define E0_DRIVER_TYPE TMC2208
 //#define E1_DRIVER_TYPE A4988
 //#define E2_DRIVER_TYPE A4988
[... 352 lines cut ...]
Cetus

### Electronics

MKS monster8 v1

### Add-ons

TMC2130 (SPI)/TMC2225 (UART)

### Bed Leveling

ABL 3-point

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-08-20:
Please confirm the steppers are connected to driver0, driver1, driver2-1 and driver 3 & that driver 2-2 is not connected.

JuliusSenkus, 2022-08-20:
TMC connection error is reported and extruder does not move, even when command was given to move it via LCD (jumpers are set for UART communication).

JuliusSenkus, 2022-08-20:
Note: TMC error is not reported while driver is in driver2-2 spot instead of driver3 spot.

Bob-the-Kuhn, 2022-08-20:
The Monster8 silkscreen is confusing.  Driver0, driver1, driver2-1, driver2-2, driver3, driver4, driver5, driver6 and driver7 refer to the 9 stepper connectors.  The 8 stepper driver slots are not labelled.

There is no stepper driver slot called driver2-2.  There is a stepper connector called driver2-2.

Stepper connectors driver2-1 and driver2-2 are electrically tied to the third stepper driver slot.  First slot is for X (driver0), second slot is for y (driver1), third slot is for Z (driver2-1 and driver2-2) and the fourth slot is for E0 (driver3).

JuliusSenkus, 2022-08-21:
So it seems to be an issue between keyboard and chair.

Thanks, resolving it.

TL;DR There are 8 slots for drivers on MKD monster8 and there are 9 connectors for steppers, one connection is duplicated (driver2-1 to driver2-2), for driver3(E0) the fifth connection should be used.

closed 2022-08-21 by JuliusSenkus