MarlinFirmware/Marlin #24659: [BUG] Bed leveling does not account for HOTEND_OFFSET when hotend with HOTEND_OFFSET offset active

ansonl, 2022-08-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bed leveling does not take into account the active hotend offset set in HOTEND_OFFSET. Bed leveling with a second offset hotend is useful to determine the correct HOTEND_OFFSET in the Z direction for switching extruder designs where the extruders are lifted and lowered. I use `#define MESH_BED_LEVELING` to perform manual leveling to determine the correct height offset for the second extruder tool since the adjusted Z height will show on the screen.

If the active hotend has an offset of ex: 19mm in X axis, the bed leveling will run past the physical minimum X endstop and hit the end of the print axes and grind. When an active hotend has an offset of 19mm in X axis, 0mm position in real life is treated as 19mm position to accommodate the offset active hotend. 

### Bug Timeline

New bug, unknown when it started

### Expected behavior

Bed leveling takes the active hotend offset into account and does not move the printhead past the physical bounds of printer. 

### Actual behavior

Bed leveling moves printhead past physical minimum endspot and hits side of printer. 

### Steps to Reproduce

1. Set HOTEND_OFFSETs to something like below where second tool is offset in X/Y direction:
```
#define HOTEND_OFFSET_X { 0.0, 19.00 } // (mm) relative X-offset for each nozzle
//#define HOTEND_OFFSET_Y { 0.0, 0.00 }  // (mm) relative Y-offset for each nozzle
#define HOTEND_OFFSET_Z { 0.0, 0.5 }  // (mm) relative Z-offset for each nozzle
```
2. Switch to second tool with T1
3. Use Bed leveling
4. print head will run into side of printer

### Version of Marlin Firmware

2.1.1 bugfix

### Printer model

Ultimaker 2+

### Electronics

Stock

### Bed Leveling

UBL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9388061/Configuration.zip)

DerAndere1, 2022-09-10:
Could you use the following approach to measure the offsets?
0) compile with HOTEND_OFFSET_Z { 0.0, 0.0 }
1): run the following G-codes
```
T0
G28
T1
```
2) Move nozzle 1 above bed at a safe Z hight *
3)
a) Move toolhead a bit closer to the bed  *
b) shove a metal piece of known thickness between printer bed and nozzle 1
c) remove the metal piece !

4) Repeat step 3) until the metal piece does not fit between printer bed and nozzle 1
5) Move up until the metal piece fits snug  between printer bed and nozzle 1 *
6) Get current position with `M119`
7) set HOTEND_OFFSET_Z for tool 1 to ( (the Z value from step 6) - ( thickness of the used metal piece ) )

Note: 
* Use G1 or the LCD move axis menu or the user interface of your host  software

ansonl, 2022-09-12:
Yes, I ended up using that method to determine the correct height using the printer LCD interface. 

Set to T0 active
Home Z axis
Move Z bed downward to a safe distance like 10mm
Manually activate second extruder and move it downwards
Move Z bed upwards by 0.1mm increments through the LCD controls until the nozzle and bed produce enough friction on a piece of paper slid in between. 
Run M218 T1 Z-NN where NN is the Z offset for T1.

Bed teaming should still be disabled or take the active extruder offset into account for safety though.

DerAndere1, 2022-09-13:
This issue is why the other bed leveling methods temporarily change to T0 and it is why HOTEND_OFFSET_Z for tool 0 must be 0. A fix is proposed in the pull request 24749 linked above.

closed 2022-11-24 by github-actions[bot]