MarlinFirmware/Marlin #24660: M600 Filament Change should allow adding a test string.

jgbrown54, 2022-08-21:
### Is your feature request related to a problem? Please describe.

Feature Request Only

### Are you looking for hardware support?

No

### Describe the feature you want

M600 Filament Change should allow inclusion of a text string to identify why a filament change is needed. Color Change, Runout, etc.

GMagician, 2022-08-21:
Are you aware of HOST_PROMPT_SUPPORT/HOST_STATUS_NOTIFICATIONS? Maybe they already already satisfy your needs

jgbrown54, 2022-08-21:
I do use an M118 to tell me what color to change to. But that only works with a host. And with other messages it scrolls out of sight. And the M117 does provide a message to the LCD if your there to see it. But then it goes away when the Continue prompt is displayed. It's basically a matter of what works and what would work better. Future programmers could make it part of the Continue message.

eoyilmaz, 2023-07-29:
Yeah this is exactly what I was looking for.

eoyilmaz, 2023-07-30:
Found a way of kind of doing this: https://github.com/MarlinFirmware/Marlin/issues/20902#issuecomment-1656982393