MarlinFirmware/Marlin #24665: [BUG] (UBL 3-point mesh leveling 'broken')

titanrw, 2022-08-21:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

G29 J seems to 'mess up' an already correct ubl grid.

This is on an ender 5 plus running mainboard 'BTT_SKR_MINI_E3_V2_0'

I was previously on 2.0.9.2, and everything worked correctly.

I upgraded to 2.1.1, and rebuilt the UBL grid, as I had to initialize eeprom.  Saved grid to eeprom and verified it survives power cycles.  This is the initial grid:
```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | -0.16  -0.05  +0.01 [-0.01] -0.09  -0.20  -0.25
Recv:    |
Recv:  5 | -0.27  -0.20  -0.17  -0.17  -0.19  -0.24  -0.22
Recv:    |
Recv:  4 | -0.26  -0.20  -0.22  -0.20  -0.20  -0.19  -0.13
Recv:    |
Recv:  3 | -0.27  -0.26  -0.22  -0.24  -0.17  -0.22  -0.01
Recv:    |
Recv:  2 | -0.26  -0.22  -0.30  -0.27  -0.21  -0.23  -0.03
Recv:    |
Recv:  1 | -0.26  -0.25  -0.30  -0.34  -0.29  -0.28  -0.06
Recv:    |
Recv:  0 | -0.27  -0.27  -0.30  -0.33  -0.30  -0.26  -0.07
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
```

Max spread about 0.3mm.  Decently good for a 350x350 bed I think?

My normal start g-code is:
```
G28 
G29 L1
G29 A
G29 J
```

I was doing the G29 J tilting correction, as the 5 plus has independent z-screws that are not 'belted' together.  I try to align them as close as possible every time I power on the printer, but just in case they aren't exactly the same, I was using the mesh tilting to compensate.

However with 2.1.1 this is the mesh after the G29 J:
```
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.85  +0.72  +0.55  +0.29  -0.02  -0.37  -0.65
Recv:    |
Recv:  5 | +0.63  +0.47  +0.26  +0.03  -0.23  -0.51  -0.73
Recv:    |
Recv:  4 | +0.54  +0.36  +0.11  -0.11  -0.34  -0.57  -0.74
Recv:    |
Recv:  3 | +0.42  +0.20  +0.00  -0.25  -0.42  -0.70  -0.73
Recv:    |
Recv:  2 | +0.33  +0.13 [-0.18] -0.39  -0.56  -0.82  -0.85
Recv:    |
Recv:  1 | +0.22  -0.00  -0.29  -0.56  -0.75  -0.97  -0.99
Recv:    |
Recv:  0 | +0.10  -0.13  -0.39  -0.66  -0.86  -1.06  -1.10
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
```

This now has a spread of about 2mm.  The z-screws are definitely not that far off.  Not realizing this messed up the mesh, I damaged the build surface as the printer applied a -1mm offset when it was originally measured as almost 0.

As a work around for now, I've commented out the G29 J tilting, and it's printing ok with UBL on.  When I get a chance, I'll try -bugfix.  I also might just run 2.0-lts.

### Bug Timeline

2.0.9.2 didn't exhibit the issue.

### Expected behavior

UBL mesh correctly modified for slight z-screw tilt.

### Actual behavior

UBL mesh multiplied almost 10-fold.

### Steps to Reproduce

Run 'G29 J' after UBL mesh populated and active.

### Version of Marlin Firmware

2.1.1

### Printer model

Ender 5 plus

### Electronics

Stock other than BTT SKR Mini E3 2.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configs, and bed visualizer images below:

[2.1.1-configs.zip](https://github.com/MarlinFirmware/Marlin/files/9389893/2.1.1-configs.zip)
Before G29 J:
[image]
After G29 J:
[image]

Roxy-3D, 2022-08-30:
It would be helpful to know the values of the points it probed.    It kind of looks like one of those points was way off and as a result it took a 'correct' mesh and tilted it funny to fit the points that were probed.

titanrw, 2022-08-30:
Sorry for my ignorance, are these tilting point values returned during probing, or do you have to query for them with G-code?

I can check later today.

I did get 2.1-bugfix compiled, but haven't had a chance to flash and test.  If it needs an eeprom init, rebuilding the grid takes a while.  Is there no way to 'save' the grid and restore via the console?

titanrw, 2022-08-31:
Did some more testing.

Figured out G29 J V4 to get the probed data.

Here's the console output:

```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | -0.16  -0.05  +0.01  -0.01  -0.09  -0.20  -0.25
Recv:    |
Recv:  5 | -0.27  -0.20  -0.17  -0.17  -0.19  -0.24  -0.22
Recv:    |
Recv:  4 | -0.26  -0.20  -0.22  -0.20  -0.20  -0.19  -0.13
Recv:    |
Recv:  3 | -0.27  -0.26  -0.22 [-0.24] -0.17  -0.22  -0.01
Recv:    |
Recv:  2 | -0.26  -0.22  -0.30  -0.27  -0.21  -0.23  -0.03
Recv:    |
Recv:  1 | -0.26  -0.25  -0.30  -0.34  -0.29  -0.28  -0.06
Recv:    |
Recv:  0 | -0.27  -0.27  -0.30  -0.33  -0.30  -0.26  -0.07
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3

Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: 0.01
Recv:                 Corrected_Z=0.26
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: 0.01
Recv:                 Corrected_Z=-0.64
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.10
Recv:                 Corrected_Z=0.24
Recv: bed plane normal = [0.0034720,-0.0016478,0.9999926]
Recv: X:225.00 Y:317.00 Z:11.97 E:0.00 Count X:18000 Y:25360 Z:9576
Recv: ok P15 B3

Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.79  +0.69  +0.55  +0.31  +0.03  -0.30  -0.56
Recv:    |
Recv:  5 | +0.58  +0.44  +0.26 [+0.05] -0.18  -0.44  -0.63
Recv:    |
Recv:  4 | +0.49  +0.34  +0.11  -0.08  -0.29  -0.49  -0.64
Recv:    |
Recv:  3 | +0.38  +0.18  +0.01  -0.22  -0.36  -0.62  -0.63
Recv:    |
Recv:  2 | +0.29  +0.12  -0.17  -0.35  -0.50  -0.73  -0.74
Recv:    |
Recv:  1 | +0.19  -0.01  -0.27  -0.52  -0.68  -0.88  -0.88
Recv:    |
Recv:  0 | +0.08  -0.13  -0.37  -0.61  -0.79  -0.96  -0.98
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3

```
[... 94 lines cut ...]
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.184  +0.262  +0.296  +0.234  +0.133  -0.014  -0.095
Recv:    |
Recv:  5 | +0.051  +0.090  +0.084 [+0.057] +0.006  -0.076  -0.087
Recv:    |
Recv:  4 | +0.039  +0.068  +0.016  +0.005  -0.027  -0.053  -0.020
Recv:    |
Recv:  3 | +0.007  -0.015  -0.006  -0.057  -0.019  -0.100  +0.073
Recv:    |
Recv:  2 | -0.005  +0.003  -0.108  -0.110  -0.081  -0.133  +0.036
Recv:    |
Recv:  1 | -0.028  -0.049  -0.131  -0.202  -0.184  -0.200  -0.021
Recv:    |
Recv:  0 | -0.060  -0.091  -0.153  -0.214  -0.216  -0.207  -0.049
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
```

This time, Z was detected as +0.20 to + 0.07.  A 0.13mm difference is more what I would have expected.

The resulting mesh is the 'same', just offset up a bit.

I'm just going to stick with 2.0.9.2 for now.

aero4, 2022-09-08:
I also have problems with UBL coming from 2.0.9.2 to 2.1.1. And I think it's the same issue. I probed the bed muliple times and saved it to EEPROM. But when validating it with G26 (of course G29 J before) the mesh seems totally tilted! This happened on two different printers and that's why I looked into here.
I have to revert to 2.0.9.2 to use the printer...
So I am already looking forward to fix this issue.

aero4, 2022-10-25:
Are there any updates concerning this bug? Is it already fixed in the bugfix branch?

aero4, 2022-12-25:
I tested version 2.1.2 and the issue didn't occur so far. So it seems to be fixed with Marlin 2.1.2 and respectively with the issue #24622.
So this issue can be closed from my point of view...

titanrw, 2022-12-30:
I just tried 2.1.2, and it's having the same issue.

2.1.2 console output:

```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.08  +0.16  +0.20  +0.14  +0.05  -0.10  -0.17
Recv:    |
Recv:  5 | -0.04  +0.01  +0.00  -0.02  -0.07  -0.14  -0.15
Recv:    |
Recv:  4 | -0.03  +0.00  -0.05  -0.05  -0.08  -0.10  -0.07
Recv:    |
Recv:  3 | -0.05  -0.07  -0.05 [-0.10] -0.06  -0.14  +0.04
Recv:    |
Recv:  2 | -0.05  -0.03  -0.14  -0.14  -0.11  -0.15  +0.02
Recv:    |
Recv:  1 | -0.05  -0.07  -0.15  -0.22  -0.19  -0.21  -0.02
Recv:    |
Recv:  0 | -0.07  -0.10  -0.16  -0.21  -0.21  -0.20  -0.04
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: -0.01
Recv:                 Corrected_Z=0.06
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.11
Recv:                 Corrected_Z=-0.70
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.06
Recv:                 Corrected_Z=0.04
Recv: bed plane normal = [0.0029235,-0.0013995,0.9999948]
Recv: X:225.00 Y:317.00 Z:12.10 E:0.00 Count X:18000 Y:25360 Z:9680
Recv: ok P15 B3
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.72  +0.63  +0.49  +0.26  -0.02  -0.33  -0.58
Recv:    |
Recv:  5 | +0.52  +0.39  +0.21 [+0.01] -0.21  -0.46  -0.64
Recv:    |
Recv:  4 | +0.44  +0.29  +0.07  -0.11  -0.31  -0.51  -0.64
Recv:    |
Recv:  3 | +0.34  +0.14  -0.02  -0.24  -0.37  -0.62  -0.62
Recv:    |
Recv:  2 | +0.26  +0.09  -0.19  -0.36  -0.50  -0.72  -0.73
Recv:    |
Recv:  1 | +0.17  -0.03  -0.28  -0.52  -0.67  -0.86  -0.85
Recv:    |
Recv:  0 | +0.06  -0.14  -0.37  -0.60  -0.77  -0.93  -0.95
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
```

Reflashed 2.0.9.2 without changing anything hardware wise on the printer.

2.0.9.2 output:

(Didn't get the pre-tilt mesh, but it was basically the same.  Corrected Z values are way less with simply a firmware flash.)

```
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: 0.01
Recv:                 Corrected_Z=0.08
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.08
Recv:                 Corrected_Z=0.09
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.03
Recv:                 Corrected_Z=0.02
Recv: bed plane normal = [-0.0000519,0.0002481,1.0000000]
Recv: X:225.00 Y:317.00 Z:11.93 E:0.00 Count X:18000 Y:25360 Z:9544
Recv: ok P15 B3
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.080  +0.166  +0.207  +0.152  +0.058  -0.082  -0.156
Recv:    |
Recv:  5 | -0.022  +0.024  +0.024 [+0.005] -0.039  -0.114  -0.118
Recv:    |
Recv:  4 | -0.004  +0.032  -0.013  -0.017  -0.042  -0.061  -0.020
Recv:    |
Recv:  3 | -0.006  -0.021  -0.005  -0.049  -0.004  -0.078  +0.102
Recv:    |
Recv:  2 | +0.012  +0.027  -0.077  -0.072  -0.036  -0.080  +0.095
Recv:    |
Recv:  1 | +0.019  +0.005  -0.069  -0.134  -0.108  -0.118  +0.068
Recv:    |
Recv:  0 | +0.017  -0.007  -0.062  -0.116  -0.110  -0.095  +0.071
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
```

Since it's only me having the problem now, it must be something in my configs.

Aero4, which printer do you have 2.1.2 running on?  Can you post your config?

aero4, 2022-12-30:
To be honest I didn't probe the bed twice and compared the values, I just probed once and there were no issues while printing... If you want I can run the same code as you did for testing.

I am running Marlin 2.1.2 on a custom printer based on the Velleman K8200 with a "3DTouch" probe. I always track the differences I modify in the config files which are the following:
[image]

titanrw, 2022-12-30:
Hello.

Thanks for the config difference from default.

What do you mean when you say you don't probe the bed twice?  I do use "#define MULTIPLE_PROBING 2" and it looks like  you don't.

The other main difference I see is that I use "#define ENABLE_LEVELING_AFTER_G28" whereas you use "#define RESTORE_LEVELING_AFTER_G28".  I don't fully understand one vs the other.  I've got a 7x7 grid and you use 6x6, but my bed is 350mm compared to 200.

What does your start gcode look like?

aero4, 2023-01-07:
Sorry for the late reply,

you're welcome.

I only probed the bed once and used these values for the bed grid and didn't repeat that.
No I don't use the MULTIPLE_PROBING setting ...

To be honest I also don't understand every single setting I used. I took most of the probing settings from tutorials and they were proved to be working so far.

This is my starting GCode:
`M117 Preparing...

M107 ; switch part fan off

M42 P6 S255 ; switch fans on
M777 P8 S1 ; switch light on

G91 ; set relative coordinates
G0 E-10 F900 ; retract 10mm with 900mm/min
G90 ; set absolute coordinates

M117 Homing...
G29 L1 ; load bed mesh from slot 1
G28 ; home all axes
M117 Auto leveling...
G29 J ; measure postion with 3 points

M117 Preparing...

G0 X0 Y0 F4800 ; go to home position with 4800mm/min speed
G0 Z0 F1500; go to z home position

T0 ; select tool 0
G91 ; set relative coordinates
G0 E22 ; extrude 22mm with 900mm/min speed
G90 ; set absolute coordinates

M117 Printing...`

titanrw, 2023-01-07:
Can you run the following G code right after the printers been powered up and post the console output after it's ran?

It won't print anything, but does need to move the head around.

```
G28 ;home

G29 L1 ;load mesh

G29 A  ;activate mesh

G29 T  ;show mesh pre tilt

G29 J V4 ;tilt mesh showing probed Z values

G29 T  ;show mesh post tilt.
```

aero4, 2023-01-07:
Sure, this is my console output when sending the commands you stated:

```
SENT: G28
READ: Active Extruder: 0
READ: Active Extruder: 0
READ: X:49.10 Y:95.00 Z:12.40 E:0.00 Count X:1577 Y:6104 Z:26453
SENT: G29 L1
READ: Mesh loaded from slot 1
READ: Done.
SENT: G29 A
READ: Unified Bed Leveling System v1.01 active
SENT: G29 T
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,185)                              (190,185)
READ:         0       1       2       3       4       5      
READ:  5 | -0.18  -0.16  -0.14  -0.09  -0.19  -0.41 
READ:    |
READ:  4 | +0.03  -0.01  -0.05  -0.09  -0.14  -0.37 
READ:    |
READ:  3 | +0.00  -0.02  -0.04  -0.06  -0.14  -0.34 
READ:    |
READ:  2 | -0.02  -0.04  -0.06 [-0.08] -0.17  -0.41 
READ:    |
READ:  1 | +0.16  +0.07  -0.02  -0.12  -0.27  -0.42 
READ:    |
READ:  0 | +0.11  +0.02  -0.07  -0.16  -0.30  -0.51 
READ:         0       1       2       3       4       5      
READ:     ( 10, 10)                              (190, 10)
READ: 
SENT: G29 J
READ: Tilting mesh (1/3)
READ: Tilting mesh (2/3)
READ: Tilting mesh (3/3)
READ: X:69.55 Y:183.00 Z:12.40 E:0.00 Count X:2234 Y:11758 Z:26453
SENT: G29 T
READ: 
READ: Bed Topography Report:
READ: 
READ:     ( 10,185)                              (190,185)
READ:         0       1       2       3       4       5      
READ:  5 | -0.22  -0.19  -0.15 [-0.09] -0.17  -0.38 
READ:    |
READ:  4 | -0.03  -0.05  -0.08  -0.10  -0.13  -0.34 
READ:    |
READ:  3 | -0.06  -0.06  -0.07  -0.07  -0.14  -0.32 
READ:    |
READ:  2 | -0.09  -0.09  -0.09  -0.09  -0.17  -0.39 
READ:    |
READ:  1 | +0.09  +0.02  -0.06  -0.14  -0.28  -0.41 
READ:    |
READ:  0 | +0.04  -0.04  -0.11  -0.18  -0.31  -0.50 
READ:         0       1       2       3       4       5      
READ:     ( 10, 10)                              (190, 10)
```

titanrw, 2023-01-07:
Thanks.  Yes, it looks very normal.

Not sure why mine ends up applying a 2mm tilt.

I suppose I can try recompiling with absolute minimal config changes made.  Like only set what's critical for stepper movement.   Axis direction, z-probe as end stop, etc.

I'll close the bug for now as I seem to be the only one having the issue.  If I can find out what's causing it I'll re-open.

Thanks again for your help.

titanrw, 2023-01-13:
Not that I can help, but can you show the output of G29 J V4?  The corrected Z values seem to be off.

For example:

Tilting point 2/3 corrected Z shows -0.7:

```
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: -0.01
Recv:                 Corrected_Z=0.06
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.11
Recv:                 Corrected_Z=-0.70
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.06
Recv:                 Corrected_Z=0.04
Recv: bed plane normal = [0.0029235,-0.0013995,0.9999948]
```

Previous firmware before I noticed the bug have all corrected Z under absolute value of 0.1:
```
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: 0.01
Recv:                 Corrected_Z=0.08
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.08
Recv:                 Corrected_Z=0.09
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.03
Recv:                 Corrected_Z=0.02
Recv: bed plane normal = [-0.0000519,0.0002481,1.0000000]
```

I did manage to narrow it down to a specific version of marlin.  I was running 2.0.9.2 as mentioned.  I tried 2.0.9.3, and it was OK.  Compiled and flashed 2.0.9.4, and it has this problem.  Something changed between 2.0.9.3 and 9.4.

I did look through the commits between 9.3 and 9.4, but I'm not much of a coder, especially in C.  I also looked at the code, trying to figure out how the corrected Z could get so 'whacked' in 9.4, but didn't really get anywhere.  I did see there were some debugging options around function get_z_correction() in ubl.h, but I couldn't figure out how to enable them.

I'm currently running my machine on 2.0.9.3 as it's the latest version that doesn't seem affected by this.

aero4, 2023-01-13:
I reverted to Version 2.0.9 and for the exact same commands and config files the mesh values are for the first G29 J V4:
```
READ:     ( 10,255)                                                      (410,255)
READ:         0       1       2       3       4       5       6       7       8      
READ:  5 | +2.227  +1.611  +1.124  +0.637  +0.320  +0.103  -0.124  -0.261  -0.398 
READ:    |
READ:  4 | +1.932  +1.416  +0.929  +0.452  +0.175  -0.032  -0.209  -0.316  -0.443 
READ:    |
READ:  3 | +1.667  +1.200  +0.734  +0.267  +0.030 [-0.127] -0.254  -0.321  -0.428 
READ:    |
READ:  2 | +1.252  +0.895  +0.538  +0.182  -0.055  -0.172  -0.249  -0.346  -0.433 
READ:    |
READ:  1 | +0.877  +0.620  +0.363  +0.106  -0.060  -0.167  -0.214  -0.331  -0.438 
READ:    |
READ:  0 | +0.572  +0.395  +0.218  +0.031  -0.066  -0.142  -0.239  -0.306  -0.383 
READ:         0       1       2       3       4       5       6       7       8      
READ:     ( 10, 10)                                                      (410, 10)
```
And the second:
```
READ:         0       1       2       3       4       5       6       7       8      
READ:  5 | +2.270  +1.647  +1.154  +0.661  +0.338  +0.115  -0.118  -0.261  -0.404 
READ:    |
READ:  4 | +1.970  +1.447  +0.954  +0.471  +0.188  -0.025  -0.208  -0.321  -0.454 
READ:    |
READ:  3 | +1.701  +1.228  +0.755  +0.282  +0.039 [-0.124] -0.257  -0.330  -0.443 
READ:    |
READ:  2 | +1.281  +0.918  +0.555  +0.192  -0.051  -0.174  -0.257  -0.360  -0.453 
READ:    |
READ:  1 | +0.901  +0.638  +0.375  +0.112  -0.061  -0.174  -0.227  -0.350  -0.463 
READ:    |
READ:  0 | +0.591  +0.408  +0.225  +0.032  -0.071  -0.154  -0.257  -0.330  -0.413 
READ:         0       1       2       3       4       5       6       7       8      
READ:     ( 10, 10)                                                      (410, 10)
```
Which are obviously way more covering...

I saw that G29 J V4 with version 2.1.2 shows Corrected_Z values in the range of 0.5mm and with 2.0.9 in the range of 0.01mm. So very likely there the problem is located...

titanrw, 2023-01-13:
Bug re-opened.

Which version of 2.0.9 specifically?  I bet 9.3 will work, and 9.4 will not.

I did damage the build plate before realizing what was going on.  Now I know to manually do a G29 J V4 (check probed values), and double check G29 T tilted mesh, to verify before printing.

aero4, 2023-01-13:
> Bug re-opened.
> 
> Which version of 2.0.9 specifically? I bet 9.3 will work, and 9.4 will not.
> 
> I did damage the build plate before realizing what was going on. Now I know to manually do a G29 J V4 (check probed values), and double check G29 T tilted mesh, to verify before printing.

It's even 2.0.9.2...
I try to avoid testing different versions because adapting the config files takes much time, but I can test with 2.0.9.4 if you want?

Yes my build plate also have one or two deep scratches because of this bug. I am back on 2.0.9 and have to revert back to this version for two more printers which are remote and I have to explain the people there how to update the firmware and hope it didn't destroy their buit plate or print head...

titanrw, 2023-01-13:
Only if you want to verify that 9.3 -> 9.4 is where it broke.  I ran 9.2 for a long time, so if it's working, why mess with it.

Setting the configs are a bit of a pain.  There's no pre-made config for my printer, with my exact mainboard.

I find its fastest to use notepad++ with the compare plugin.  Open the old_config and new_config, then run the compare.  It'll show which lines are different, and you can 'move over' config options you need from the old to new files.

Still takes about half an hour, but at least the bed mesh seems to persist.

aero4, 2023-01-13:
Good to know which version to avoid...

I tracked my changes compared to stock and have to adapt about 80 settings, which as you said needs about half an hour.

Yes I will also stick to the old version, the new features are unable to compensate this huge bug.

Why are we the only two with that issues (or care about it)? Shouldn't that concern everyone with version >2.0.9.3 and having UBL enabled?

titanrw, 2023-01-13:
I don't know why we're the only ones to notice.  From what I can tell, it would affect anyone on 2.0.9.4 and later, using UBL, and tilting the mesh with G29 J.  Maybe it's also tied to a specific config option we're both using.  Other than #enable UBL that is.  Haven't figured that one out yet.

If my printer had the dual Z screws mechanically 'tied' together, then I wouldn't both with G29 J, and this wouldn't affect me.  But I figured that since I'm not going to get them perfectly set every single time I power on the printer, it could compensate.

titanrw, 2023-01-13:
> The bug re-occured! On my second printer i swapped the mainboard and used Marlin 2.1.1 with similar setting than my first printer.

***snip***

I just noticed you change the mainboard.  Maybe the particular mainboard used has something to do with whether this bug manifests or not.

As mentioned in the first post, my printer is driven by a BigTreeTech SKR Mini E3 v2.0.

What mainboard did you change ***from***, and which one did you switch ***TO***?

aero4, 2023-01-13:
I just swapped the board. So my previous BTT SKR 1.4 was damaged and I bought another one to replace.
So its very likely not the board.

titanrw, 2023-01-16:
I tested the latest -bugfix from today (2023-01-16).  The issue still occurs:

```
Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Jan 16 2023 14:07:37) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin

Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:    |
Recv:  5 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:    |
Recv:  4 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:    |
Recv:  3 |  0.00   0.00   0.00 [ 0.00]  0.00   0.00   0.00
Recv:    |
Recv:  2 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:    |
Recv:  1 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:    |
Recv:  0 |  0.00   0.00   0.00   0.00   0.00   0.00   0.00
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)

Send: G28
Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.08  +0.16  +0.20  +0.14  +0.05  -0.10  -0.17
Recv:    |
Recv:  5 | -0.04  +0.01  +0.00  -0.02  -0.07  -0.14  -0.15
Recv:    |
Recv:  4 | -0.03  +0.00  -0.05  -0.05  -0.08  -0.10  -0.07
Recv:    |
Recv:  3 | -0.05  -0.07  -0.05 [-0.10] -0.06  -0.14  +0.04
Recv:    |
Recv:  2 | -0.05  -0.03  -0.14  -0.14  -0.11  -0.15  +0.02
Recv:    |
Recv:  1 | -0.05  -0.07  -0.15  -0.22  -0.19  -0.21  -0.02
Recv:    |
Recv:  0 | -0.07  -0.10  -0.16  -0.21  -0.21  -0.20  -0.04
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: echo:busy: processing
Recv: Bed X: 50.00 Y: 50.00 Z: 0.06
Recv:                 Corrected_Z=0.13
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: 0.00
Recv:                 Corrected_Z=-0.59
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: 0.12
Recv:                 Corrected_Z=0.11
Recv: bed plane normal = [0.0027639,-0.0012966,0.9999954]
Recv: X:225.00 Y:317.00 Z:12.10 E:0.00 Count X:18000 Y:25360 Z:9680
Send: G29 T
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.75  +0.66  +0.54  +0.31  +0.05  -0.26  -0.50
Recv:    |
Recv:  5 | +0.55  +0.43  +0.26 [+0.07] -0.14  -0.38  -0.55
Recv:    |
Recv:  4 | +0.48  +0.35  +0.13  -0.04  -0.23  -0.42  -0.55
Recv:    |
Recv:  3 | +0.38  +0.20  +0.05  -0.16  -0.29  -0.53  -0.52
Recv:    |
Recv:  2 | +0.31  +0.16  -0.12  -0.28  -0.41  -0.62  -0.62
Recv:    |
Recv:  1 | +0.22  +0.04  -0.20  -0.43  -0.58  -0.75  -0.73
Recv:    |
Recv:  0 | +0.13  -0.06  -0.28  -0.51  -0.67  -0.82  -0.82
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
```

apulver, 2023-01-21:
G29J does a least-squares approximation to fit a plane and obtain a linear transform to apply to the mesh points, correct? I don't understand the point of this unless a sample larger than three points is used. Otherwise, isn't it equivalent to tramming the bed?

apulver, 2023-01-21:
This is a long shot, but try running a G29P5C so that the grid is centered before using G29J.

titanrw, 2023-01-22:
> This is a long shot, but try running a G29P5C so that the grid is centered before using G29J.

Here's including G29 P5C before a G29 J:

```
Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.03  +0.12  +0.17  +0.13  +0.04  -0.09  -0.16
Recv:    |
Recv:  5 | -0.08  -0.03  -0.02  -0.03  -0.06  -0.13  -0.13
Recv:    |
Recv:  4 | -0.07  -0.03  -0.06  -0.06  -0.07  -0.08  -0.03
Recv:    |
Recv:  3 | -0.08  -0.09  -0.06 [-0.10] -0.04  -0.11  +0.08
Recv:    |
Recv:  2 | -0.07  -0.04  -0.14  -0.13  -0.08  -0.12  +0.07
Recv:    |
Recv:  1 | -0.07  -0.07  -0.14  -0.19  -0.16  -0.16  +0.03
Recv:    |
Recv:  0 | -0.08  -0.09  -0.14  -0.18  -0.17  -0.14  +0.03
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Send: G29 P5C
Recv: # of samples: 49
Recv: Mean Mesh Height: -0.062754
Recv: Standard Deviation: 0.080287
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.09  +0.18  +0.23  +0.19  +0.10  -0.03  -0.09
Recv:    |
Recv:  5 | -0.02  +0.03  +0.04  +0.03  -0.00  -0.07  -0.06
Recv:    |
Recv:  4 | -0.01  +0.04  +0.00  +0.01  -0.01  -0.02  +0.03
Recv:    |
Recv:  3 | -0.02  -0.02  +0.00 [-0.03] +0.02  -0.04  +0.14
Recv:    |
Recv:  2 | -0.01  +0.02  -0.08  -0.06  -0.02  -0.05  +0.13
Recv:    |
Recv:  1 | -0.00  -0.01  -0.08  -0.13  -0.10  -0.10  +0.10
Recv:    |
Recv:  0 | -0.01  -0.03  -0.07  -0.12  -0.11  -0.08  +0.09
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: -0.10
Recv:                 Corrected_Z=-0.08
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.11
Recv:                 Corrected_Z=-0.87
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: -0.03
Recv:                 Corrected_Z=-0.09
Recv: bed plane normal = [0.0030143,-0.0014809,0.9999944]
Recv: X:225.00 Y:317.00 Z:12.10 E:0.00 Count X:18000 Y:25360 Z:9680
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.61  +0.53  +0.40  +0.17  -0.10  -0.41  -0.65
Recv:    |
Recv:  5 | +0.41  +0.29  +0.12 [-0.07] -0.29  -0.53  -0.71
Recv:    |
Recv:  4 | +0.34  +0.20  -0.01  -0.19  -0.38  -0.57  -0.71
Recv:    |
Recv:  3 | +0.24  +0.05  -0.10  -0.32  -0.44  -0.69  -0.68
Recv:    |
Recv:  2 | +0.16  +0.01  -0.27  -0.43  -0.57  -0.78  -0.78
Recv:    |
Recv:  1 | +0.08  -0.11  -0.36  -0.59  -0.74  -0.92  -0.90
Recv:    |
Recv:  0 | -0.02  -0.22  -0.44  -0.67  -0.83  -0.99  -0.99
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
```

Corrected_Z is still 'wrong' when tilting.

This is -bugfix from Jan 16.

Flashing back to 2.0.9.3 now...

apulver, 2023-01-28:
The "Corrected_Z" value for the second point seems off. I had a cursory look at the diff for G29.ccp between 2.0.9.3 and the current version. A few things were changed but nothing stood out. 

Numerical algorithms can be tricky and certain ways of performing rotations are more numerically stable than others. When I have the time/motivation I'll have a closer look at what they're doing. In the meantime, could you try a G29J4 or G29J5? If your bed isn't the most even then a 4x4, 5x5, or even 3x3 sample of the bed will yield a much better approximation of the bed's planar component than the default three point sample. 

Incidentally, it may be beneficial if UBL's G29 aligned the planar component of the mesh itself with the printer's XY axis at the time that the mesh is sampled and/or as an option to G29J: https://github.com/MarlinFirmware/Marlin/issues/25262#issue-1551892682 This could probably be implemented reusing some code from G29J, provided that G29J works in the first place.

titanrw, 2023-01-28:
> The "Corrected_Z" value for the second point seems off. I had a cursory look at the diff for G29.ccp between 2.0.9.3 and the current version. A few things were changed but nothing stood out.

The problem starts to occur between 2.0.9.3 and 9.4.  So any changes past 9.4 shouldn't apply to this problem.

 
> Numerical algorithms can be tricky and certain ways of performing rotations are more numerically stable than others. When I have the time/motivation I'll have a closer look at what they're doing. In [...]

I'll try a > 3 point tilt, both on 9.3 and 9.4 and compare.  Realistically I'm only looking for any possible left-right (X axis) tilt, as that's what would be introduced by non perfectly aligned dual z screws on the ender 5 plus.

titanrw, 2023-01-29:
Here's 2.0.9.3 with a G29 J4 V4:

```
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Recv: ok P15 B3
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.041  +0.129  +0.173  +0.121  +0.029  -0.107  -0.179
Recv:    |
Recv:  5 | -0.065  -0.017  -0.013  -0.030  -0.071  -0.143  -0.144
Recv:    |
Recv:  4 | -0.051  -0.012  -0.054  -0.055  -0.077  -0.094  -0.050
Recv:    |
Recv:  3 | -0.057  -0.068  -0.050 [-0.091] -0.043  -0.114  +0.069
Recv:    |
Recv:  2 | -0.042  -0.024  -0.125  -0.117  -0.078  -0.120  +0.058
Recv:    |
Recv:  1 | -0.038  -0.050  -0.121  -0.183  -0.154  -0.161  +0.028
Recv:    |
Recv:  0 | -0.044  -0.065  -0.117  -0.168  -0.160  -0.141  +0.027
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
Send: G29 J4 V4
Recv: Tilting mesh point 1/16
Recv: 
Recv: Bed X: 50.00 Y: 50.00 Z: -0.04
Recv:                 Corrected_Z=0.01
Recv: Tilting mesh point 2/16
Recv: 
Recv: Bed X: 50.00 Y: 136.67 Z: -0.02
Recv:                 Corrected_Z=0.02
Recv: Tilting mesh point 3/16
Recv: 
Recv: Bed X: 50.00 Y: 223.33 Z: -0.02
Recv:                 Corrected_Z=0.01
Recv: Tilting mesh point 4/16
Recv: 
Recv: Bed X: 50.00 Y: 310.00 Z: 0.01
Recv:                 Corrected_Z=0.01
Recv: Tilting mesh point 5/16
Recv: 
Recv: Bed X: 136.67 Y: 310.00 Z: 0.07
Recv:                 Corrected_Z=0.05
Recv: Tilting mesh point 6/16
Recv: 
Recv: Bed X: 136.67 Y: 223.33 Z: -0.03
Recv:                 Corrected_Z=0.02
Recv: Tilting mesh point 7/16
Recv: 
Recv: Bed X: 136.67 Y: 136.67 Z: -0.07
Recv:                 Corrected_Z=0.04
Recv: Tilting mesh point 8/16
Recv: 
Recv: Bed X: 136.67 Y: 50.00 Z: -0.09
Recv:                 Corrected_Z=0.05
Recv: Tilting mesh point 9/16
Recv: 
Recv: Bed X: 223.33 Y: 50.00 Z: -0.03
Recv:                 Corrected_Z=0.13
Recv: Tilting mesh point 10/16
Recv: 
Recv: Bed X: 223.33 Y: 136.67 Z: 0.01
Recv:                 Corrected_Z=0.09
[... 418 lines cut ...]
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.65  +0.58  +0.47  +0.26  +0.02  -0.28  -0.50
Recv:    |
Recv:  5 | +0.46  +0.35  +0.20 [+0.02] -0.17  -0.40  -0.56
Recv:    |
Recv:  4 | +0.38  +0.26  +0.07  -0.09  -0.27  -0.44  -0.55
Recv:    |
Recv:  3 | +0.29  +0.12  -0.02  -0.22  -0.32  -0.55  -0.52
Recv:    |
Recv:  2 | +0.21  +0.07  -0.18  -0.33  -0.45  -0.65  -0.62
Recv:    |
Recv:  1 | +0.13  -0.04  -0.27  -0.49  -0.61  -0.78  -0.74
Recv:    |
Recv:  0 | +0.03  -0.15  -0.36  -0.56  -0.71  -0.85  -0.84
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3

```

Tilting point 2/3 is still weird.

apulver, 2023-01-31:
A good place to start debugging this might be to examine the data passed to and the data changed by these function calls, and compare them between 2.0.9.3 and 2.0.9.4. 

https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L1456
https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L1470

Less likely but still possible, it could be a loss of precision somewhere along the line, as computations such as https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/src/libs/vector_3.cpp#L68 
https://github.com/MarlinFirmware/Marlin/blob/2.0.9.3/Marlin/src/libs/vector_3.cpp#L117
may not be numerically stable, considering the relatively small z values. 

The code in general seems to use a lot of macros and inline functions, and frequently there are functions defined within header files.

apulver, 2023-01-31:
If/when I have time I'll try to reproduce the problem on my machine and see what I can do.

jamespearson04, 2023-03-01:
Think I might have found something, line 268 of ubl.h in 2.0.9.4, get_mesh_x is a static float, but x0 and x1 which are set using it are being initialized as const uint8_t.
In 2.0.9.3 there was no x0 and x1 variables, the function was used directly as an argument on what becomes line 269 and 270.

https://github.com/MarlinFirmware/Marlin/blob/f471eab1a2834c4e65477d978ea9f0349542b302/Marlin/src/feature/bedlevel/ubl/ubl.h#L268
https://github.com/MarlinFirmware/Marlin/blob/369542db3bb34c28196a77838a2f018180a7da74/Marlin/src/feature/bedlevel/ubl/ubl.h#L268

I'll try changing the variable type and test it tomorrow, will let you all know how it goes.

jamespearson04, 2023-03-02:
This alteration yields the following leveling topologies (First is after a fresh probing G29 P1 and P3, second is after G29 J), to me it looks as though it could be correct, it's much much closer than it was previously:

Bed Topography Report:

    ( 10,290)                                                              (290,290)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.23  +0.07  -0.09  -0.25 [-0.11] +0.01  +0.31  +0.49  +0.69  +0.79
   |
 8 | +0.35  +0.19  +0.03  -0.13  -0.02  +0.11  +0.34  +0.49  +0.70  +0.80
   |
 7 | +0.25  +0.13  +0.01  -0.11  +0.01  +0.13  +0.33  +0.45  +0.61  +0.69
   |
 6 | +0.33  +0.19  +0.05  -0.09  +0.06  +0.21  +0.36  +0.45  +0.59  +0.65
   |
 5 | +0.37  +0.23  +0.09  -0.05  +0.10  +0.22  +0.37  +0.45  +0.57  +0.62
   |
 4 | +0.23  +0.14  +0.06  -0.02  +0.08  +0.24  +0.37  +0.44  +0.56  +0.63
   |
 3 | +0.27  +0.17  +0.06  -0.04  +0.09  +0.18  +0.36  +0.42  +0.56  +0.68
   |
 2 | +0.23  +0.12  +0.01  -0.09  +0.01  +0.19  +0.32  +0.44  +0.60  +0.75
   |
 1 | +0.29  +0.14  -0.00  -0.15  +0.11  +0.20  +0.32  +0.41  +0.65  +0.76
   |
 0 | -0.06  -0.12  -0.17  -0.23  -0.08  +0.20  +0.29  +0.37  +0.59  +0.73
        0       1       2       3       4       5       6       7       8       9
    ( 10, 10)                                                              (290, 10)

ok
Tilting mesh (1/3)
Tilting mesh (2/3)
Tilting mesh (3/3)
X:121.50 Y:298.00 Z:11.70 E:0.00 Count X:9720 Y:23840 Z:4680
ok

Bed Topography Report:

    ( 10,290)                                                              (290,290)
        0       1       2       3       4       5       6       7       8       9
 9 | +0.35  +0.19  +0.02  -0.15  -0.01 [+0.10] +0.39  +0.57  +0.76  +0.86
   |
 8 | +0.46  +0.30  +0.13  -0.03  +0.07  +0.19  +0.42  +0.57  +0.77  +0.86
   |
 7 | +0.36  +0.23  +0.11  -0.02  +0.09  +0.21  +0.41  +0.52  +0.67  +0.74
   |
 6 | +0.44  +0.29  +0.14  -0.01  +0.14  +0.28  +0.43  +0.51  +0.65  +0.70
   |
 5 | +0.47  +0.32  +0.17  +0.03  +0.17  +0.29  +0.43  +0.50  +0.62  +0.67
   |
 4 | +0.32  +0.23  +0.14  +0.05  +0.15  +0.30  +0.42  +0.49  +0.60  +0.67
   |
 3 | +0.36  +0.24  +0.13  +0.02  +0.15  +0.24  +0.41  +0.46  +0.60  +0.71
   |
 2 | +0.31  +0.19  +0.08  -0.03  +0.07  +0.24  +0.36  +0.47  +0.63  +0.77
   |
 1 | +0.36  +0.21  +0.06  -0.10  +0.16  +0.24  +0.36  +0.44  +0.67  +0.78
   |
 0 | +0.01  -0.06  -0.12  -0.18  -0.04  +0.24  +0.32  +0.40  +0.61  +0.74
        0       1       2       3       4       5       6       7       8       9
    ( 10, 10)                                                              (290, 10)

I'll do my best to continue to validate this

jamespearson04, 2023-03-02:
Done some G26s, and am pretty convinced that this has solved the issue. I'll try and make a push request once I'm out of work this evening.

This was tested on 2.1.x bugfix , though I think the UBL code hasn't changed since 2.9.4, so it should work for all versions between.

titanrw, 2023-03-02:
I didn't really understand your original post as to what you figured was wrong.  My C skills are kinda non-existent.  Can you post a diff, or the changes you made?  I can verify tonight on my Ender 5 plus.

jamespearson04, 2023-03-02:
In the refered file, ubl.h, change line 267 and 268 from: 
```
    const uint8_t mx = _MIN(cx, (GRID_MAX_POINTS_X) - 2) + 1, my = _MIN(cy, (GRID_MAX_POINTS_Y) - 2) + 1,
                  x0 = get_mesh_x(cx), x1 = get_mesh_x(cx + 1);
```
to:
```
    const uint8_t mx = _MIN(cx, (GRID_MAX_POINTS_X) - 2) + 1, my = _MIN(cy, (GRID_MAX_POINTS_Y) - 2) + 1;
const float x0 = get_mesh_x(cx), x1 = get_mesh_x(cx + 1);
```

jamespearson04, 2023-03-02:
Got a PR open, so can clone from here until it's done: https://github.com/jamespearson04/Marlin

apulver, 2023-03-02:
Good find, though it's not immediately obvious to me how it relates to the bug in this issue. I wouldn't think that G29J has anything to do with the interpolation procedure itself.

titanrw, 2023-03-02:
> Got a PR open, so can clone from here until it's done: https://github.com/jamespearson04/Marlin

Thanks.

I'll try in 4-5 hours.  I manually changed (patched) ubl.h for 2.1.2.

Reading the PR and it now makes sense.  My 3 point 'tilt' only had issues with point 2, which is the farthest right (highest X value) at x=310.

My 4x4 16 point tilting only had problems with x>255 tilting points.

This bug wouldn't have affected me if I had a smaller machine (ender 3 ish).  All meshes here show a bed of X > 255.

jamespearson04, 2023-03-02:
I can't say I've traced back the particular part where G29 J uses it, but the level probe points don't necessarily lie on top of the mesh points, so the interpolation will be used at some point along the line to calculate the mesh height at the level probe points in order to work out the difference for the adjustment.

If the bed is big enough, the rightmost point will have an x coordinate above 255mm, which will bring out the rollover issue, giving completely incorrect values at the interpolation stage, which when used for the tit gives another completely wrong result.

jamespearson04, 2023-03-02:
@titanrw Yep, exactly. Even if we didn't screw the leveling with G29 J, the same function is used for the leveling whilst printing, so any prints past x=255mm would level erratically anyway. Explains some other issues people have been having with G26 past that point too.

jamespearson04, 2023-03-02:
@apulver This is the function for the tilt leveling:
https://github.com/MarlinFirmware/Marlin/blob/49911c8f5d7740071e53d5a38f102ca1b1474dfa/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L1480

Plenty of uses of  ```get_z_correction()```, for the 3 point:
https://github.com/MarlinFirmware/Marlin/blob/49911c8f5d7740071e53d5a38f102ca1b1474dfa/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L1515

and the grid version:
https://github.com/MarlinFirmware/Marlin/blob/49911c8f5d7740071e53d5a38f102ca1b1474dfa/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L1604

apulver, 2023-03-02:
> Plenty of uses of `get_z_correction()`, for the 3 point:

I see now, I think you've got it. 

>  the level probe points don't necessarily lie on top of the mesh points, so the interpolation will be used at some point along the line to calculate the mesh height at the level probe points in order to work out the difference for the adjustment. 

I see, so in this case, the planar component of the mesh itself is also approximated by taking the plane defined by those three points on the mesh. This seems less than optimal since it can be computed exactly by using least squares on the whole mesh, which is essentially part of what I suggest here https://github.com/MarlinFirmware/Marlin/issues/25262#issue-1551892682, but I digress.

jamespearson04, 2023-03-02:
So if I'm understanding correctly, your FR is essentially about pre-tilting the mesh to it's most level position before storing it? Interesting, though the bed should be trammed before probing, and shouldn't move a large amount after, so I'd wonder if it'd be worthwhile - more transformations might result in a loss of accuracy, and since in theory you don't need to tilt it initially, you may be starting with a less accurate mesh.

With regards to the leveling though, the fact it uses only the 3 points means that any high or low spots don't affect the measurement. If you were to try and compare planes, a high spot under one of your 3 points would result in the measured plane being very different to the computed plane. Whereas with the current method, any such high point would also be present in the mesh, so would automatically be accounted for.

apulver, 2023-03-02:
> So if I'm understanding correctly, your FR is essentially about pre-tilting the mesh to it's most level position before storing it? 

Yes, essentially.

>Whereas with the current method, any such high point would also be present in the mesh, so would automatically be accounted for.

I do understand that. What they're doing now is a quick way to account for how the bed's planar component has changed. That is, by taking the delta for each point to compute the transformation. There's nothing terribly wrong with that, but I think having the UBL mesh handle the bed's non-planar component and G29J handle the planar component would be a clearer and cleaner way to organize the code and documentation and would actually simplify some of the math. 

Incidentally, I imagine that the presence of G29J's menu option is rather confusing for most users. "Do I make a mesh or use three point leveling? Why are they both present?".

jamespearson04, 2023-03-02:
I have to disagree entirely, it just seems kind of pointless and not beneficial at all. If it even works, the only effect to be seen will be increased resource usage.

You mention overwriting in your FR, but that isn't relevant either because:

- You don't save the tilted mesh in the slot, you keep the original
- The mesh loaded into memory must be tilted, else the move planner has to do it on the fly, which will hurt performance.

Basically there's no good way around the fact you need the tilted mesh in memory. If you try and seperate the plane from the surface, you just have to put them back together again anyway. 

And the fact that levelling the bed will align it with the mesh is moot, because you should've levelled the bed before probing it anyway, so they should inherently align. Sorry for being a bit of a downer, this just might be a bad use of your time.

And as is basically essential for UBL, if they read the documentation, they won't have any issues. Just the unfortunate effect of a lack of LCD space I imagine.

titanrw, 2023-03-02:
Just tested 2.1.2, with your PR merged (manually).  Everything seems good now.  Here's the console log:

```
Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Recv: ok P15 B3
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Recv: ok P15 B3
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.09  +0.17  +0.20  +0.14  +0.03  -0.11  -0.20
Recv:    |
Recv:  5 | -0.02  +0.02  +0.01  -0.01  -0.07  -0.15  -0.16
Recv:    |
Recv:  4 | -0.00  +0.02  -0.03  -0.04  -0.07  -0.10  -0.07
Recv:    |
Recv:  3 | -0.01  -0.03  -0.02 [-0.08] -0.04  -0.12  +0.05
Recv:    |
Recv:  2 | +0.00  +0.01  -0.10  -0.10  -0.07  -0.13  +0.04
Recv:    |
Recv:  1 | +0.01  -0.01  -0.10  -0.17  -0.15  -0.17  +0.01
Recv:    |
Recv:  0 | +0.00  -0.03  -0.09  -0.15  -0.16  -0.15  +0.01
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Send: G29 J V4
Recv: Tilting mesh (1/3)
Recv: Bed X: 50.00 Y: 50.00 Z: -0.09
Recv:                 Corrected_Z=-0.08
Recv: Tilting mesh (2/3)
Recv: Bed X: 310.00 Y: 50.00 Z: -0.16
Recv:                 Corrected_Z=-0.02
Recv: Tilting mesh (3/3)
Recv: Bed X: 180.00 Y: 310.00 Z: -0.00
Recv:                 Corrected_Z=-0.02
Recv:  T:21.54 /0.00 B:20.93 /0.00 @:0 B@:0
Recv: bed plane normal = [-0.0002288,-0.0001365,1.0000000]
Recv: X:225.00 Y:317.00 Z:10.00 E:0.00 Count X:18000 Y:25360 Z:8000
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,359)                                      (359,359)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.04  +0.13  +0.17  +0.13  +0.04  -0.10  -0.16
Recv:    |
Recv:  5 | -0.08  -0.03  -0.02 [-0.03] -0.07  -0.14  -0.14
Recv:    |
Recv:  4 | -0.07  -0.03  -0.07  -0.07  -0.09  -0.10  -0.05
Recv:    |
Recv:  3 | -0.09  -0.09  -0.07  -0.11  -0.06  -0.13  +0.06
Recv:    |
Recv:  2 | -0.08  -0.06  -0.16  -0.15  -0.10  -0.14  +0.04
Recv:    |
Recv:  1 | -0.08  -0.09  -0.16  -0.22  -0.19  -0.19  +0.00
Recv:    |
Recv:  0 | -0.10  -0.12  -0.16  -0.21  -0.20  -0.18  -0.01
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (359,  1)
Recv: 
Recv: ok P15 B3
```

Tilting point 2/3 at -0.02, even at X=310.

Considering this bug's been open since Aug last year, I'm glad it's fixed now.  Thought I'd be on 2.0.9.3 forever.

Now I can consider trying input shaping as that version of Marlin is usable now.

jamespearson04, 2023-03-02:
@titanrw Fantastic, glad I could be of help! And glad I didn't just opt to re-write my config for 2.0.9.3 lol.

Might be useful if you drop a comment on my PR just so the powers have some more affirmation: https://github.com/MarlinFirmware/Marlin/pull/25453

apulver, 2023-03-03:
> you need the tilted mesh in memory.
> If you try and seperate the plane from the surface, you just have to put them back together again anyway.

I'm not disputing that. However, I don't see that there's significant extra overhead here. The mesh would be loaded and transformed according to the model of the bed's planar surface. G29J does the same thing each time you run it, except in this case, it can assume that the mesh is already level (thus avoiding the call to get_z_correction). The object here though is to avoid having to run G29J in the first place as the vector normal to the bed's planar component should tend to be orthogonal with the printer's physical X and Y anyway due to tramming.  Modeling the planar component separately could also assist more accurate tramming. Whether it's worth the minor overhaul of UBL is certainly disputable and your point there is well-taken.

thisiskeithb, 2023-03-03:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25453

closed 2023-03-03 by thisiskeithb