MarlinFirmware/Marlin #24666: indianness choice in u8glib converter.html

toto-toa, 2022-08-22:
### Is your feature request related to a problem? Please describe.

I used your u8glib converter to generate bitmap (XBM files). The generated code used on Espressif devices (esp32/esp8266) has a  'indianness' inversion, certainly induced by the structure of the esp ¨?
 So it will be nice of you to add an option to choose beetween little or big-indian.
Thank you in advance.
The generated file:

[image]

..and the displayed one (normal and inverted)

[image]

### Describe the feature you want

choice of the indianness to generate the XBM file

toto-toa, 2022-08-22:
... and if I reconstruct the picture...

[image]

EvilGremlin, 2022-08-22:
Hello Columbus Junior!
Endian! E! from End!
Anyway, this is no concern for marlin, endianness is different everywhere and dealt with by platformio frameworks and display code, image array itself stay same for all platforms.

closed 2022-08-22 by thisiskeithb