MarlinFirmware/Marlin #24667: Need help with CONTROLLER_FAN_PIN

orrinwitt, 2022-08-23:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I mentioned this over in the Archim github [page,](https://github.com/ultimachine/Archim/issues/3) but it doesn't look very active, so I'm wondering if anyone here can shed some light.
I need to have a fan in my case to cool those stepper drivers, and would like for it to come on automatically with the CONTROLLER_FAN_PIN and M710 feature. 
I promise I've googled and also scoured over the schematic and pin files of my board and I'm just not having any luck.

Any help is greatly appreciated!

### Version of Marlin Firmware

2.1.1

### Printer model

Airwolf 3D XL

### Electronics

upgraded to Archim2 board

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[files.zip](https://github.com/MarlinFirmware/Marlin/files/9399358/files.zip)

ellensp, 2022-08-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-08-23 by ellensp

orrinwitt, 2022-08-24:
@ellensp Sorry for posting this with the wrong queue. It's definitely not a bug in Marlin, I just wasn't sure how else to get my question out there. 
The Archim github looks like it's been real dead for a few years. Where can I get support for this kind of thing? If I need to assign the pins manually, I'm open to that. But if not here, where should I post to find out how to do that?

Also, dealing with a dead github page makes me appreciate the steady work y'all do here even more. Thank you!

orrinwitt, 2022-08-24:
UPDATE: I worked smarter, not harder, and reassigned an extra heater pin from the mosfet rail to be the controller fan. Working like a champ now!