MarlinFirmware/Marlin #24671: [BUG] M118 Output to SERIAL_PORT_2

mulcmu, 2022-08-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

'M118` only outputs blank lines to secondary serial port.  Other text to secondary serial port is received okay such as temperature auto reports.

`M118 P0 TEST` and `M118 P2 TEST` only output blank lines to `SERIAL_PORT_2`

### Bug Timeline

Old issue

### Expected behavior

String provided to `M118` should be output to ports specified by `P` parameter.

### Actual behavior

`M118` output to secondary serial port is only a blank line for each 

### Steps to Reproduce

Send `M118P0TEST` or `M118P2TEST` in host software results in blank line output on secondary serial port.

Same behavior when above included in gcode printed from SD card.

### Version of Marlin Firmware

Foxies-CSTL/Marlin_2.1.x fork for FLSUN printers

### Printer model

FLSUN QQ-S Pro

### Electronics

Stock

### Add-ons

`SERIAL_PORT_2` is connected to an ESP8266 module installed in the motherboard.  ESP8266 is running ESPHOME

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

See https://github.com/Foxies-CSTL/Marlin_2.1.x for QQSP configuration.

I wasn't able to find root cause but tested a work around:

https://github.com/MarlinFirmware/Marlin/blob/e201f4e656f787666420cc157d5efd436c1bbb14/Marlin/src/gcode/host/M118.cpp#L59

was replaced with 
```
  SERIAL_ECHO(p);
  SERIAL_EOL();
```
and expected behavior was confirmed.  String was output on secondary serial port by `M118`

`SERIAL_ECHOLN(p)` seems to treat `p` as string pointer by the primary serial port but only as a char on the secondary serial port.  Bug might be more general that `SERIAL_ECHOLN(char *)` behaves differently based on port or underlying serial stream/device.

ellensp, 2022-08-24:
Provided config only has serial 0

ellensp, 2022-08-29:
I setup a ramps, enabled SERIAL_PORT_2 2  and hooked up a usb/serial adapter. Config is otherwise stock default config 

On this image Top serial console is on ramps serial port 2, bottom  serial port it on ramps serial port 0

[image]

```
Results.
Test   |  from | gcode String     | result
1      |   0   | M118 P2 TESTING  |  OK
2      |   0   | M118 P2 TESTING2 |  OK
3      |   2   | M119 P0 TESTING3 |  FAILED
4      |   2   | M119 P0 TESTING4 |  FAILED
```
3 and 4 failed as it also send test string to serial 2

I also saw something similar to reported issue.  But cannot reliably reproduce it.

ellensp, 2022-08-29:
Nope I need to RTFM!

[Pn<0\|1\|2>] | Send message to host serial port (1-9).Pn0: Send message to all ports.Pn1: Send message to main host serial port.Pn2: Send message to secondary host serial port. Requires SERIAL_PORT_2.

mulcmu, 2022-08-29:
@ellensp Are you able to test the ramps board setup with SERIAL_PORT_2 using SoftwareSerial instead of using a hardware UART?

@Foxies-CSTL Did you have any insight into why M118 to SERIAL_PORT_2 on QQSP has this behavior?  Could you test M118P2TEST on one of your machines?

ellensp, 2022-08-29:
@mulcmu M118 only supports hardware uarts

ellensp, 2022-08-29:
Version of Marlin Firmware
Foxies-CSTL/Marlin_2.1.x fork for FLSUN printers

This is a fork of marlin, you need to submit bug reports with the forks maintainer

closed 2022-08-29 by ellensp

Foxies-CSTL, 2022-08-31:
> @Foxies-CSTL Did you have any insight into why M118 to SERIAL_PORT_2 on QQSP has this behavior? Could you test M118P2TEST on one of your machines?

You need to define with " SERIAL_PORT_2 1" the communication channel. So add  "#define NUM_SERIAL 2" and it should work.

Don't forget that the Mks firmware of the Wifi module only works with the Mks UI (TFT_LVGL_UI) because of the resolution of the Stock screen (320x240).

mulcmu, 2022-08-31:
@Foxies-CSTL Thanks

I have `MOD_WIFI` defined in FLSUN_config, which defines `MOD_AUX`.  This then picks up the 
```
      #define SERIAL_PORT_2 1
      #define NUM_SERIAL 2
```
in Configuration.h

The firmware of the Wifi module has been flashed with my [ESPHome](https://github.com/mulcmu/esphome-marlin-uart) based uart interface.  The wifi module rx/tx works as expected except for the issue reported with `M118`.  My troubleshooting identified something fundamentally different between output from `SERIAL_ECHOLN(char *);` vs the workaround I implemented using `SERIAL_ECHO(char *); SERIAL_ECHO_EOL();` on the second QQSP serial port.

Foxies-CSTL, 2022-08-31:
> The firmware of the Wifi module has been flashed with my [ESPHome](https://github.com/mulcmu/esphome-marlin-uart) based uart interface. The wifi module rx/tx works as expected except for the issue [...]

So it might be necessary to define the two pins that Mks and ESP3D v3.0 use for data transfer and acquisition. To do this, we must modify the QQSP pins file by declaring:
  // ESP-WIFI module
  #define WIFI_IO0_PIN PA8 // MKS ESP WIFI IO0 PIN
  #define WIFI_IO1_PIN PC7 // MKS ESP WIFI IO1 PIN
  #define WIFI_RESET_PIN PA5 // MKS ESP WIFI RESET PIN
  
  Open the problem on my repository because I think it does not concern Marlin official. Maybe later ;-)

thisiskeithb, 2022-08-31:
> Open the problem on my repository because I think it does not concern Marlin official

Yes, please open an issue in [Foxies-CSTL’s](https://github.com/Foxies-CSTL/Marlin_2.1.x/issues) repo

mulcmu, 2022-08-31:
https://github.com/Foxies-CSTL/Marlin_2.1.x/issues/6 has been opened for this issue.