MarlinFirmware/Marlin #24676: [BUG] Pause_Park_Nozzle_Timeout not working

ePiiKDiiSasTeR, 2022-08-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have enabled PAUSE_PARK_NOZZLE_TIMEOUT to turn off the nozzle on change filament mid print if I'm not near the print. After the time limit, the screen changes to the message "Press button to heat nozzle" but temperature does not decrease over time. If I click the button the change filament continues with the process of loading, because the nozzle is still on temperature.

I tried latest bugfix, and 2.0.9.5 version and the problem still occurs. Otherwise on 2.0.9.3 it works great.

### Bug Timeline

2.0.9.5

### Expected behavior

I expect the nozzle to turn off for safety.

### Actual behavior

It does not turn off the nozzle

### Steps to Reproduce

1. Change Filament mid print.
2. Wait for park and filament to unload.
3. Wait to test if the nozzle turn off after the time specified on PAUSE_PARK_NOZZLE_TIMEOUT

### Version of Marlin Firmware

2.1.1

### Printer model

Creality Ender 3 Pro

### Electronics

Stock electronics

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/9419977/Configuration.Files.zip)

https://user-images.githubusercontent.com/19728634/186540386-31fe646b-83ea-4d00-915d-56098f3226a8.mp4

GMagician, 2022-08-25:
@tombrazier MPC seems has a issue with it:
`float pid_output = power * 254.0f / constants.heater_power + 1.0f;`
since pid_output =1 when power=0 due to idle time

While standard PID seems missing this check
you may change:
`const float pid_output = hotend_pid[ee].get_pid_output();`
to:
`const float pid_output = TERN1(HEATER_IDLE_HANDLER, !heater_idle[ee].timed_out) ? hotend_pid[ee].get_pid_output() : 0;`
in temperature.cpp file

paulluby, 2022-08-25:
I'm having exactly the same problem on two printers.

Hotend stays on, 1st indicated temp is solid and 2nd indicated temp is flashing, both at same value.

As I said two printers, one has BTT SKR V1.3 Board and other has BTT SKR V1.1 Pro board.

Okay with Marlin 2.1 and below, issue only happens for me with Marlin 2.1.1 and Bugfix 2.1.x

tombrazier, 2022-08-25:
Looks like this logic changed for PID in #24389.

MPC is fine. The value returned by `get_pid_output_hotend()` get bit shifted to the left. So the value of 1 goes to zero. It's there to make sure that rounding works correctly.

paulluby, 2022-08-25:
> Looks like this logic changed for PID in #24389.
> 
> MPC is fine. The value returned by `get_pid_output_hotend()` gets bit shifted to the left. So the value of 1 goes to zero. It's there to make sure that rounding works correctly.

So take it its not a quick fix?

GMagician, 2022-08-25:
@paulluby are you using MPC or classic PID?

GMagician, 2022-08-25:
@paulluby try PR proposed solution if you can

paulluby, 2022-08-25:
> @paulluby try PR proposed solution if you can

PR?

paulluby, 2022-08-25:
> Yep you see link above (draft). You may copy temperature.cpp in your marlin folder and recompile
> 
> edit: [temperature.zip](https://github.com/MarlinFirmware/Marlin/files/9426433/temperature.zip)
> 
> copy to marlin/src/module

Can do, BTW if you mean-

"
While standard PID seems missing this check
you may change:
const float pid_output = hotend_pid[ee].get_pid_output();
to:
const float pid_output = TERN1(HEATER_IDLE_HANDLER, !heater_idle[ee].timed_out) ? hotend_pid[ee].get_pid_output() : 0;
in temperature.cpp file
"
Which you posted a couple of hours ago, done that, compiled it and its printing a small block now with a colour change halfway through :)

I'll have a look through the zip file and if its not that I'll stop print, insert and recompile accordingly :)

tombrazier, 2022-08-25:
@paulluby A "PR" is a "pull request" which is a github way of saying "I have made some changes to this project and would like you to include them". @GMagician is asking for his fix to be added to Marlin. When he created the PR it caused an automatic post above saying "GMagician mentioned this issue 1 hour ago". Below that is a link to the PR, which has the changes. I hope that clarifies some of the language above. Took me some time to understand it all when I started with github.

GMagician, 2022-08-25:
> @paulluby A "PR" is a "pull request" which is a github way of saying "I have made some changes to this project and would like you to include them". @GMagician is asking for his fix to be added to [...]

Thanks, I often give this as standard understood but that's not true. My fault

paulluby, 2022-08-25:
Cheers, sort of figured it out after a while.

Its on its second attempt compile, have cleared it for a retry, loads of errors listed at last go.

GMagician, 2022-08-25:
@paulluby my source file (temperature.cpp) if for latest 2.1 bugfix. It may give errors if you are using an old marlin version, even a 2.1 release version

paulluby, 2022-08-25:
The compile has fallen down again

paulluby, 2022-08-25:
> @paulluby my source file (temperature.cpp) is for latest 2.1 bugfix. It may give errors if you are using an old marlin version, even a 2.1 release version

Okay no probs, I have a bugfix from last night ready to go will insert a give it a go :)

paulluby, 2022-08-25:
Okay, first multiple compile fails so I'm gonna start with a fresh Bugfix download.

Second, the following that you mentioned earlier-

"
While standard PID seems missing this check
you may change:
const float pid_output = hotend_pid[ee].get_pid_output();
to:
const float pid_output = TERN1(HEATER_IDLE_HANDLER, !heater_idle[ee].timed_out) ? hotend_pid[ee].get_pid_output() : 0;
in temperature.cpp file
"

Seems to have worked on my printer that has a BTT SKR V1.3 board. I take it this just re-queries whether the advanced pause heater requirement has timed out and if so sets heater off?

Can you let me know if this will cause other issues and I'll go back to 2.1

In meantime I'll do new bugfix and insert your replacement temperature file.

GMagician, 2022-08-25:
The fix I've done pratically does the same thing in a different way and it should affect only heaters when they are idle.
Remember that bugfix is always changing and I can't assure that new bugs are not added to the fresh download

paulluby, 2022-08-25:
> `const float pid_output = hotend_pid[ee].get_pid_output();`
> to:
> `const float pid_output = TERN1(HEATER_IDLE_HANDLER, !heater_idle[ee].timed_out) ? hotend_pid[ee].get_pid_output() : 0;`
> in temperature.cpp file

Okay here's some test results.

Note I printed a 20 x 20 x 20mm cube with a filament change at z=10mm for each test.

On both my printers with Marlin 2.1.1 installed and the following change carried out in the temperature.cpp file;

Replace;

const float pid_output = hotend_pid[ee].get_pid_output();

with

const float pid_output = TERN1(HEATER_IDLE_HANDLER, !heater_idle[ee].timed_out) ? hotend_pid[ee].get_pid_output() : 0;

the temperature indication and actual hot-end temperature drops accordingly during a filament change and goes up to temp when selected, all indications are as seen by me in previous versions.

On both printers with Bugfix Marlin 2.1.x installed and using the temperature.cpp file you forwarded to me the temperature indication and actual hot-end temperature drops accordingly during a filament change and it goes up to temp when selected, all indications are as seen by me in versions up to and including Marlin 2.1

Hopes this helps in some way.

Thanks for sorting.

Paul.

GMagician, 2022-08-25:
@paulluby thanks for testing. It works to me it's enough

GMagician, 2022-09-10:
Merged, may be closed

closed 2022-09-10 by ellensp