MarlinFirmware/Marlin #24677: [BUG] BTT MMU DIP not connecting to SKR Mini E3 v3

pennypacker007, 2022-08-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Main board is SKR Mini E3 v3 trying to connect to MMU DIP via TFT port. Connection below

SKR me3v3         MMU DIP
TFT Pins                Pins
+5v--------------------+5v
Gnd-------------------Gnd
Tx----------------------Rx
Rx----------------------Tx 
Reset------------------Reset

Port settings
Marlin 2.1.x Bugfix 
Configuration.h

#define SERIAL_PORT -1
#define BAUDRATE 115200
//#define SERIAL_PORT_2

Configuration.adv.h
MMU2_SERIAL_PORT 2
#define MMU2_DEBUG

The above setup sends Marlin into a boot loop. Serial monitor returns  attached error. Any help would be appreciated.

[image]

 

### Bug Timeline

New issue after setup and install of MMU2S with BBT MMU DIP board

### Expected behavior

I expected the MMU to connect with the printers board 

### Actual behavior

Communication between the two boards fails. 

### Steps to Reproduce

Restart printer 

### Version of Marlin Firmware

2.1.x Bugfix

### Printer model

AM8

### Electronics

BTT SKR MINI E3 V3, MMU DIP

### Add-ons

MMU2S

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-08-25:
looks more like you have an issue with the mmu pinda probe since it crashed when you ask it to read the pinda state

pennypacker007, 2022-08-25:
It crashes at different points in the debug not always at P0

ellensp, 2022-08-25:
Using external power?  Usb is not sufficient

pennypacker007, 2022-08-25:
All external power

GMagician, 2022-08-25:
I checked code, for specified event, there is nothing evidentely wrong in code. Code is in 'case -4' waiting for an answer.

GMagician, 2022-08-25:
P0 command is sent also time to time then it may be a canditate for the cause.
What is see, maybe not correct, is `sscanf(rx_buffer, "%hhuok\n", &finda);`. finda is int8 and %hhu is uint8 but not sure it maybe the cause

GMagician, 2022-08-25:
You may add a debug print in case -4, before sscanf and print rx_buffer. Just to see if it enters here or not

pennypacker007, 2022-08-25:
Thank you I will try that and post the result

GMagician, 2022-08-27:
oh, add the same "debug print" in case 2. This is when P0 is sent periodically

pennypacker007, 2022-08-28:
[image]

GMagician, 2022-08-28:
is this message ("test output") what you get in rx_buffer?

GMagician, 2022-08-28:
and "'P0' complete" where does it come from?

pennypacker007, 2022-08-28:
[image]

pennypacker007, 2022-08-28:
[image]

GMagician, 2022-08-28:
That's not what I meant when I said to print out rx_buffer and also, in case - 4, inside Else you do an unnecessaru sscanf.
what I was thinking something like:
```c++
      if (rx_ok()) {
        DEBUG_ECHOLNPGM ("answer=>", rx_buffer, "<");
        sscanf(rx_buffer, "%hhuok\n", &finda);
```
or similar in output (if this doesn't compile)

From your code it seems that it hang while waiting for "ok" answer, or because of sscanf

pennypacker007, 2022-08-29:
[image]

pennypacker007, 2022-08-29:
Thanks I tried your code and got the above response. Marlin then reboots continuously.

GMagician, 2022-08-29:
Since I don't see message:
"MMU 1"
"MMU - ENABLED"
it seems that sscanf or the debug print itself is the issue.
I highly suspect sscanf but you may continue to move the 'rx_buffer print' down one line each time to see.
Maybe sscanf doesn't handle correctly %hhu but in such case this is out of Marlin team control, maybe %hhi will behave correctly (and since variable is signed is a more sensed thing to do despite of result you get)

pennypacker007, 2022-09-05:
hi, have tried %hhi and same result

pennypacker007, 2022-09-05:
below is the output from pronterface:

Connecting...
echo:V86 stored settings retrieved (748 bytes; crc 9557)
//action:notification Stored settings retrieved
//action:prompt_end
MMU <= reset
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
echo:SD card ok
Printer is now online.
MMU => 'start'
MMU <= 'S1'
MMU => 107
MMU <= 'S2'
MMU => 134
MMU <= 'P0'
answer=>0ok
<
Exception in thread read thread:
Traceback (most recent call last):
  File "printrun\printcore.py", line 333, in _readline
  File "printrun\printcore.py", line 329, in _readline_nb
  File "serial\serialwin32.py", line 275, in read
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

GMagician, 2022-09-05:
What you get here is an PC error, not a Marlin fault.
Sure system you are using doesn't send a M112?

ellensp, 2022-09-05:
Googling the error shows "The error you see happens when the device on the other side of the connection resets or closes the connection for any other reason."

Yes its a system error, to the serial port abruptly vanishing 

Here is where is gets "interesting" 

The MMU is is connected the the controllers serial UART 2
Pronterface is connected to the controllers CDC USB serial port, this is the one that is vanishing abruptly.

Which mean hard crash or power issue.

Do you have a lcd?  (no config, so cannot check myself) 

Does the Controller reboot or just stop talking? (if you have a lcd you would see the boot screen again)

GMagician, 2022-09-05:
@ellensp I suspect he has a lcd due to his statement:

> The above setup sends Marlin into a boot loop

What I don't understandi is that a message before sscanf is printed while the one after isn't. Does Marlin reboot because of sscanf?
I read somewhere that %hhu may write to a 16 bits integer (https://stackoverflow.com/questions/15825254/why-is-scanfhhu-char-overwriting-other-variables-when-they-are-local). If this is true then reboot may occurs because of memory corruption

GMagician, 2022-09-05:
I checked and C89 is really old and we are using, as older, c++11 that is year after C89

pennypacker007, 2022-09-05:
Here are the config files. Yes @GMagician, I have an LCD connected and the machine reboots continuously which is why the serial connection disappears 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9491766/Configuration.zip)

GMagician, 2022-09-06:
I don't know where to look at. All seems bring to sscanf, but no real evidence, just suspects. To remove any doubt you may replace sscanf, where `finda` variable is involved, with `finda = true;`. You'll have issues using it (filament always present) but if it stops to reboot we know who is guilty.

pennypacker007, 2022-09-07:
@GMagician thank you, I have tried the above, commented sscanf out and  finda = true; now receive the MMU - Enabled message and marlin then reboots which again drops the serial connection. 
[image]

GMagician, 2022-09-07:
Then sscanf is not guilty. No idea how to find where/why marlin hangs, sorry

closed 2022-11-18 by github-actions[bot]