MarlinFirmware/Marlin #24686: [BUG] miss_spelled

pedrobecerra, 2022-08-25:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Marlin-2.0.x
In Conditionals_post.h
Line 2788
there is:   #if PIN_EXISTS(CONTROLLER_FAN_PIN)
must be:  #if PIN_EXISTS(CONTROLLER_FAN)

### Bug Timeline

It's a BUG

### Expected behavior

I expected to compile with no errors.

### Actual behavior

It do not compile.

### Steps to Reproduce

1. In Configuration_adv.h     line 502      uncomment #define USE_CONTROLLER_FAN
2. In line 504  type any valid pin.  Ex: FAN2_PIN
3. Build the project.
4. You'll get an error:
c:\marlin-2.0.x\marlin\src\inc\SanityCheck.h:2102:6: error: #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

### Version of Marlin Firmware

2.0.9.5

### Printer model

Creality Ender 3

### Electronics

BTT SKR 1.4 turbo

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GMagician, 2022-08-25:
bugfix 2.1 seems has fixed it

pedrobecerra, 2022-08-25:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9428170/Configuration.zip)

closed 2022-08-25 by thisiskeithb