MarlinFirmware/Marlin #24687: [BUG] (bug summary)  #define ADAPTIVE_FAN_SLOWING    leads to Marlin\src\module\temperature.cpp:298:8: error: 'bool Temperature::adaptive_fan_slowing' is not a static data member of 'class Temperature'

Quantenkosmos, 2022-08-26:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Just activated #define ADAPTIVE_FAN_SLOWING
CONFIGURATION_ADV_H_VERSION 02010100 (from 8.8.2022)

( #define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO )

compiles with 

```
Compiling .pio\build\LPC1769\src\src\sd\SdVolume.cpp.o
Compiling .pio\build\LPC1769\src\src\sd\cardreader.cpp.o
Marlin\src\module\temperature.cpp:298:8: error: 'bool Temperature::adaptive_fan_slowing' is not a static data member of 'class Temperature'
  298 |   bool Temperature::adaptive_fan_slowing = true;
      |        ^~~~~~~~~~~
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9430846/Configuration.zip)

Marlin\src\module\temperature.cpp: In member function 'void Temperature::tr_state_machine_t::run(const_celsius_float_t, const_celsius_float_t, heater_id_t, uint16_t, celsius_t)':
Marlin\src\module\temperature.cpp:2818:15: error: 'adaptive_fan_slowing' was not declared in this scope
 2818 |           if (adaptive_fan_slowing && heater_id >= 0) {
      |               ^~~~~~~~~~~~~~~~~~~~
*** [.pio\build\LPC1769\src\src\module\temperature.cpp.o] Error 1
```

### Bug Timeline

26.08.2022

### Expected behavior

I expected normal compile

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.1.1 (Aug 26 2022 00:09:00) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D

### Printer model

Prusa like

### Electronics

BOARD_BTT_SKR_V1_4_TURBO 

### Add-ons

BLTOUCH

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-08-26:
no config files, cannot reproduce on bugfix

Quantenkosmos, 2022-08-26:
Sorry, I was sure it was uploaded, here it comes
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9431130/Configuration.zip)

ellensp, 2022-08-26:
you have PIDTEMP diabled

but you deliberatly enabled defines that require PIDTEMP
ie 
```CPP
//#if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
    #define NO_FAN_SLOWING_IN_PID_TUNING    // Don't slow fan speed during M303 //MD20220826 - sonst andere Variable ned definiert... issn bug
  //#endif

```
This is not a bug, this is malicious deliberate sabotage

ellensp, 2022-08-26:
ADAPTIVE_FAN_SLOWING + MPCTEMP is not valid either, but marlin should do something about that

Quantenkosmos, 2022-08-26:
> 

You're right, i tried that to circumvent the problem. Even with activated 
`#if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
    #define NO_FAN_SLOWING_IN_PID_TUNING    // Don't slow fan speed during M303
  #endif`
it's not compiling

Quantenkosmos, 2022-08-26:
> ADAPTIVE_FAN_SLOWING + MPCTEMP is not valid either, but marlin should do something about that

Ok, I understand, I wasn't aware of that. I changed from PID to MPCTEMP, and it worked perfect! 
Actually I had a thermal runaway (in bed!), so I saw that possibility, and wondered why it didn't compile.

Also with MPCTEMP I expected lowered fan speed {later lowered printing speed??} when the temperature drops.

closed 2022-11-06 by github-actions[bot]