MarlinFirmware/Marlin #24688: [FR] ADAPTIVE_BED_FAN_SLOWING (like ADAPTIVE_FAN_SLOWING for hotend) for falling BED temperature (when fans cools the bed unwantededly)

Quantenkosmos, 2022-08-26:
### Is your feature request related to a problem? Please describe.

When I installed a new hotend- cooler, the fan blew so strong (ind the false direction) that the bed temperature fell down about 10°C, so thermal runaway was triggered (and the print was ruined).

Looking for a solution I found that `ADAPTIVE_FAN_SLOWING`
but didn't work with `#define MPCTEMP`.
That wouldn't slove MY problem with the bed, buit is the initial idea of that.

Of course mechanical solution is the first obvious solution, but maybe we can help with software here also.

### Describe the feature you want

`ADAPTIVE_BED_FAN_SLOWING`, similar to `ADAPTIVE_FAN_SLOWING`, but referred to bed temperature runaway.

we could also introduce a parameter to
* calibrating fan temperature - influence to bed (and hotend) 
- let it run at various speeds, measure temp drop / PID values, 
- store in EEPROM,
- future future: calculate allowed max speed of fan and later printing speed in respect to that calibration, fan position (over printed bed areas, over clean bed area) and so on

classicrocker883, 2022-08-31:
i think the easiest solution is to have a chamber. or what about instead of slowing down the fan because thats essential to part cooling, is to have the bed raise in temp?

too bad beds dont have segments that can be heated individually

EvilGremlin, 2022-08-31:
PID already doing this, problem is weak power/heater and wrong fan duct design, and these problems should be always solved in hardware.

InsanityAutomation, 2022-08-31:
Hotend made sense because it allows for over powered blowers to be used for bridging, and keeping the system under control otherwise. Bed should have a massive thermal capacity and if the part cooler is lowering it.... There is likely a bigger problem.

Chamber fan and chamber temp I can see it applying if they're trying to vent / exhaust too much and heaters don't have the wattage to keep up.

GMagician, 2022-08-31:
> Bed should have a massive thermal capacity and if the part cooler is lowering it.... 

I'll be not so sure. I still have some issues with it when fan blows just over the bed where thermistor resides.
My CNF is a DIY Bear (Prusa Mk3S like printer).
Bed changes from 70 to 66/67 (Fan 60%) and triggered thermal error. I solved by playing with fan speed with repetier until everithing stable. Not the first time I face it

Quantenkosmos, 2022-08-31:
My settings are even worse (cheap bed, but when uprgading then a 400mm x 400mm least, so I'll stay with it for a while longer).
Because I did everything myself I have to design a new fan duct, you're right. I also added a block in the fan duct, so it helped a little, you know, software is always there to repair the hardware problems ;-).

```
/**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD        180  // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS      15  // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD                180 // Seconds
  #define WATCH_BED_TEMP_INCREASE               1 // Degrees Celsius
#endif
```