MarlinFirmware/Marlin #24689: Menu Back Items missing on direct-connected "Ultipanel" LCDs (i.e CR10_STOCKDISPLAY) since #24618

oliver-eifler, 2022-08-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After updating to marlin bugfix2.1.x (Aug 26, 2022, including "Fix http://" links commit) I've realized that all back menu items are gone on my CR10_STOCKDISPLAY

### Bug Timeline

it start's with #24618

### Expected behavior

back menu item should be displayed when entering on of the sub menus (i.e Move Axis) on CR10_STOCKDISPLAY and any other simple LCD Displays (graphical and text)

### Actual behavior

No back menu item is shown in any submenu on my CR10_STOCKDISPLAY

### Steps to Reproduce

in configuration.h
#define CR10_STOCKDISPLAY 
(or any other LCD Display)
enter 'Move Axis' Menu -> No back menu item shown

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Aug 26 2022 13:08:15) 

### Printer model

Ender 3 Pro and others

### Electronics

BTT SKR E3 Turbo

### Add-ons

bltouch, independent dual z axis

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

in marlin/src/inc/Conditionals_post.h
changing line 3772
from
```
#if DISABLED(DISABLE_ENCODER) && ANY(HAS_ROTARY_ENCODER, HAS_ADC_BUTTONS) && ANY(TFT_CLASSIC_UI, TFT_COLOR_UI)
  #define HAS_BACK_ITEM 1
#endif
```
to
```
#if DISABLED(DISABLE_ENCODER) && ANY(HAS_ROTARY_ENCODER, HAS_ADC_BUTTONS) && ANY(HAS_MARLINUI_MENU, TFT_CLASSIC_UI, TFT_COLOR_UI)
  #define HAS_BACK_ITEM 1
#endif
```
seemed to fix the issue

oliver-eifler, 2022-08-26:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9433472/configuration.zip)

EvilGremlin, 2022-08-26:
Ooops, that'll be on me, was too focused on TFT. Wiill throw fix PR ASAP.

oliver-eifler, 2022-08-26:
That was fast ;)

thisiskeithb, 2022-08-26:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24690

closed 2022-08-26 by thisiskeithb

thisiskeithb, 2022-08-27:
New PR with fix: https://github.com/MarlinFirmware/Marlin/pull/24694