MarlinFirmware/Marlin #24692: [BUG] (bug summary)

lukasloetkolben, 2022-08-26:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I am trying to configure my Octopus V1.1 with two independent Z-Axis Motors and two Endstops.
I defined these endstops:
```
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
#define USE_IMAX_PLUG
```
Drivers:
```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
```

And enabled Multi Endstops:
```
#ifdef Z2_DRIVER_TYPE
  #define Z_MULTI_ENDSTOPS          // Other Z axes have their own endstops
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP   _IMAX_   // Z2 endstop board plug. Don't forget to enable USE_*_PLUG.
    #define Z2_ENDSTOP_ADJUSTMENT 0   // Z2 offset relative to Y endstop
  #endif 
  [...]
```
  When i am compiling like this M119 does not show the z2_max status.
  If i change `#define Z2_USE_ENDSTOP   _IMAX_` to -> `#define Z2_USE_ENDSTOP   _XMAX_` 
  It does show z2_max with M119.
  
  Can anyone help me with this issue?

  

### Version of Marlin Firmware

2.1.1

### Printer model

Qidi X-Plus

### Electronics

BTT OCTOPUS V1.1

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/9436004/config.zip)

GMagician, 2022-08-26:
why do you have USE_IMAX_PLUG?

GMagician, 2022-08-26:
What pin do you expect is connected to IMAX? I can't see it in pins definition, but I'm not used with such board...

GMagician, 2022-08-26:
I can't see any I_MAX_PIN defined in .h

lukasloetkolben, 2022-08-26:
i have connected the Z2 Endstop to Z2_DIAG_PIN (PG11)
What does `#define Z2_USE_ENDSTOP   _IMAX_` actually mean?
I can only find _IMAX_ in macros.h `#define _IMAX_   0x24`, i have not quite understood how this works.
I am only surprised that its woking with _XMAX_, but then X homing isn't working anymore.

thisiskeithb, 2022-08-26:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-08-26 by thisiskeithb

lukasloetkolben, 2022-08-26:
Isn't it a bug?

GMagician, 2022-08-27:
@lukasloetkolben code does something like in code:
```c++
#if Z2_USE_ENDSTOP == _XMIN_
  #define Z2_MAX_PIN X_MIN_PIN
```
then defines pin used as endstop. Code supports only X, Y, Z MIN/MAX then _IMAX_ should say to use I axis MAX endstop, but this isn't supported by code.

lukasloetkolben, 2022-08-27:
Thank you. 
I noticed the same yesterday. I am using Y-MIN Plug now, but it doesn't make sense to me.

GMagician, 2022-08-27:
Maybe defining: 
`#define Z2_USE_ENDSTOP   _E?DIAG_   // Z2 endstop board plug. Don't forget to enable USE_*_PLUG.`

and replace ? with a number from 0 to 3 may solve (and you have to connect to PG12 to 15)

GMagician, 2022-08-27:
or even you may try to add:
`#define Z2_MAX_PIN Z2_DIAG_PIN`
commenting Z2_USE_ENDSTOP