MarlinFirmware/Marlin #24697: [BUG] platforms.h HAL_STM32 compile error (FIX)

classicrocker883, 2022-08-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

as you can see this does not work

platforms.h line 40 
```
#elif defined(ARDUINO_ARCH_STM32)
  #ifndef HAL_STM32
    #define HAL_STM32
  #endif
  #define HAL_PATH(PATH, NAME) XSTR(PATH/STM32/NAME)
```

### instead it should be written like this
```
#elif defined(ARDUINO_ARCH_STM32) || !defined(HAL_STM32)
  #define HAL_STM32
  #define HAL_PATH(PATH, NAME) XSTR(PATH/STM32/NAME)
```

### Version of Marlin Firmware

bugfix 2.1

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-08-27:
What is the error and how do we reproduce it?

classicrocker883, 2022-08-27:
when using PROUI it doesnt really give an error. but it shows problems as though it isnt defined.
this helps keep the reported problems down. at least on my end

sjasonsmith, 2022-09-29:
I'm also not understanding how this is misbehaving. If `HAL_PATH` wasn't being defined you should have hit the `#error` case at the bottom of the file. It shouldn't be silently failing.

@classicrocker883 for anything to happen with this you will need to provide your configuration headers which are being impacted by this.

sjasonsmith, 2022-10-17:
Closing, since we can't really tell what the problem is and OP is unresponsive.

closed 2022-10-17 by sjasonsmith