MarlinFirmware/Marlin #24699: [BUG] (ESP3D Web UI not working with Marlin)

UmerSoftwares, 2022-08-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am not sure if I should post this issue on Marlin github or ESP3D github.

Basically, I have made my own DIY PCB for 3D Printer board based on ESP32 DEVKIT V1 that uses the ESP-WROOM-32 chip with 4MB flash. All the pin connections are the same as [FYSTEC E4](https://github.com/FYSETC/FYSETC-E4) board.

I used the latest available version of Marlin Bugfix. The firmware is compiled and uploaded. It connected to my WiFi and I opened the ESP32 WebUI in the browser using the IP address. Now, I had to upload the index.html.gz file. Since I could not find it in the Marlin code, I used the one from ESP3D here [https://github.com/luc-github/ESP3D/blob/2.1.x/esp3d/data/index.html.gz](https://github.com/luc-github/ESP3D/blob/2.1.x/esp3d/data/index.html.gz). 
Now, in the WebUI, I can see the SDCARD files. But other controls including the terminal and the movement controls do not work.

### Bug Timeline

New. Posted on 27-08-2022

### Expected behavior

I expected all controls from the WebUI would work.

### Actual behavior

Following are not working:

- Movement Controls
- Temperature Control
- Extruder Control
- Command Terminal
- The printer configuration is showing nothing

### Steps to Reproduce

Download Marlin code and open in VSCODE using the Platform IO extension.

In platformio.ini, I changed the default_envs to FYSTEC_E4
[image]

In the ini/esp32.ini folder, I modified the board_build.partitions option in env:FYSETC_E4 and commented the default_16MB.csv because my ESP32 is 4MB flash and not 16MB
[image]

In Configuration.h, I define the motherboard
`#define MOTHERBOARD BOARD_FYSETC_E4`

As well I set the bed thermistor and set the baudrate to 115200

In the Configuration_adv.h, I set the WiFi Support

```
/**
 * WiFi Support (Espressif ESP32 WiFi)
 */
//#define WIFISUPPORT         // Marlin embedded WiFi managenent
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host

  /**
   * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
   * the following defines, customized for your network. This specific file is excluded via
   * .gitignore to prevent it from accidentally leaking to the public.
   *
   *   #define WIFI_SSID "WiFi SSID"
   *   #define WIFI_PWD  "WiFi Password"
   */

  #define WIFI_SSID "My WiFi"
  #define WIFI_PWD  "My WiFi Password"
  //#include "Configuration_Secure.h" // External file with WiFi SSID / Password
#endif
```

After Marlin compiles and uploads, I open its IP address in my browser and upload the ESP3D files present here [https://github.com/luc-github/ESP3D/tree/2.1.x/esp3d/data](https://github.com/luc-github/ESP3D/tree/2.1.x/esp3d/data)

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Aug 27 2022 15:42:25) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:FYSETC_E4 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Ender 3 Pro

### Electronics

Custom Board bases in FYSTEC E4 design

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9437843/Configuration.zip)

ellensp, 2022-08-27:
ESP3D_WIFISUPPORT is when you are running marlin on a esp32!!!

for a esp32 plugged into another board you enable a the serial port it is plugged into!

ellensp, 2022-08-27:
never mind.. didnt realise [FYSTEC E4](https://github.com/FYSETC/FYSETC-E4) is a esp32

EvilGremlin, 2022-08-27:
read Luc's notes: warningCurrently the development version need special version of Marlin which may not sync with latest version of Marlin yet : https://github.com/luc-github/Marlin/tree/ESP3DLibV3.0

luc-github, 2022-08-27:
you need to uncomment `#define SERIAL_PORT_2 -1` to let the communication working

UmerSoftwares, 2022-08-28:
Thanks for help. Uncommenting `#define SERIAL_PORT_2 -1` solved the issue.

You solved the last issue I was having with my custom board. This is going to be the first 3D Printer control board made in my country.
[image]
[image]

closed 2022-08-28 by UmerSoftwares