MarlinFirmware/Marlin #24702: [BUG] TOUCH_IDLE_SLEEP_MINS does not work with TFT_COLOR_UI

spasecookee, 2022-08-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In Marlin 2.0.9 it is possible to use TFT_COLOR_UI, with a generic TFT (FSMC), TOUCH_SCREEN and a value for TOUCH_IDLE_SLEEP. In 2.1.1, this combination causes an error when Linking: 

Marlin\src\lcd/marlinui.cpp:1183: undefined reference to `MarlinUI::sleep_display(bool)'  

If TFT_CLASSIC_UI is used, there is no linking error, but it should also work with TFT_COLOR_UI.

### Bug Timeline

This is a newer issue

### Expected behavior

I expect linking should succeed.

### Actual behavior

Linking fails with Marlin\src\lcd/marlinui.cpp:1183: undefined reference to `MarlinUI::sleep_display(bool)'  

### Steps to Reproduce

In Configuration.h, set BOARD_CHITU3D_V6
Set SERIAL_PORT 1
Set TFT_GENERIC
Set TFT_DRIVER ILI9488
Set TFT_INTERFACE_FSMC
Set TFT_RES_480x320
Set TFT_COLOR_UI
Set TOUCH_SCREEN
Set TOUCH_IDLE_SLEEP_MINS 5
[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/9438642/MarlinConfig.zip)

### Version of Marlin Firmware

2.1.1

### Printer model

nj

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2022-08-31:
try enabling the `LCD_BACKLIGHT_TIMEOUT`
in settings.cpp
under   _TERN_(HAS_LCD_BRIGHTNESS, ui.refresh_brightness());

add ` TERN_(LCD_BACKLIGHT_TIMEOUT, ui.refresh_backlight_timeout());`

in dwin.cpp (ProUI)
add to advancedsettings menu
` #if LCD_BACKLIGHT_TIMEOUT
      EDIT_ITEM(ICON_Box, MSG_LCD_TIMEOUT_SEC, onDrawPIntMenu, SetTimer, &ui.lcd_backlight_timeout); //added
    #endif`  
then add the function
```
  #if LCD_BACKLIGHT_TIMEOUT
  void SetTimer()   { SetPIntOnClick(LCD_BKL_TIMEOUT_MIN * 10, LCD_BKL_TIMEOUT_MAX);
    DWIN_UpdateLCD();
   }
  #endif
```

under HMI_Init()
add
```
#if LCD_BACKLIGHT_TIMEOUT
      ui.refresh_backlight_timeout();
    #endif  
```

and under EachMomentUpdate()
```
 #if LCD_BACKLIGHT_TIMEOUT
    if (ui.backlight_off_ms && ELAPSED(ms, ui.backlight_off_ms)) {
      TurnOffBacklight(); ; // Backlight off
      ui.backlight_off_ms = 0;
    }
  #endif
```

you may have to add
`  const millis_t ms = millis();` to marlinui.cpp
`under if LCD_BACKLIGHT_TIMEOUT`

thisiskeithb, 2022-10-01:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24826/

closed 2022-10-01 by thisiskeithb