MarlinFirmware/Marlin #24703: [BUG] Update platform/platform-packages in stm32h7.ini for STM32H743Vx_btt to avoid TMC CONNECTION ERROR

JoaquinBerrios, 2022-08-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When building with the default values in stm32h7.ini for platform and platform-packages, the resulting build has UART issues with at least TMC2208, TMC2209 and EZ2209 stepper drivers. Updating to use more current platform and platform-packages resolve those issues. 

This was a day one issue for me regardless of the version of Marlin I was building (2.0.x and 2.1.x). Hardware tested was a BTT SKR 3 EZ with EZ2209 as well as TMC2208 v3.0 stepper sticks on an Ender 3 Max (not that it matters).  

After doing testing, everything is operating as expected and the MCU is no longer reporting TMC connection errors. 

There is very likely a better way to do this, but my successful build that resolved the TMC CONNECTION ERRORS was using the values in STM32h7.ini show below. 
```
...
#
# BigTreeTech SKR V3.0 / V3.0 EZ (STM32H743VIT6 ARM Cortex-M7)
#
[env:STM32H743Vx_btt]
extends                     = stm32_variant
#platform                    = ststm32@~14.1.0
platform                    = ststm32@~15.4.1
#platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
platform_packages           = framework-arduinoststm32@~4.20200.220530
board                       = marlin_STM32H743Vx
board_build.offset          = 0x20000
board_upload.offset_address = 0x08020000
build_flags                 = ${stm32_variant.build_flags}
                            -DPIN_SERIAL1_RX=PA_10 -DPIN_SERIAL1_TX=PA_9
                            -DPIN_SERIAL3_RX=PD_9 -DPIN_SERIAL3_TX=PD_8
                            -DPIN_SERIAL4_RX=PA_1 -DPIN_SERIAL4_TX=PA_0
                            -DSERIAL_RX_BUFFER_SIZE=1024 -DSERIAL_TX_BUFFER_SIZE=1024
                            -DTIMER_SERVO=TIM5 -DTIMER_TONE=TIM2
                            -DSTEP_TIMER_IRQ_PRIO=0
                            -DD_CACHE_DISABLED
upload_protocol             = cmsis-dap
debug_tool                  = cmsis-dap
```

### Bug Timeline

This has happened from every build I have attempted to compile for this board. 

### Expected behavior

Firmware would build and operate normally with error free UART communications to the stepper drivers. 

### Actual behavior

Before making the changes, the MCU would report TMC CONNECTION ERROR message to the display, and M122 command would show ALL LOW errors that would jump around from stepper to stepper. Stepper drivers with errors were also not properly configures as expected for a stepper driver the MCU is not reliably able to communicate with. Output from M122 command is shown below:
```
M122
		X	Y	Z	Z2	E
Address		0	0	0	0	0
Enabled		true	false	false	false	false
Set current	560	560	560	560	600
RMS current	994	550	550	994	581
MAX current	1402	776	776	1402	819
Run current	17/31	17/31	17/31	17/31	18/31
Hold current	8/31	8/31	8/31	8/31	9/31
CS actual	0/31	8/31	8/31	0/31	9/31
PWM scale					
vsense		0=.325	1=.18	1=.18	0=.325	1=.18
stealthChop	false	true	true	false	true
msteps		256	16	16	256	16
interp		false	true	true	false	true
tstep		0	max	max	0	max
PWM thresh.					
[mm/s]						
OT prewarn	false	false	false	false	false
triggered
 OTP		false	false	false	false	false
pwm scale sum	0	10	10	0	11
pwm scale auto	0	0	0	0	0
pwm offset auto	0	36	0	0	36
pwm grad auto	0	14	14	0	14
off time	0	4	4	0	4
blank time	24	24	24	16	24
hysteresis
 -end		-3	2	2	-3	2
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	0	40	40	0	40
DRVSTATUS	X	Y	Z	Z2	E
sg_result	0	0	0	0	0
stst		*		*	*	
olb						
ola						
s2gb						
s2ga						
otpw						
ot						
157C						
150C						
143C						
120C						
s2vsa						
s2vsb						
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0xC0:08:00:00
		Z	0xC0:08:00:00
		Z2	0x00:00:00:00	 Bad response!
		E	0xC0:09:00:00

Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... Error: All LOW
Testing E connection... OK
ok

M122
		X	Y	Z	Z2	E
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	560	560	560	560	600
RMS current	994	550	550	994	581
MAX current	1402	776	776	1402	819
Run current	17/31	17/31	17/31	17/31	18/31
Hold current	8/31	8/31	8/31	8/31	9/31
CS actual	0/31	8/31	8/31	0/31	9/31
PWM scale					
vsense		0=.325	1=.18	1=.18	0=.325	1=.18
stealthChop	false	true	true	false	true
msteps		16	16	16	256	16
interp		false	true	true	false	true
tstep		0	max	0	0	max
PWM thresh.					
[mm/s]						
OT prewarn	false	false	false	false	false
triggered
 OTP		false	false	false	false	false
pwm scale sum	0	10	10	0	11
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	0	36
pwm grad auto	0	14	14	0	14
off time	0	4	4	0	4
blank time	16	24	24	16	24
hysteresis
 -end		-3	2	2	-3	2
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	0	40	40	0	40
DRVSTATUS	X	Y	Z	Z2	E
sg_result	0	0	0	0	0
stst		*			*	
olb						
ola						
s2gb						
s2ga						
otpw						
ot						
157C						
150C						
143C						
120C						
s2vsa						
s2vsb						
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0xC0:08:00:00
		Z	0xC0:08:00:00
		Z2	0x00:00:00:00	 Bad response!
		E	0xC0:09:00:00

Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing Z2 connection... Error: All LOW
Testing E connection... OK
ok
```

### Steps to Reproduce

1. Build Marlin with unmodified stm32h7.ini file 
2. Define either TMC2208 or TMC2209 *_DRIVER_TYPE in Configuration.h
3. Build in VSC/PIO/ABM
4. Flash firmware.bin to printer and TMC CONNECTION ERROR is displayed on the LCD with M122 showing ALL LOW and misconfigured drivers. 

### Version of Marlin Firmware

2.0.x and 2.1.x

### Printer model

Ender 3 Max

### Electronics

SKR 3 EZ motherboard with EZ2209 or TMC2208 v3.0 BTT stepper drivers

### Add-ons

Unlikely to be involved, but the printer is also configured with a CRTouch, TFT35 E3 display,  BIGTREETECH ESP8266 WiFi Transceiver Module (ESP12S), 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Enabling or disabling other features had not effect on the UART communication errors with the stepper drivers. I attempted builds that just configured the board, and stepper drivers and it still resulted in issued. The attached configurations enable all the features I wanted and produced errors with the default stm32h7.ini, but the errors were resolved with when I modified the file as noted above. 

Configurations:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9438660/Configurations.zip)

There was also discussion of this issue on BTT github:
https://github.com/bigtreetech/SKR-3/issues/13

thisiskeithb, 2022-08-27:
Weird. I build for this board/MCU with 2209s all the time without needing to modify anything and don’t have this issue.

JoaquinBerrios, 2022-08-28:
You are not alone, but we had other folks as noted on the BTT git issue that were not able to build without the modifications to the ini file.

thisiskeithb, 2022-08-28:
It’s obvious some of those reports are from users running BTT’s Marlin fork, but I’m wondering if their VSCode / PIO install was damaged.

Some suggestions:

Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

EvilGremlin, 2022-08-28:
Reports are, clean build 100% fail on one machine and 100% succeed on the other, exact same code obviously. No statistics by OS though.
But isn't this what you fixed here? https://github.com/thisiskeithb/Arduino_Core_STM32/commit/8b3522051ac21809c4e76146bbbc546970271039

thisiskeithb, 2022-08-28:
> But isn't this what you fixed here? 

🤷‍♂️ A similar fix was originally proposed by BTT for the Biqu BX, which is what that fork is meant for.

@rhapsodyv showed me that it was properly fixed in later versions of Arduino_Core_STM32, so I backported that change.

JoaquinBerrios, 2022-08-28:
@EvilGremlin, this occurred for me on various clean installs of VSC/PIO on Windows 10, 11, MacOS 11, and Linux (Ubuntu 22.04 in an x64 VM, and Bullseye on a RPi4) so it's not as simple as doing a clean install of POI (I tired). The only way I was able to get a build with working UART to the drivers was by making the edits to explicitly pull in newer versions of ststm32 and framework-arduinoststm32. Others on the referenced BTT git issue had similar experience.

EvilGremlin, 2022-08-28:
Oh woe! We (and i mean all people who understand at least somehing about this code) are always reluctant to bump ststm32 verision because updates always break everything, it's constant catch-up game. But i gues it's this time again. Please tell what exact ststm32 version works, and if you noticed any isuues with it.

donmahallem, 2022-08-28:
@thisiskeithb I do face this issue and I did use straight the latest Bugfix Marlin and just modified the parameters for board and drivers. The never touched the outdated btt marlin version

SteveStation, 2022-08-28:
I have the same problem, i was getting crazy ticking it was my problem or hardware issue, so i am not alone, installed the vscode on 3 computer if the same result, 2 of the computers never had the vscode instaled + platformio.

Pardiman, 2022-08-28:
Hello, I come from the GitHub of BTT.

In case it helps. Updating the platform and packages manually works, but technically, you don't need to update anything to solve the problem, at least in my case. Here's how it worked for me the first time:

https://github.com/bigtreetech/SKR-3/issues/13#issuecomment-1229138177

> I have managed to successfully compile a firmware. I haven't been able to test if everything works, but the M122 is correct and my Z axis seems to have correct torque.
> 
> If you look at the build log the problem seems to be in:
> 
> `- framework-arduinoststm32 @ 0.0.0`
> 
> When you delete the following line in "`..\ini\stm32h7.ini`":
> 
> `platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip`
> 
> Platformio installed version 2.0.0 in a predefined way, looking in the build log like this:
> 
> `- framework-arduinoststm32 @ 4.20000.210603 (2.0.0)`
> 
> Here it already works, but by adding the following to the "stm32h7.ini" file, instead of the previously deleted line, I have managed to install version 2.2.0:
> 
> `platform_packages = framework-arduinoststm32@>4.20000.210603`
> 
> Or also:
> 
> `platform_packages = framework-arduinoststm32@4.20200.220530`
> 
> Here I have stayed, there is version 2.3.0, which is the one that should be installed with the original code, but I have not managed to install it. Please try one of you to confirm that this is correct or if there is a better way to do it.

I did not see the build log of anyone who works TMC without modifying anything, but is it normal that when building the version is not shown? 

 `- framework-arduinoststm32 @ 0.0.0`

I'm sorry I can't be more technical, but my knowledge is limited.

donmahallem, 2022-08-28:
@Pardiman as stated in the other issue:

The master branch is referenced which in itself has no version(tag). So the cli infers/sets it as 0.0.0, because it can't determine its version

Pardiman, 2022-08-28:
True @donmahallem , I remember you already told me something about it.

From here I get lost and I think I can't help.

JoaquinBerrios, 2022-08-28:
@EvilGremlin, I'm using the versions noted in the original bug report, with a fair number of features enabled with no issues that I have noticed. My printer is currently printing the first real print since I got a working build and everything is working as expected five hours into a 14 hour print. I know it's a sample size of one and only one set of stepper drivers using only UART, but wanted to provide the feedback nonetheless.

gplv2, 2022-08-31:
I can confirm this has also fixed it for me :  also on SKR 3 EZ motherboard with EZ2209 steppers.  This is a good fix.

thisiskeithb, 2022-08-31:
@bigtreetech: Are there undocumented versions of the SKR 3 series? Why does this only affect some users?

JoaquinBerrios, 2022-08-31:
@thisiskeithb, someone on the BTT issue posted a build that a friend built for them using the same config files they were using. The build from the friend did not have the TMC connection issues and theirs did. My guess is that there is a tweak in the build environment somewhere for those who can get a working build vs those that can't without updating the default ini files. 

I think it's unlikely to be hardware related since the builds that work seem to work on all boards. This seems all the more likely since the folks with good builds seem to be more involved with developing and building Marlin more frequently for STM32 that the folks posting on the BTT issue with the problem and therefore more likely to have modified their environment somewhere along the way vs. a new user who is potentially building Marlin for the first time for STM32. I myself have built Marlin several times, but this is the first time building for an STM32 MCU and I certainly don't do it every day or anything. 

Still, it would be interesting to test a broken build on a board for someone who has been building with out issues, but my guess is that the broken builds will be broken on all boards. 

I can provide a .bin file using a clean install of VSC/PIO that triggers the connection issue on my board for you if you wanted to test and trusted a total stranger's binary, if you think that would be helpful. I can take a default config and just configure a serial port and four TMC 2209 steppers (default 800 ma current),  TMC debugging enabled, with end stop switches and a 200^3 mm build volume, or if you want to provide configuration.h and configuration_adv.h files I can build against those. 

It would also be great if @bigtreetech  could confirm one way or the other whether there have been any hardware revisions or not.

thisiskeithb, 2022-08-31:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24711

closed 2022-08-31 by thisiskeithb

gplv2, 2022-09-01:
@thisiskeithb Just tried your patch and looks to be working perfectly.  Thanks for the work.

Tezzatron81, 2022-10-24:
Can someone explain which file needs to be edited to get this to work. I have skr3 ez with ez2209 and getting the same tmc errors.

Pardiman, 2022-10-24:
@Tezzatron81 No longer necessary if you don't want to. The problem is fixed in the bugfix-2.1.x version.