MarlinFirmware/Marlin #24705: [BUG] Missing "Back" item on Ender 3 Pro

jaevans, 2022-08-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With latest bugfix-2.1.x custom build for Ender-3 Pro w/Creality 4.7.2 main board the LCD menus have no "back" capability. Looking at the past few days of changes, it looks like commit 2f91154cbdc added a conditional to control the back button. I've checked the [Configurations](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427) repo for updates around this, but didn't find anything that looks right to me.

For the conditional
```C
#if DISABLED(DISABLE_ENCODER) && ANY(HAS_ROTARY_ENCODER, HAS_ADC_BUTTONS) && ANY(TFT_CLASSIC_UI, TFT_COLOR_UI)
```
all values are 0 in my config. It seems like `HAS_ROTARY_ENCODER` and `TFT_CLASSIC_UI` should be set?

I think I can hack around it for the moment by defining `HAS_BACK_ITEM` as `1` in my `Configuration_adv.h`, but I believe this should be on by default. 

Looks like it was addressed for some configurations in https://github.com/MarlinFirmware/Marlin/pull/24690, but that doesn't fix my setup.

### Bug Timeline

new, created by #24618

### Expected behavior

Back menu item appears in sub-menus

### Actual behavior

No way to go back up the tree

### Steps to Reproduce

1. Compile and flash firmware including commit 2f91154cbdcec3b81da9913bdd16556e8c775d06

### Version of Marlin Firmware

Bugfix-2.1.x 2022-08-27

### Printer model

Creality Ender 3 Pro

### Electronics

Creality mainboard 4.2.7

### Add-ons

BLtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Tried with stock `Configuration.h` and `Configuration_adv.h` from Configurations repo with same resul
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9439940/configuration.zip)
t

EvilGremlin, 2022-08-28:
https://github.com/MarlinFirmware/Marlin/pull/24694

thisiskeithb, 2022-08-28:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24689

closed 2022-08-28 by thisiskeithb