MarlinFirmware/Marlin #24712: SD card is not reading...

FreeZuu69, 2022-09-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

SD card is not reading.

my system

Mks Gen L v2.1
Mks H43 Display
Mks H43 Adapter
Mks PWC v3.0

[image]

### Bug Timeline

a new error, it does not read the card in any way.

### Expected behavior

As soon as the printer is turned on, the contents of the sd card should appear, but the files are not visible. There is no such problem in the previous version. Marlin-V2.X-MKS-H43-main

### Version of Marlin Firmware

Marlin-2.1.1

### Printer model

Corexy

### Electronics

Mks  gen L 2.1 , Mks Pwc, Mks H43 LCD, bltouch, h43 a adapter

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9467841/Marlin.zip)

borland1, 2022-09-02:
Disabled by default...

Configuration.h
 ```
* SD CARD
 *
 * SD Card support is disabled by default. If your controller has an SD slot,
 * you must uncomment the following option or it won't work.
 */
//#define SDSUPPORT

/**
 * SD CARD: ENABLE CRC
 *
 * Use CRC checks and retries on the SD communication.
 */
//#define SD_CHECK_AND_RETRY
```

FreeZuu69, 2022-09-02:
I tried both of them differently. nothing has changed, the problem persists.

ellensp, 2022-09-28:
Provided Config does not compile
```
Compiling .pio/build/mega2560/src/src/core/serial.cpp.o
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/HAL.cpp:24:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3186:4: error: #error "TMC2208 or TMC2209 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN."
```

Updated to current bugfix configs and added Z2 pins

[Configuration-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/9662770/Configuration-2.0.x.zip)

ellensp, 2022-09-28:
no sd detect pin either
  #define SD_DETECT_PIN                       49

ellensp, 2022-09-28:
Odd behavior during testing, with the MKS H43 Adapter-A connected  and the SD_DETECT_PIN defined

If you disable DGUS_LCD_UI_MKS and enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
The sd card works as expected

But as soon as you enable DGUS_LCD_UI_MKS and disable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
The sd card fails to work...

ellensp, 2022-09-28:
It seem to be a memory issue...  to many things enabled on old mega2560 with only 4k 
If you for example change TMC2209 to TMC2209_STANDALONE it frees up enough ram to work as expected.

jmger, 2023-02-03:
YES, I have the same issue. I have the Creality Ender 3 Max with BTT Octopus Pro F429 board. I enabled SDSUPPORT and CRC check. The SDCARD was working in 2.0.x. My BTT BOARD uploaded and installed the 2.1.2 Firmware file just fine. Once or twice I get the message Initialization fail. For the SDCARD.

jmger, 2023-02-04:
HI for at least my issue where SDCARD was not detected/working in Marlin 2.1.2. I had to add the line #define SDCARD_CONNECTION ONBOARD to the Configuration.adv file. Apparently, This line was left out. Now everything is working.

inspectionsbybob, 2023-03-26:
Forgive me, but I am having similar problems with the MKS_Robin_Pro

When I look at the Config_Adv.H file I can find nothing mentioning the "# define SDcard_CONNECTION_ONBOARD" and normally there is a "If SDcard_CONNECTION_OMBOARD " followed by a whole lot of defines and other stuff. Where did you find the code that uses the flag you say is missing?

agolinko, 2023-07-08:
Not working for me too.
Hardware: Kingroon K3PS/Robin nano board,  Marlin 2.1.2.1
Setting SDCARD_CONNECTION to ONBOARD didn't help

bhargavbe, 2023-08-19:
I am also facing the same issue now, suddenly !!!, was working fine before,  in 2.1.2.1 branch, It had another issue of stopping print intermediately without any reason.
Now print doesn't even start. 
For additional information.
1. I have tried to print from SD card from menu. 
2. I have tried to print from SD card from prontoface.
3. I have tried to print from prontoface as a host and selected file from my PC.
 In all above scenario "Not even first line execute". So it should not be the issue specific in SD Card but it's the issue related to print.
 
 All commands sent from serial port directly works fine,

mechano, 2023-09-11:
> Not working for me too. After boot, contents of SDCARD is visible. But when select something to print, after confirm, it does nothing. When trying to open SD card again it shows empty card.
> 
> Hardware: Kingroon K3PS/Robin nano board, BLTouch Marlin 2.1.2.1

Same beaviour, everything perfect and fully working but the SD reader. Files appears I select one and it doesn't print. When I go back to media the files disappear. If I extract and insert again the SD card they appeat again but still no printing. Tryed both 2.1.2.1 and 2.1.2. 

I saw around 2.1.2 firmwares fully working. So I think it's onlt a misconfiguration.

ElMedic0, 2024-01-28:
I had the same issue, fixed by uncommenting those if you are using the lcd sdcard slot,
//#define SDCARD_CONNECTION LCD
//#define NO_SD_DETECT 
And also left the power loss commented (disabled).
//#define POWER_LOSS_RECOVERY

thinkyhead, 2026-10-08:
Thanks for reporting this. A lot has changed in this area since it was filed. Could you please flash the current `bugfix-2.1.x` and tell us whether the problem still happens? If it does, please attach your current `Configuration.h` and `Configuration_adv.h` and a short description of the steps. We're trying to clear the decks for the 2.1.3 release, so a quick retest helps a lot.

closed 2026-10-20 by thinkyhead