MarlinFirmware/Marlin #24717: [BUG] error: 'NUM_ANALOG_FIRST' was not declared in this scope; did you mean 'NUM_ANALOG_INPUTS'?

Altxiety, 2022-09-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After i start building using Auto Martin, it gives me this error: 

> Marlin\src\gcode\config\../../pins/../HAL/STM32/pinsDebug.h:171:23: error: 'NUM_ANALOG_FIRST' was not declared in this scope; did you mean 'NUM_ANALOG_INPUTS'?
[image]

Im currently trying to make SKR mini E3 V3.0 work with my Anet A6

### Version of Marlin Firmware

2.1.1

### Printer model

Anet A6

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9480465/Marlin.zip)
 config files

thisiskeithb, 2022-09-02:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23386

thisiskeithb, 2022-09-02:
Disable `PINS_DEBUGGING` since it doesn’t work for this board.

closed 2022-09-02 by thisiskeithb