MarlinFirmware/Marlin #24724: PowerLoss Recovery Fail

FreeZuu69, 2022-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I resume it after a power cut, the nozzle and bed do xy home without getting hot, then it warms up and tries to continue. I think there is a problem with the instructions necessary for the start in the plr file he saved. I'm asking for help on this. You can find the relevant files in zip form attached.

### Bug Timeline

after power failur

### Expected behavior

I was waiting for it to make xy home after the nozzle and bed warmed up. Since the nozzle, which has cooled down in its current state, sticks to the part, the part is removed from the table during xy home.

### Actual behavior

nozzle and bed must be warmed before moving

### Version of Marlin Firmware

Marlin-V2.X-MKS-H43-main

### Printer model

Corexy

### Electronics

Mks  gen L 2.1 , Mks Pwc, Mks H43 LCD, bltouch, h43 a adapter

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Uploading Marlin-V2.X-MKS-H43-main.zip…]()
[power loss recovery.zip](https://github.com/MarlinFirmware/Marlin/files/9482929/power.loss.recovery.zip)

FreeZuu69, 2022-09-03:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9483175/Configuration.zip)

ellensp, 2022-09-05:
https://github.com/makerbase-mks/Marlin-V2.X-MKS-H43 is a fork of marlin.

Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2022-11-15 by github-actions[bot]