MarlinFirmware/Marlin #24725: [BUG] (Marlin 2.1.1 will not compile using creality ender 3/ender 3 pro 422 board example configs)

Mikabella, 2022-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am attempting to compile a new firmware for someone who is using a creality 422 board on an ender 3 pro and before i make any changes i like to double check that a base firmware will indeed compile. However when i attempt this with Marlin 2.1.1 i keep running into issues if i try to compile using the example configurations for Crealitys Ender 3 or Ender 3 Pro 422 board.

It works just fine building any other config file for me. [as an example, BTT config files compile just fine](https://i.imgur.com/B2SrJEy.png) 
For the example its the exact same folder, only switched out configuration.h & configuation_adv.h files.

### Bug Timeline

New, As of 9/2/2022 9pm EST

### Expected behavior

Marlin 2.1.1 using ender 3 (Or ender 3 pro) Config files to compile into a firmware.bin file using autobuild marlin or platformio

### Actual behavior

It simple errors out and fails to compile.

Using Autobuild Marlin Extension: [Image Link](https://i.imgur.com/QknmEW6.png)
Using Platformio extension: [Image Link](https://i.imgur.com/wmiBlfw.png)

Error Coding: 
```
PS C:\Firmware\Jenn FW\Marlin-2.1.1> platformio run -e STM32F103RE_creality
Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\Mikab\\.platformio\\packages\\framework-arduinoststm32\\variants\\MARLIN_F103Rx':
  File "C:\Users\Mikab\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\Mikab\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Mikab\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
  File "C:\Firmware\Jenn FW\Marlin-2.1.1\buildroot\share\PlatformIO\scripts\generic_create_variant.py", line 52:
    variant_dir.mkdir()
  File "C:\Users\Mikab\.platformio\python3\lib\pathlib.py", line 1323:
    self._accessor.mkdir(self, mode)
================================================================================================== [FAILED] Took 1.84 seconds ==================================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:01.838
============================================================================================= 1 failed, 0 succeeded in 00:00:01.838 ============================================================================
PS C:\Firmware\Jenn FW\Marlin-2.1.1> echo "done" >"C:\Users\Mikab\AppData\Local\Temp\ipc"
PS C:\Firmware\Jenn FW\Marlin-2.1.1>
```

### Steps to Reproduce

Download Marlin 2.1.1 & the corresponding config files.
Replace default config & adv config in marlin folder
Open vscode
Attempt to build firmware using either autobuild marlin or platformio

### Version of Marlin Firmware

2.1.1

### Printer model

Ender 3 Pro

### Electronics

Stock 422 Board

### Add-ons

Cr Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

My Marlin folder - [Marlin-2.1.1.zip](https://github.com/MarlinFirmware/Marlin/files/9483129/Marlin-2.1.1.zip)

EvilGremlin, 2022-09-03:
Not a bug, wait a bit and run build again. First run is kinda expected to fail with obscure error, pio and vscode doing a lot in background.

Mikabella, 2022-09-03:
> Not a bug, wait a bit and run build again. First run is kinda expected to fail with obscure error, pio and vscode doing a lot in background.

I left vscode open overnight and tried again in the morning and it still did not compile :/

jimmysdiy, 2022-09-04:
this is not a bug. I tested 4.2.7 and 4.2.2 and SKR Mini e3 v2. They all compile fine

ellensp, 2022-09-04:
delete the directory "C:\Users\Mikab\.platformi`  and restart vscode,
It will reinstall platformio. 
Hit build and wait for it to download all required tools and build.

closed 2022-09-04 by ellensp