MarlinFirmware/Marlin #24727: [FR] Enable WiFi standlone module is confused

Johnny-Martin, 2022-09-04:
### Is your feature request related to a problem? Please describe.

My board is Fysetc Spider 2.3, and I buy a wifi module(SPIDER-RRF-WIFI) for the board so that I can control by 3D printer with my laptop.

The Wifi module need a serial port, and the problem is, I don't know how to configrate these two serial ports to enable the WiFi module by reading the comment in the header file.

`/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 1
.........
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
`

What value should I assign to these two serial ports? 
And how to turn on the wifi module?  

Please help me, thanks a lot!

### Describe the feature you want

I hope the user can understand every number means for the serial port by reading the comment in the header file.

ellensp, 2022-09-04:
You need to look at the specs and circuit diagrams for your controller. 

https://wiki.fysetc.com/images/Spider_V2.3_Pinout.jpg

Its plugged into exp1 amd 2 so it has to be a serial port on those connectors

PC6 and PC7 are UART 6

So it looks like you need serial port 6 enabled

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2022-09-05:
reopening..
issue is not only serial port
SPIDER-RRF-WIFI is running DuetWiFiSocketServer firmware code...  that marlin doesn't know how to talk to   
https://github.com/gloomyandy/DuetWiFiSocketServer/releases

thisiskeithb, 2022-09-05:
> reopening.. issue is not only serial port SPIDER-RRF-WIFI is running DuetWiFiSocketServer firmware code... that marlin doesn't know how to talk to https://github.com/gloomyandy/DuetWiFiSocketServer/releases

They are trying to run RRF-related firmware on that module. It needs to be flashed with something like ESP3D for it to work with Marlin, so it's not really a bug.

ellensp, 2022-09-06:
That is why its a feature request.

Johnny-Martin, 2022-09-13:
Thanks for reopening this issue, I had reverse engineering of SPIDER-RRF-WIFI module, and here is the schematic:
[image]

The EXP1  of this WiFi module is for UART communication.

I will try to flash this module with ESP3D firmware in few days.

thisiskeithb, 2024-04-16:
Any updates @Johnny-Martin?